adafruit/circuitpython #6811: ESP32-S2 pullups enabled when idle

HelenFoster, 2022-08-24:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; Adafruit QT Py ESP32S2 with ESP32S2
```

### Code/REPL

```python
# A1 connected to ground through 10K
# Start REPL: 0.7V
import board, digitalio
d = digitalio.DigitalInOut(board.A1)  # goes to 0V
d.pull = digitalio.Pull.UP  # goes to 0.7V
d.pull = None  # goes to 0V
d.deinit()   # goes to 0.7V
```

### Behavior

A0 is documented to have a 10K pullup, but all the other GPIOs seem to have ~40K pullup. Other boards have the pins floating in this case, right?

dhalbert, 2022-08-25:
The pins on Espressif boards are reset with a pull-up on. We are using the ESP-IDF's default reset pin routine. Using a pull-up actually causes less power consumption than having the pin float, according to Eespressif. See https://github.com/espressif/esp-idf/blob/23d5a582cb27d7c909d52557ca0ddceade510db8/components/driver/gpio.c#L393-L406.

Originally we floated the pin. This also caused some issues with random behavior of NeoPixels attached to pins. The reset was changed to the ESP-IDF default in PR #5892.

HelenFoster, 2022-08-25:
OK, thanks for the info.

closed 2022-08-25 by HelenFoster