adafruit/circuitpython #6812: Web Workflow: Renaming a folder to be inside itself removes it

makermelissa, 2022-08-24:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0
```

### Code/REPL

```python
N/A
```

### Behavior

If you go into the offline editor http://circuitpython.local/fs/ and rename a folder, it shows the full path. If you set it so it is inside itself (for instance you rename /test/ to /test/test2/) then the folder disappears with no error. I would expect an error message.

makermelissa, 2022-09-01:
I looked into this a bit and it appears renaming just uses `f_rename`, which appears to be part of tinyusb (https://github.com/hathach/tinyusb/blob/master/lib/fatfs/ff.c#L3500), so this issue may be more pervasive than I thought and possibly in the wrong place.

dhalbert, 2022-09-01:
I think it is using the `f_rename()` in `circuitpython/lib/oofatfs/ff.c`. TinyUsb has a version too, but we don't use that implementation of fatfs.

makermelissa, 2022-09-01:
Ah, ok. Thanks.

makermelissa, 2022-09-01:
Ok, so the way it looks like it works is:
1. It checks that the new location exists and there isn't another entry of the same name
2. It creates a copy of the folder inside there
3. It removes the original folder

Perhaps a check needs to be added to see if the

dhalbert, 2022-09-06:
This was actually fixed years ago in #1645, but that fix is in `fat_vfs_rename()` in `extmod/vfs_fat.c`, and the web workflow code is calling `f_rename()` directly, which bypasses the fix. We need to study a bit more about whether to move the check into `f_rename()` or call `fat_vfs_rename()` instead.

makermelissa, 2022-09-09:
It looks like Scott used `f_rename()` instead of `fat_vfs_rename()` so that he could get some status information about specific failures and relay that info back via HTTP error codes. Adding some additional checks from `fat_vfs_rename()` to Web Workflow should be pretty straightforward.

PR #6883 (2022-10-12): Prevent folder from trying to move inside itself
Fixes #6812. 

Tested on Feather ESP32-S3 TFT using Web Workflow offline file manager:
- Tested that attempting to move a folder into itself fails
- Tested that attempting to move folder into another folder that doesn't exist fails
- Tested that moving a folder into another folder that does exist still works
- Tested that moving a file into a folder that exists still works
- Tested that renaming a folder still works
- Tested that renaming a file still works
- Tested renaming and moving in the same operation works
closed 2022-10-12 by dhalbert