adafruit/circuitpython #6821: Os.stat missing documentation

thetazero, 2022-08-25:
The [os.stat](https://docs.circuitpython.org/en/7.3.x/shared-bindings/os/index.html#os.stat) docs do not fully describe the behavior. 
From the docs it is unclear what the returned tuple represents.
[image]

BrianPugh, 2022-08-28:
the returned tuple is the [same as in normal python](https://docs.python.org/3/library/os.html#os.stat).

Of which, the following are probably properly populated:

`os.stat()[0]`  file type, regular or directory. 
`os.stat()[6]`  file size
`os.stat()[7], os.stat()[8], os.stat()[9]` are all the same and tell the modification time. For that value to be correct, the time has to be set.

thetazero, 2022-08-29:
Thanks @BrianPugh, I've created a pull improving the documentation based on your comments.

PR #6841 (2022-09-01): Improve os.stat docs
To resolve #6821.
closed 2022-09-01 by MicroDev1