adafruit/circuitpython #6824: please merge similar messages

jepler, 2022-08-26:
I noticed both of these...
```
msgid "All I2C peripherals are in use"
msgid "I2C peripheral in use"
```

dhalbert, 2022-08-29:
The first means there are no more peripherals available. The second means you chose pins for a particular peripheral  instance that is already in use. For example, on RP2040 you chose pins assigned to I2C0, which happened to already be in use, but I2C1 might be free.

jepler, 2022-08-29:
If it's important that these messages be distinct then close this up... I made this issue without checking into it deeply.

jepler, 2022-08-29:
```
locale/circuitpython.pot:msgid "%q in use"
locale/circuitpython.pot:msgid "A hardware interrupt channel is already in use"
locale/circuitpython.pot:msgid "All CAN peripherals are in use"
locale/circuitpython.pot:msgid "All I2C peripherals are in use"
locale/circuitpython.pot:msgid "All I2C targets are in use"
locale/circuitpython.pot:msgid "All PCNT units in use"
locale/circuitpython.pot:msgid "All RX FIFOs in use"
locale/circuitpython.pot:msgid "All SPI peripherals are in use"
locale/circuitpython.pot:msgid "All UART peripherals are in use"
locale/circuitpython.pot:msgid "All channels in use"
locale/circuitpython.pot:msgid "All event channels in use"
locale/circuitpython.pot:msgid "All state machines in use"
locale/circuitpython.pot:msgid "All sync event channels in use"
locale/circuitpython.pot:msgid "All timers for this pin are in use"
locale/circuitpython.pot:msgid "All timers in use"
locale/circuitpython.pot:msgid "Bus pin %d is already in use"
locale/circuitpython.pot:msgid "Cannot vary frequency on a timer that is already in use"
locale/circuitpython.pot:msgid "Clock unit in use"
locale/circuitpython.pot:msgid "DAC already in use"
locale/circuitpython.pot:msgid "Device in use"
locale/circuitpython.pot:msgid "EXTINT channel already in use"
locale/circuitpython.pot:msgid "Hardware in use, try alternative pins"
locale/circuitpython.pot:msgid "I2C peripheral in use"
locale/circuitpython.pot:msgid "PWM slice already in use"
locale/circuitpython.pot:msgid "PWM slice channel A already in use"
locale/circuitpython.pot:msgid "Peripheral in use"
locale/circuitpython.pot:msgid "Pin interrupt already in use"
locale/circuitpython.pot:msgid "SPI peripheral in use"
locale/circuitpython.pot:msgid "Serializer in use"
locale/circuitpython.pot:msgid "data pin #%d in use"
```
I think there might be _something_ here to reduce redundant strings...

deshipu, 2023-07-29:
I don't think this is a good first issue anymore, because it's not clear what needs to be done. I'm going to remove the label.

jepler, 2023-07-29:
This is probably not particularly actionable.

closed 2023-07-29 by jepler