adafruit/circuitpython #6838: ESP32-SE TFT board.DISPLAY.brightness no longer works

alison-gravley, 2022-08-29:
### CircuitPython version

```python
Version: 8.0.0-beta.0
```

### Code/REPL

```python
import time
import board
import adafruit_si7021
import digitalio
import busio
import displayio
import sensorDataDisplay
import alarm
import neopixel
from adafruit_displayio_layout.layouts.grid_layout import GridLayout
  
# The battery sensor is current broken and needs to be fixed in the firmware.
env_sensor = adafruit_si7021.SI7021(board.I2C())
# battery_sensor = LC709203F(board.I2C())

# Main Display Group
main_group = displayio.Group()
np = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness=.05, auto_write=True)

layout = GridLayout(x=0,
                    y=0,
                    width=32+50,
                    height=(32*2)+40,
                    grid_size=(2, 2),
                    cell_padding=10
                    )

tempLayout = sensorDataDisplay.createTemperatureLayout()
humidLayout = sensorDataDisplay.createHumidtyLayout()
layout.add_content(cell_content=tempLayout[0],
                   grid_position=(0,0),
                   cell_size=(1,1))

layout.add_content(cell_content=tempLayout[1],
                   grid_position=(1,0),
                   cell_size=(1,1))

layout.add_content(cell_content=humidLayout[0],
                   grid_position=(0,1),
                   cell_size=(1,1))

layout.add_content(cell_content=humidLayout[1],
                   grid_position=(1,1),
                   cell_size=(1,1))

np[0] = (0,100,0)
board.DISPLAY.brightness = 1.0
board.DISPLAY.show(layout)
dimTime = time.monotonic()
currentlyDimmed = False
loopNumber = 0

while True:
    tempLayout[1].text = sensorDataDisplay.temperatureText(env_sensor)
    humidLayout[1].text = sensorDataDisplay.humidityText(env_sensor)
    time.sleep(1)
    currentTime = time.monotonic()
    loopNumber+=1
    if((currentTime > dimTime + 30)) :
        if currentlyDimmed :
            # We have been Dim for 20 seconds, go to sleep
            currentlyDimmed = False
            dimTime = currentTime
            board.DISPLAY.brightness = 0
            np[0] = (50,0,0)
            time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic()+10)

            #sensorDataDisplay.turnPowerOff(board.TFT_CS)
            alarm.light_sleep_until_alarms(time_alarm)
            #sensorDataDisplay.turnPowerOn(board.TFT_CS)
            # Wake up and set brightness back to full
            np[0] = (0,100,0)
            board.DISPLAY.brightness = 1.0
        else :
            # Set brightness to 10% and set light to blue
            board.DISPLAY.brightness = .01
            np[0] = (0,0,100)
            currentlyDimmed = True
            dimTime = time.monotonic()
```

### Behavior

No error messages. Screen stays at max brightness and does not turn off.

### Description

Changing board.DISPLAY.brightness does not change the brightness of the screen anymore. Worked in alpha version, but does not work in beta.

### Additional information

Have not had time to look in to it, but I will be able to this week sometime. Let me know if you need any additional information.

alison-gravley, 2022-08-29:
I am still digging through the codebase, but I did figure out how to change the brightness. If I turn auto_refresh off, set the brightness, and then manually refresh in my While loop, it works as expected. Otherwise, brightness just automatically goes back to 1.0 no matter what I set it to. So your issue is somewhere in your auto refresh code. Hope this helps!

```
...
board.DISPLAY.auto_refresh = False
board.DISPLAY.brightness = 1.0
board.DISPLAY.show(layout)
dimTime = time.monotonic()
currentlyDimmed = False
loopNumber = 0

while True:
... # same timers/if code as original post
board.DISPLAY.refresh() # added to end of While loop
```

dhalbert, 2022-09-12:
Fixed by #6894, but now the brightness must be set manually at the beginning, or the display is dark. Opening a new issue for that.

closed 2022-09-12 by dhalbert