adafruit/circuitpython #6843: Outputing a MemoryError when executing processes that play mp3 audio 13 or 14 times

iechahk, 2022-08-30:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.1 on 2022-06-22; Adafruit ItsyBitsy nRF52840 Express with nRF52840
Board ID:itsybitsy_nrf52840_express
```

### Code/REPL

```python
################## main.py ######################
import asyncio
import time
import module1
import config
import async_task

DistRange = [config.DISTANCE['KEIKOKU'], config.DISTANCE['KIKEN']]
checkResult = True
try:
    checkResult = module1.check_config(DistRange)
except:
    checkResult = False

if checkResult:
    asyncio.run(async_task.main())
else:
    module1.mp3l("センサ設定誤り.mp3")
    time.sleep(3)
################## main.py ######################

################## async_task.py ######################
import asyncio
import board
import time
import digitalio
import sensor_ble
import adafruit_hcsr04
import config
import module1
from adafruit_ble import BLERadio

sonar = adafruit_hcsr04.HCSR04(trigger_pin=board.D10, echo_pin=board.D9)

risk = 0

led = digitalio.DigitalInOut(board.L)
led.direction = digitalio.Direction.OUTPUT

ble = BLERadio()
advertisement = sensor_ble.advel()

def keihol(onselfDL, opponentDL):
    if onselfDL >= 1 and opponentDL >= 1:
        if opponentDL == 1:
            return opponentDL
        if opponentDL == 2:
            return opponentDL

async def sensor_task():
    global risk
    dist = [0, 0]
    dist[1] = config.SENSOR['MAX']
    DistRange = [config.DISTANCE['KEIKOKU'], config.DISTANCE['KIKEN']]

    while not ble.connected:
        return
    while ble.connected:
        try:
            dist[0] = dist[1]
            dist[1] = sonar.distance
            risk = module1.kikendl(DistRange, dist)
            sensor_ble.writel(risk)
        except RuntimeError:
            risk = 0
            sensor_ble.writel(risk)

        await asyncio.sleep(config.INTERVAL)

async def kikend_task():
    while not ble.connected:
        return
    while ble.connected:
        opprisk = sensor_ble.readl()
        if opprisk:
            result = keihol(risk, opprisk)
            if result == 1:
                module1.mp3l("警告音声.mp3")
                cnt = 0
                while True:
                    led.value = False
                    await asyncio.sleep(0.5)
                    led.value = True
                    await asyncio.sleep(0.5)
                    cnt += 1
                    if cnt == 3:
                        break
            if result == 2:
                module1.mp3l("危険音声.mp3")
                cnt = 0
                while True:
                    led.value = False
                    await asyncio.sleep(0.25)
                    led.value = True
                    await asyncio.sleep(0.25)
                    cnt += 1
                    if cnt == 6:
                        break

        await asyncio.sleep(config.INTERVAL)

async def main():
    while True:
        ble.start_advertising(advertisement)

        while not ble.connected:
            led.value = False
            pass
        ble.stop_advertising()

        while ble.connected:
            led.value = True
            await asyncio.gather(sensor_task(),  kikend_task())
################## async_task.py ######################

################## module1.py ######################
import board
import time
from audiomp3 import MP3Decoder
import config

try:
    from audioio import AudioOut
except ImportError:
    try:
        from audiopwmio import PWMAudioOut as AudioOut
    except ImportError:
        pass  # not always supported by every board!
audio = AudioOut(board.SDA)

def mp3l(name):
    if audio.playing:
        pass

    mp3 = open(name, "rb")
    decoder = MP3Decoder(mp3)
    decoder.file = open(name, "rb")
    audio.play(decoder)

def check_config(configlist):
    result = True
    for key in configlist:
        print(key)
        if (key < config.LIMIT['LOWER'] or key > config.LIMIT['UPPER']):
            result = False
        elif (configlist[0] <= configlist[1]):
            result = False
        else:
            pass
    return result

def kikendl(listA, listB):

    keikoku = listA[0]
    kiken = listA[1]

    zenkai_k = listB[0]
    konkai_k = listB[1]

    result = 0

    if konkai_k < zenkai_k :
        if konkai_k <= keikoku and kiken < konkai_k :
            result = 1  
        elif konkai_k <= kiken :
            result = 2
        else :
            pass
    else :
        pass
    return result
################## module1.py ######################
```

### Behavior

Traceback(most recent call last):
  File "asyncio/core.py", line 214, in run_until_complete
  File "/lib/async_task.py", line 57, in kikend_task
  File "/lib/module1.py", line 13, in mp3l
MemoryError: memory allocation failed, allocating 4608 bytes

[source.zip](https://github.com/adafruit/circuitpython/files/9451981/source.zip)

### Description

I'm Japanese so please excuse my poor English.
Outputing an error when executing processes that play mp3 audio 13 or 14 times.
Please tell me how can i remove this error.

Specific of mp3:
 size : 23.5KB
 bit rate : 64kbps
 sound source length : 3 seconds

dhalbert, 2022-09-05:
Could you trying closing the file after you finish playing it?

jepler, 2022-09-06:
.. and from the same guide page here's the "official" way to play multiple MP3 files sequentially. as far as I know, this can play dozens and dozens of files without encountering MemoryError (I've left it playing an album on repeat overnight):
```python
# You have to specify some mp3 file when creating the decoder
mp3 = open(mp3files[0], "rb")
decoder = MP3Decoder(mp3)
audio = AudioOut(board.A0)

while True:
    for filename in mp3files:
        # Updating the .file property of the existing decoder
        # helps avoid running out of memory (MemoryError exception)
        decoder.file = open(filename, "rb")
        audio.play(decoder)
        print("playing", filename)

        # This allows you to do other things while the audio plays!
        while audio.playing:
            pass

        print("Waiting for button press to continue!")
        while button.value:
            pass
```

iechahk, 2022-09-07:
Hi jepler! Thank you for answering.
I'll refer to the web site and review the source code.

snkYmkrct, 2022-09-23:
@jepler 

This seems to be more of a mode of use problem, than an actual bug. 

I think one way to definitively solve this is to turn MP3Decoder into a singleton, and not allow multiple instances of it.
Would this be possible  / desirable in CircuitPython ?

jepler, 2022-10-03:
While the object should almost always be used "like a singleton", there are some uses for multiple MP3 decoders, used in conjunction with AudioMixer, and this is actually even possible on the samd51 if you're careful about doing it. (you create two MP3Decoder objects, both near the top of the script)

So, even though it would be helpful for simple programs if the MP3Decoder was actually a singleton, we don't want to foreclose on more complicated usage especially as microcontrollers get more RAM and processing power.

closed 2022-10-14 by jepler