adafruit/circuitpython #6849: FunHouse Crash into the HardFault_Handler.

slootsky, 2022-08-31:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.2 on 2022-07-20; Adafruit FunHouse with ESP32S2
Board ID:adafruit_funhouse
```

### Code/REPL

```python
import adafruit_logging as logging
from config import config

logger=logging.getLogger('logger')
# logger.setLevel(config.get('loglevel',logging.WARNING))

import another
```

### Behavior

```Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

### Description

okay.  I get that there's a lot going on here, and a lot of it isn't perfect.  Some of it maybe even should generate a runtime error.  But getting it to crash into the HardFault_Handler seems unexpected :-)

I've attached all the files involved in this  (renamed from .py to .txt because that's what the uploader would accept).  [don't worry, secrets.py doesn't really have any secrets in it anymore]

This is the output of circup freeze
```adafruit_ahtx0==1.0.16
adafruit_bme680==3.4.10
adafruit_ccs811==1.3.12
adafruit_debouncer==2.0.5
adafruit_dotstar==2.2.5
adafruit_fakerequests==1.0.11
adafruit_logging==4.1.6
adafruit_miniqr==1.3.14
adafruit_pixelbuf==1.1.8
adafruit_requests==1.12.8
[config.txt](https://github.com/adafruit/circuitpython/files/9457686/config.txt)
[secrets.txt](https://github.com/adafruit/circuitpython/files/9457687/secrets.txt)
[another.txt](https://github.com/adafruit/circuitpython/files/9457688/another.txt)
[code.txt](https://github.com/adafruit/circuitpython/files/9457689/code.txt)

adafruit_scd4x==1.3.6
adafruit_ticks==1.0.7
adafruit_tmp117==1.1.6
neopixel==6.3.6
simpleio==3.0.8
adafruit_bitmap_font==1.5.11
adafruit_bus_device==5.2.3
adafruit_display_text==2.22.10
adafruit_dps310==2.1.8
adafruit_funhouse==2.1.18
adafruit_io==5.6.12
adafruit_minimqtt==5.4.2
adafruit_portalbase==1.14.3
adafruit_register==1.9.13

### Additional information

This is just about as minimal as I could get the code and still have the problem.

Any of the following will make it work again

- removing the logger import and code 
- removing the from config import config 
- removing the import adafruit_funhouse from another.py

slootsky, 2022-08-31:
I just tested with 7.3.3 and I get the same result there.

Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit FunHouse with ESP32S2
Board ID:adafruit_funhouse

dhalbert, 2022-08-31:
If you put the `from adafruit_funhouse import FunHouse` into `code.py` instead of in `another.py`, does it work fine in 7.3.x?

This _might_ be a manifestation of #2687, but usually that doesn't cause a crash. The fact that 8.0.0-beta.0 doesn't crash may or may not reflect an 8.0.0-whatever fix.

slootsky, 2022-08-31:
@dhalbert   Yes.  Moving from adafruit_funhouse import FunHouse to code.py makes it go away.

This of course means that I can't use FunHouse in another.py though.

dhalbert, 2023-10-28:
Given that this seems to be another manifestation of #2687, this is fixed in 9.0.0-alpha.2 because long-lived heap was removed.

closed 2023-10-28 by dhalbert