adafruit/circuitpython #6850: audiomp3 only accepts io.FileIO objects, documentation suggests it could take io.BytesIO object as well

stephennelson384, 2022-08-31:
### CircuitPython version

```python
CP 7.3.3
```

### Code/REPL

```python
mp3 = io.BytesIO(my_mp3_as_bytes)
decoder = MP3Decoder(mp3)
```

### Behavior

TypeError: file must be a file opened in byte mode

### Description

I read a small (32kb) mp3 file into memory as bytes, from serial. Currently I must write those bytes to a file on the filesystem and then reopen it, because MP3Decoder only accepts FileIO objects. It would be useful to be able to give a BytesIO object to MP3Decoder, especially since the objects are functionally very similar. 

### Additional information

The source line in question: https://github.com/adafruit/circuitpython/blob/main/shared-bindings/audiomp3/MP3Decoder.c#L96

jepler, 2022-08-31:
The C implementation works only if the MP3 resides on a FAT filesystem such as CIRCUITPY or an SD card, because it makes calls like `f_read` to read data and `f_lseek` to seek.

It would be great if a community member could change the implementation so that it is in fact compatible with an in-memory file. At the same time, maybe it would be possible to implement streaming of mp3s over wifi.

Failing that, a doc fix for the type signature would also be good.

dhalbert, 2026-11-13:
Claude wrote this:

This works now. Since #9284, `MP3Decoder` reads through the generic stream protocol, so any binary stream is accepted. On 11.0.0-alpha.3 (Feather RP2350), `audiomp3.MP3Decoder(io.BytesIO(data))` constructs and reports the sample rate and channel count.

closed 2026-11-13 by dhalbert