adafruit/circuitpython #6851: msgpack not choosing type correctly

polygnomial, 2022-08-31:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-dirty on 2022-05-27; PyCubedv05-MRAM with samd51j19
```

### Code/REPL

```python
import msgpack, io
c = io.BytesIO()
msgpack.pack(145, c)
c.seek(0)
c.getvalue()
```

### Behavior

Output in CircuitPython is `b'\xd1\x00\x91'`, which is an `int16`, but this is encoded on unix as `b'\xcc\x91'`, which is a `uint8`. The value is < 255 and positive, so I think it should be a uint8. Additionally, CircuitPython decodes `b'\xcc\x91'` as -111, which I'm pretty sure is incorrect per the msgpack spec.

### Description

This causes problems for passing msgpack'd data back and forth between the two platforms. I recompiled the firmware with different pin definitions, but I don't think that is the issue.

dhalbert, 2022-08-31:
@ghost -- of interest to you.

dhalbert, 2022-08-31:
@bboser @iot49 of interest to you

polygnomial, 2022-08-31:
I'm poking around in the `msgpack` implementation, and best I can tell it never packs anything as a uint. That doesn't seem to be a huge problem in my use case, but unpacking is a bit of an issue. I am thinking for now I will get around this by on both sides encoding as a negative number (which will cause the packer to encode as a signed int) and then I will flip the sign after unpacking.

iot49, 2022-08-31:
Python does not have native uint's - hence the encoder never generates them in the output. The decoder converts received uint's to int's.

You may consider adding special encoders (e.g. pack_uint) to serve your purpose.

iot49, 2022-08-31:
```Additionally, CircuitPython decodes b'\xcc\x91' as -111, which I'm pretty sure is incorrect per the msgpack spec.```

Yes, that would be a bug. I have no time to fix right now.

Neradoc, 2022-08-31:
> I think adding uints is straightforward

To be more accurate when I say that, a simple (uint*_t) cast works for ints that fit in a CP int (below `0x3FFFFFFF`) but higher numbers require to use long ints,. I'll see if I can look into that.

> Python does not have native uint's - hence the encoder never generates them in the output.

Note that C-Python _does_ use uint for positive ints in msgpack. I don't think we need it in CP because it's more code that does not add functionality, it only makes the packing potentially smaller.

```py
❯ python
Python 3.9.6 (default, Aug 11 2021, 01:29:00) 
>>> import msgpack
>>> msgpack.packb(151)
b'\xcc\x97'
```

closed 2026-11-03 by dhalbert