adafruit/circuitpython #6852: Windows PC tries to boot from CIRCUITPY drive

rsbohn, 2022-09-01:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0-7-g4ae97bec6 on 2022-08-23; Adafruit FunHouse with ESP32S2
```

### Code/REPL

```python
#none
```

### Behavior

computer tries to boot from attached CircuitPython board.

### Description

When I restart my computer it tries to boot from CIRCUITPY. This has happened with CP7.x as well.
I'm running Windows 10 on a Dell Precision 5500.
Usually it eventually boots on the internal disk but this time (after a bios update) it went into some kind of error recovery mode.

dhalbert, 2022-09-01:
This is very likely due to the BIOS/UEFI settings that put booting from a USB drive first in priority. Check your boot drive BIOS settings and reorder them or remove the possibility of a USB drive from the boot drive list.

rsbohn, 2022-09-01:
MulhorandEmperor — Today at 6:16 PM
happens to me on 7.x though not on 8.x
with a qt py esp32-s2 (via Discord)

dhalbert, 2022-10-05:
Closing because it's not a bug. We can explore further in discord or the forums.

closed 2022-10-05 by dhalbert