adafruit/circuitpython #6857: ESP32-C3 microcontroller.on_next_reset

bill88t, 2022-09-02:
### CircuitPython version

```python
Applicable to CircuitPython 8.0.0-beta-0, and main branch builds.
Adafruit CircuitPython 0a7c1ef15 on 2022-09-02; DFRobot Beetle ESP32-C3 with ESP32-C3FN4
```

### Code/REPL

```python
>>> import microcontroller
>>> microcontroller.on_next_reset(microcontroller.RunMode.BOOTLOADER)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Invalid run_mode
```

### Behavior

ValueError

### Description

Doens't work with any of the modes.

dhalbert, 2022-09-02:
This is deliberate. As far as I can see, it's not possible to programmatically enter bootloader-only mode on the ESP32-C3. GPIO9 must be held low, and GPIO8 must be held high. So this run mode is caught as invalid.

bill88t, 2022-09-02:
Supposedly to flash to this board at least (beetle esp32c3) I have to connect GPIO9 with a 10k to ground and hold reset for 3s.
But even when circuitpy is running, I can just run esp-idf and flash it..

MicroDev1, 2022-09-10:
Only `RUNMODE_UF2` isn't supported on `esp32c3`, other modes should work fine. I'll PR a fix for this.

closed 2022-09-10 by dhalbert