adafruit/circuitpython #6862: UnexpectedMaker ProS3 board.I17 should be board.IO17

rsbohn, 2022-09-03:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0-8-gb48141154 on 2022-08-24; ProS3 with ESP32S3
```

### Code/REPL

```python
import board
board.I17
board.IO17
```

### Behavior

'''>>> board.I17
board.NEOPIXEL_POWER
>>> board.IO17
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'IO17'
```

### Description

    { MP_ROM_QSTR(MP_QSTR_LDO2), MP_ROM_PTR(&pin_GPIO17) }, // Second LDO Enable control
    { MP_ROM_QSTR(MP_QSTR_I17), MP_ROM_PTR(&pin_GPIO17) }, // Second LDO Enable control

(Not sure where these came from, Discord perhaps?)

dhalbert, 2022-09-05:
@unexpectedmaker Was the name deliberate?

UnexpectedMaker, 2022-09-05:
> @UnexpectedMaker Was the name deliberate?

Oh, I missed this, sorry. Hmm, seems like a typo. I'll do a PR today to fix it, sorry.

DJDevon3, 2022-09-06:
@UnexpectedMaker looks like the same issue exists on the FeatherS3 for GPIO39
`{ MP_ROM_QSTR(MP_QSTR_I39), MP_ROM_PTR(&pin_GPIO39) }, // Second LDO Enable control`

UnexpectedMaker, 2022-09-07:
Thanks @DJDevon3 :)

Both have been fixed in this PR https://github.com/adafruit/circuitpython/pull/6873

closed 2022-09-07 by dhalbert