adafruit/circuitpython #6866: ESP32-S2 wifi scanning instability upon soft reset

bill88t, 2022-09-05:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0-30-g14498f793 on 2022-09-04; Waveshare ESP32-S2-Pico with ESP32S2

It's a build from adafruit-circuit-python.s3.amazonaws.com
```

### Code/REPL

```python
import wifi
a = wifi.radio.start_scanning_networks()
for i in a: print(str(i.ssid))
wifi.radio.stop_scanning_networks()
del wifi
```

### Behavior

On first boot:

[image]

After exiting to REPL and reloading:

[image]

Meanwhile if I do it in real REPL it works (somewhat, it only sees the network 10cm away from it),
even after many reloads:

[image]

To further test, I made code.py the Code/REPL provided code, and pressed Ctrl-D a few times:

[image]

### Description

Even if the Pi400 network which is 10cm from it, is seen, it can't connect to it.

### Additional information

On first boot and circuitpy 7.x it works fine.
Does not go away if I toggle wifi.radio.enabled.

anecdata, 2022-09-05:
Can you test with `8.0.0-beta.0` to rule out some quirk in that daily build?

With `Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; Adafruit QT Py ESP32S2 with ESP32S2`, this `code.py` runs continously with good, full AP results:
```py
import time
import supervisor
import wifi

def scan_wifi():
    for network in wifi.radio.start_scanning_networks():
        print("{0:02X}:{1:02X}:{2:02X}:{3:02X}:{4:02X}:{5:02X}".format(*network.bssid), end=" ")
        print("{:>2d}".format(network.channel), end=" ")
        print(network.rssi, network.authmode, network.country, network.ssid)
    wifi.radio.stop_scanning_networks()

scan_wifi()
time.sleep(5)
supervisor.reload()
```

bill88t, 2022-09-05:
Repeated the exact same code.py I did above, on beta-0:

[image]

This time it *once more worked*.. but only once..

Tried with your code snippet too, pretty much same results..
It only once registed:
`E4:5F:01:05:51:4E  1 -29 [wifi.AuthMode.WPA, wifi.AuthMode.PSK] XY Pi400`
(I do not mind sharing my pi's mac)

bill88t, 2022-09-05:
I perhaps should mention:
 - There is no .env file for web workflow provided.
 - This board is in a diy airtight box, safe from any tampering.
 - The Pi400 wifi is 10cm away from it.
 - The power is supplied via the Pi400, it is more than adequate.

anecdata, 2022-09-05:
Lack of `.env`, or of any valid Station or AP, should not be a problem. Scanning is independent of Station and AP. I wonder if the box material is attenuating wifi signals, but it's still hard to explain why it would only receive on the first try. Can you test it outside of the box, and also closer to some other APs? Do you have another (perhaps different) device you can try inside the box?

bill88t, 2022-09-05:
Chief, da box plastic.
[image]

Tested it outside the box in the same setup, and it's the same.

Tested on fone, near more ap's:
[image]

Also, those char's are new and annoying.

pepijndevos, 2022-09-06:
I can confirm similar behavior on an ESP32-S3 sitting on my desk right next to the router.

I'm basically just doing 

```python
wifi.radio.connect(secrets["ssid"], secrets["password"])
```

and sometimes after resetting it just errors with

```
ConnectionError: No network with that ssid
```

and indeed scanning at this point returns no networks. A hard reset will bring it back to working normally.

bill88t, 2022-09-08:
Cannot reproduce it on my beetle ESP32-C3..
Did 20 reloads and tried some resets, all fine, no signal reduction.

So, S2 and S3 have the issue but not C3.
I will update the title respectively.

DJDevon3, 2022-09-10:
Cannot reproduce using adafruit feather s2, ran anecdata's snippet for an hour np. Tested during adafruit's Community Help Desk on Saturday morning 9/10/22

KeithTheEE, 2022-09-10:
Running @anecdata's  demo code on an `Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; Adafruit Metro ESP32S2 with ESP32S2` and I can replicate the bug. 

On a hard reset, the first attempt to use the wifi works, it can scan nearby networks and connect to one with it's ssid/password. It can then do standard networking, get and post data, etc.

Softresets thereafter fail to find any wifi network on the scan.  

To do an extra test I erased the flash, reloaded the binary and tried the demo code again and the issue persists. 

Now I've noticed that while most scans fail to find any network, every once in a while it will manage to see one or two networks on a scan. 

After moving the device around, and just watching the few networks it does see, it never notices any network with an rssi below -59 after the soft reload, but on a hard reset it noticed networks all the way down to -88. (These are probably soft signal strength boundaries). There's also enough nearby networks that router distance shouldn't be an issue. But it does apparently contribute. Still, I can set the metro next to the router and frequently will fail to connect so the issue isn't exclusively that. 

In `Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Metro ESP32S2 with ESP32S2` this metro runs the demo without issue, noticing the networks consistently on each soft reboot.

bill88t, 2022-09-10:
Tried again on `8.0.0-beta.0-41-g33a100611` on ESP32-S2:
With some slightly different results:
[image]
This is with 2 soft reloads, same hardware setup, on ljinux since the results don't differ anyways.
It can see the networks that are not **too** far or **too** close..

bill88t, 2022-09-10:
Actually, on first boot, those networks are considered to have, bad reception..
[image]
Something very weird is going on..

DJDevon3, 2022-09-10:
That brings up another point about the RSSI data actually being flawed.  In my test I have 2 AP's sitting right next to each other which usually provide near identical signal according to my wifi manager app on my tablet and phone... but scanning with that code within esp32-s2 showed an unusual amount of deviation between my AP's to the point it gives me reason to question the accuracy.

bill88t, 2022-09-10:
The networks however should still be visible and connectable, regardless of RSSI..

DJDevon3, 2022-09-11:
Oh right, that's an inconsequential issue if it doesn't even work for you, my apologies.

KeithTheEE, 2022-09-11:
I grabbed the nightly from `Adafruit CircuitPython 8.0.0-beta.0-41-g33a100611 on 2022-09-10; Adafruit Metro ESP32S2 with ESP32S2` this morning to check if the bug had been fixed and it hadn't. 

I left the code running while I went about my day and came back to

```py
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 7, in <module>
MemoryError: Failed to allocate Wifi memory

```

hitting `CTRL+D` to reload generates the same error. 

Entering the REPL I ran the following:

```
Adafruit CircuitPython 8.0.0-beta.0-41-g33a100611 on 2022-09-10; Adafruit Metro ESP32S2 with ESP32S2
>>> import gc
>>> gc.mem_free()
2047984
```

And then exiting with `CTRL+D` to reload code.py  the program runs into the Wifi Memory Error again.

DJDevon3, 2022-09-11:
Have you tried manually adding gc.collect in spots?

KeithTheEE, 2022-09-14:
No, I believe gc.collect impacts the memory allotted to the program space that code.py uses, whereas given the error,

```py
MemoryError: Failed to allocate Wifi memory
```
it appears to be during the core's startup and initialization and how it allocates memory for the code managing the wifi components so gc.collect shouldn't have access to that block. But just in case I tried it as well (because it was a great suggestion and even if it didn't solve it, it does help confirm that it's associate with the core's allocation) and put that call at the end just before the sleep+soft reboot  calls and the program still ran out of memory after a period of soft rebooting.

So the useful info I have is that a Metro ESP32-S2 seems to experience the original bug @bill88t had reported, and that another likely attribute of the bug is a memory error after enough soft reboot's take place, and now that gc.collect() does not impact the `Failed to allocate Wifi memory` aspect.

anecdata, 2022-09-14:
@KeithTheEE yes, the memory error is in wifi init, and arises from an esp-idf call. Using the `espidf` module, could track usage of espidf memory <https://docs.circuitpython.org/en/latest/shared-bindings/espidf/index.html>. Could also allocate some PSRAM to esp-idf memory using `CIRCUITPY_RESERVED_PSRAM=` in the `.env` file, but I really don't think that should be necessary if things are working right... just for a scan.

KeithTheEE, 2022-09-15:
Running the same sample code, but adding the line, `print("ESPIDF Heap caps free: ", espidf.heap_caps_get_free_size())` once per program run, I end up with the output: 

```py

ESPIDF Heap caps free:  58808

Code done running.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
ESPIDF Heap caps free:  56688

Code done running.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
ESPIDF Heap caps free:  58636
```

On the first couple of loops.

After running for a long while it fails with the memory error. Right before it fails due to memory error it has far less memory: 

```py 

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
ESPIDF Heap caps free:  4420

Code done running.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
ESPIDF Heap caps free:  4384

Code done running.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 3, in <module>
MemoryError: Failed to allocate Wifi memory

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

The two edge cases--start and end of the program--show changes in memory that are not representative of the typical memory change. 

Most loops show 32 (bytes? free heap caps?) less than the previous call. 

It does vary, sometimes. Usually on the scale of one loop will drop 28, and then a loop or two later it'll drop 36. 

Occasionally (and I noticed this took place around the same times it saw wifi networks) it would jump up by 1000 or so. Sometimes it'd drop back after two or three loops, sometimes it'd slowly drop back down at roughly 32 unit intervals. 

### No `import wifi`
Completely removing the `scan_wifi()` call, and removing the import wifi function, the code loops on a soft reboot and holds steady at 

`ESPIDF Heap caps free:  100076` independent of the number of soft reboot's prior. 

### importing wifi but not calling `scan_wifi()`

This drops the initial output of `espidf.heap_caps_get_free_size()` from the above 100076 to 54844, and then every loop thereafter drops that 54844 by around 32. 

(roughly estimating) Setting the sleep to 1 second from the 5 seconds in the example code, it should take around half an hour to hit a memory error when `import wifi` is present in the code.

anecdata, 2022-09-15:
It appears (to me) that: (1) there is a small memory leak in the CP core or (more likely) in the esp-idf; (2) `supervisor.reload()` does not clear esp-idf memory usage since the exception happens in wifi init (from importing). So `microcontroller.reset()`, hard reset (button), or power cycle is necessary to get back to a clean state. I don't recall this being an issue in older version of esp-idf, perhaps it was introduced with an update.

anecdata, 2022-09-15:
Curiously, on my `Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; Adafruit QT Py ESP32S2 with ESP32S2`, where repeated scans and reloads show a full complement of APs, the esp-idf free memory stays relatively constant at around 50-51kB. Using:
```py
print(f"{gc.mem_free()} {espidf.heap_caps_get_total_size()} {espidf.heap_caps_get_free_size()} {espidf.heap_caps_get_largest_free_block()}")
```
Which brings up additional complicating factors: • differences in network environment; • possibly differences in behavior between different ESP32-S2 chips / modules / antennas.

KeithTheEE, 2022-09-15:
I did not have web workflow enabled either, so that might be a key bit to narrow down the issue

anecdata, 2022-09-16:
Since it works with web workflow enabled (device is auto-connected to an AP), I tried connecting before the scan w/o web workflow, and it gets results the first time then `ConnectionError: No network with that ssid` on subsequent reloads. So it's something more involved than the `connect()` that lets scanning work with web workflow enabled.

bill88t, 2022-09-16:
I will test again on s2 and c3 tonight. This time w/ and wo/ web workflow on both, and plot `espidf.heap_caps_get_free_size()` over reloads

saketvora, 2022-09-16:
Sorry I'm not contributing technically to this, but I'm hitting this exact same issue on my Feather ESP32-S3 (4MB Flash with 2MB PSRAM). Running 8.0.0-beta.0, I see connect() successfully get an IP address on first power up or hard reset, but then return `ConnectionError: No network with that ssid` every subsequent time, even through soft resets. I'm connecting to a 2.4GHz network (ASUS Router XT8). 
Happy to try to pull any logs or try other troubleshooting to help with data collection, just need some pointers as I'm new to this particular platform. Eagerly following this thread though....

bill88t, 2022-09-18:
Made https://github.com/bill88t/CircuitPythonWifiTester to gather more data automatically.
It gathers:
 - How many wifi scans were successful
 - How many wifi scans had significantly reduced results
 - How many scans failed
 - Espidf heap mem stuff before-after

And does the test automatically with and without the .env
I only managed to find time to run it on my S2 though. C3 will have to wait.
The result is clear. The .env makes it work somewhat ok, but as soon as it's gone, scanning immediately breaks.
Here (Data has been beautified to be more readable):
```
{
    "version": "8.0.0-beta.0-41-g33a100611",
    "board": "Waveshare ESP32-S2-Pico",
    "cpu": "ESP32S2",
    "Pass": 47,
    "PartialFail": 22,
    "CompleteFail": 30,
    "espmem_total": 148292,
    "espmem_before":
        [
            59064,
            58752,
            58716,
            58644,
            58620,
            58576,
            58528,
            58512,
            58492,
            58420,
            58412,
            58352,
            58352,
            58312,
            58276,
            58248,
            58216,
            58196,
            58148,
            58120,
            58080,
            58044,
            58040,
            57984,
            57956,
            57920,
            57884,
            57832,
            57816,
            57800,
            57760,
            57712,
            57680,
            57664,
            57624,
            57604,
            57568,
            57548,
            57516,
            57492,
            57456,
            57420,
            57368,
            57336,
            57308,
            57268,
            57248,
            57208,
            57172,
            57144
        ],
    "espmem_after":
        [
            59020,
            58712,
            56700,
            56620,
            56604,
            56552,
            56516,
            56484,
            56468,
            56396,
            56396,
            56332,
            56336,
            56300,
            56260,
            56236,
            56192,
            56172,
            56136,
            56096,
            56068,
            56028,
            56028,
            55972,
            57912,
            57876,
            57856,
            57792,
            57048,
            55776,
            55736,
            55692,
            57636,
            57620,
            57588,
            57560,
            55532,
            55524,
            57480,
            57448,
            57412,
            55396,
            55344,
            55312,
            55288,
            55248,
            55228,
            55184,
            55148,
            55120
        ],
    "espmem_before_env":
        [
            51688,
            51720,
            51716,
            51720,
            51720,
            51720,
            51720,
            51720,
            51720,
            51720,
            51720,
            51720,
            51720,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51332,
            51332,
            51332,
            51332
        ],
    "espmem_after_env": 
        [
            50756,
            51716,
            51720,
            51720,
            51720,
            51720,
            51720,
            51720,
            51720,
            51720,
            51720,
            51720,
            51720,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51516,
            51332,
            51332,
            51332,
            51332,
            51332
        ]
}
```
I will provide:
 - more info on how it works
 - plug the data onto excel
 - run the same test on c3 and latest builds
in the next few days.

bill88t, 2022-09-19:
Did some changes to better seperate the data:
```
{
    "version": "8.0.0-beta.0-41-g33a100611",
    "board": "Waveshare ESP32-S2-Pico",
    "cpu": "ESP32S2",
    "iterations": 101,
    "want_env": false,
    "og_networks": 14,
    "espmem_total": 148292,
    "Pass_env": 44,
    "Pass": 1,
    "PartialFail_env": 5,
    "PartialFail": 16,
    "CompleteFail_env": 0,
    "CompleteFail": 33,
    "espmem_before_env": [51852, 51640, 51684, 51696, 51700, 51700, 51700, 51696, 51700, 51696, 51700, 51696, 51700, 51696, 51700, 51696, 51700, 51696, 51700, 51696, 51700, 51696, 51700, 51696, 51700, 51696, 51700, 51696, 51692, 51700, 51692, 51700, 51692, 51700, 51692, 51700, 51692, 51700, 51692, 51700, 51700, 51700, 51700, 51700, 51696, 51700, 51696, 51700, 51696, 51700],
    "espmem_after_env": [50328, 51404, 51696, 51696, 51700, 51700, 51700, 51700, 51700, 51700, 51700, 51700, 51700, 51700, 51696, 51700, 51696, 51700, 51696, 51696, 51700, 51696, 51700, 51696, 51700, 51696, 51700, 51692, 51700, 51692, 51700, 51692, 51700, 51692, 51700, 51692, 51700, 51692, 51700, 51700, 51700, 51700, 51700, 51700, 51700, 51700, 51700, 51696, 51700, 51700],
    "espmem_before": [59064, 58752, 58716, 58644, 58640, 58588, 58564, 58532, 58512, 58476, 58432, 58424, 58368, 58348, 58340, 58300, 58260, 58220, 58180, 58156, 58120, 58076, 58036, 58008, 57968, 57944, 57900, 57872, 57836, 57816, 57768, 57744, 57692, 57664, 57648, 57612, 57604, 57576, 57552, 57516, 57476, 57440, 57404, 57384, 57352, 57296, 57260, 57236, 57192, 57172],
    "espmem_after": [59020, 58712, 56700, 56800, 56624, 56564, 56536, 56512, 56488, 56464, 56408, 56404, 56344, 56332, 56316, 56276, 56240, 56196, 56152, 56132, 56096, 56064, 56016, 55988, 57940, 57908, 55896, 55856, 55824, 57768, 55752, 56632, 55680, 55652, 57612, 55596, 57568, 55560, 55536, 55500, 55464, 55424, 55388, 55356, 55336, 55280, 55244, 55220, 55176, 55156]
}
```
So you can see:
 - With the .env:
   - The check passed 44/50.
   - Partially failed (more than 5 networks were lost) 5/50.
   - Did not completely fail at all (all networks lost).
 - Without the .env:
   - The check passed 1/50 (the first time).
   - Partially failed 16/50.
   - Completely failed 33/50.

bill88t, 2022-09-19:
And on C3 I see similar results:
```
{
    "version": "8.0.0-beta.0-47-gae64f9fd7",
    "board": "DFRobot Beetle ESP32-C3",
    "cpu": "ESP32-C3FN4",
    "want_env": false,
    "iterations": 101,
    "og_networks": 13,
    "espmem_total": 262736,
    "Pass_env": 46,
    "Pass": 2,
    "PartialFail_env": 3,
    "PartialFail": 3,
    "CompleteFail_env": 0,
    "CompleteFail": 45,
    "espmem_before_env": [67556, 67780, 67568, 67572, 67568, 67572, 67568, 67572, 67572, 67572, 67572, 67572, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67552, 67548, 67560, 67548, 67560, 67548, 67560, 67548, 67560, 67548, 67560, 67548, 67560, 67548, 67560, 67548, 67560, 67548],
    "espmem_after_env": [66928, 67772, 67572, 67572, 67572, 67568, 67548, 67572, 67548, 67572, 67544, 67548, 67556, 67556, 67556, 67556, 67556, 67556, 67532, 67556, 67556, 67556, 67556, 67556, 67556, 67556, 67532, 67556, 67556, 67560, 67532, 67528, 67548, 67560, 67548, 67560, 67524, 67536, 67548, 67560, 67548, 67560, 67548, 67560, 67548, 67560, 67548, 67536, 67524, 67560],
    "espmem_before": [75112, 74788, 74764, 74696, 74684, 74588, 74584, 74532, 74544, 74488, 74464, 74428, 74388, 74360, 74304, 74296, 74240, 74208, 74184, 74164, 74136, 74092, 74052, 74032, 73988, 73956, 73908, 73888, 73864, 73816, 73784, 73724, 73712, 73680, 73668, 73640, 73608, 73572, 73536, 73472, 73448, 73412, 73404, 73360, 73340, 73320, 73276, 73244, 73208, 73176],
    "espmem_after": [75068, 74752, 74720, 74664, 72660, 74556, 74012, 73596, 73604, 72472, 74420, 73848, 73268, 73056, 74272, 74272, 74208, 72924, 74148, 72148, 74096, 74060, 72028, 73988, 73952, 73912, 73872, 73852, 73828, 73780, 73748, 73688, 71696, 73644, 73632, 73604, 73572, 73536, 73500, 73436, 73408, 73372, 73356, 73324, 73300, 73288, 73244, 73208, 73172, 73140]
}
```
So it is affected afterall, but since i was using the .env (cuz duh no usb) I didn't experience it.
Now imma test again with latest build

bill88t, 2022-09-19:
C3 latest:
```
{
    "version": "8.0.0-beta.0-61-g5593d23a0",
    "board": "DFRobot Beetle ESP32-C3",
    "cpu": "ESP32-C3FN4",
    "want_env": false,
    "og_networks": 13,
    "iterations": 101,
    "Pass_env": 48,
    "Pass": 1,
    "PartialFail_env": 1,
    "PartialFail": 0,
    "CompleteFail_env": 0,
    "CompleteFail": 49,
    "espmem_total": 262848,
    "espmem_before_env": [67652, 67888, 67672, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684],
    "espmem_after_env": [67056, 67880, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676, 67684, 67676],
    "espmem_before": [75224, 74900, 74876, 74808, 74784, 74748, 74740, 74696, 74644, 74632, 74584, 74540, 74512, 74468, 74468, 74424, 74360, 74352, 74336, 74304, 74268, 74236, 74204, 74160, 74144, 74096, 74072, 74016, 74004, 73960, 73908, 73892, 73852, 73820, 73784, 73768, 73740, 73704, 73668, 73656, 73604, 73588, 73540, 73512, 73464, 73452, 73404, 73396, 73360, 73324],
    "espmem_after": [75180, 74864, 74832, 74772, 74740, 74700, 74160, 73396, 74608, 72616, 72568, 74504, 74476, 74432, 74432, 74388, 74324, 74316, 74300, 74268, 74232, 74200, 74168, 74124, 74100, 74060, 74032, 73980, 73968, 73924, 73872, 73856, 73816, 73784, 73748, 73732, 73704, 73668, 73632, 73620, 73568, 73548, 73504, 73464, 73428, 73412, 73368, 73360, 73324, 73288]
}
```
Same pretty much.

bill88t, 2022-09-19:
S2 latest:
```
{
    "version": "8.0.0-beta.0-61-g5593d23a0",
    "board": "Waveshare ESP32-S2-Pico",
    "cpu": "ESP32S2",
    "want_env": false,
    "iterations": 101,
    "og_networks": 14,
    "Pass_env": 47,
    "Pass": 1,
    "PartialFail_env": 2,
    "PartialFail": 17,
    "CompleteFail_env": 0,
    "CompleteFail": 32,
    "espmem_total": 148292,
    "espmem_before_env": [51852, 51648, 51708, 51712, 51708, 51712, 51708, 51712, 51504, 51508, 51504, 51508, 51512, 51504, 51512, 51512, 51512, 51512, 51512, 51504, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51504, 51512, 51512, 51512, 51504, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51328, 51328],
    "espmem_after_env": [50340, 51408, 51712, 51708, 51712, 51708, 51712, 51708, 51508, 51504, 51508, 51512, 51504, 51512, 51512, 51512, 51512, 51512, 51504, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51504, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51512, 51328, 51328, 51328],
    "espmem_before": [59064, 58752, 58716, 58680, 58652, 58628, 58592, 58564, 58504, 58468, 58420, 58412, 58368, 58328, 58296, 58264, 58232, 58196, 58164, 58116, 58096, 58064, 58032, 58000, 57968, 57936, 57904, 57872, 57840, 57804, 57768, 57744, 57704, 57668, 57640, 57604, 57588, 57572, 57536, 57504, 57468, 57436, 57400, 57352, 57344, 57300, 57264, 57244, 57220, 57188],
    "espmem_after": [59020, 56732, 56700, 56664, 56636, 56612, 56576, 56548, 56488, 56448, 56392, 56392, 56352, 56312, 56280, 56248, 56216, 56180, 56148, 56100, 56080, 56048, 56016, 55984, 55952, 55920, 57320, 55856, 55824, 55788, 55752, 55728, 55688, 55832, 57604, 57568, 57552, 55556, 55520, 55488, 55452, 55420, 55380, 55324, 55324, 55284, 55248, 57208, 55204, 55172]
}
```
Similar results here too..

bablokb, 2022-09-20:
I am using a Magtag (with an esp32-s2) and the code that connects to my router with 7.3.3 will not connect with 8.0.0-beta. No network with this ssid.

When I look at some recent commits, I see some tinkering with the resetting (or not resetting) of wifi. Maybe this  is the cause.

saketvora, 2022-09-21:
@bill88t Would it be helpful if I ran your WiFiTester script for the extra data? I tried running it on my Feather ESP32-S3 4MB Flash / 2MB PSRAM but I didn't know how to handle the .env part (forgive my ignorance on this).
Happy to try and help, just need a pointer or two.

This bug is totally stalling my project on the Feather ESP32-S3. :/ I don't seem to have the option for this platform to downgrade to CircuitPython 7.3.3 either. I've ordered a Feather ESP32-S2 to see if I can reproduce this issue with my router using both 7.3.3 and 8.0.0 on the S2 platform.

bill88t, 2022-09-21:
> Would it be helpful if I ran your WiFiTester script for the extra data? I tried running it on my Feather ESP32-S3 4MB

@saketvora It would be useful to get a reading from ESP32-S3 as to confirm it's behaviour.
For the .env please take a look [here](https://docs.circuitpython.org/en/latest/docs/workflows.html#web)
If the board has no CIRCUITPY usb access, you can from within the repl do something like:
```
>>> from storage import remount
>>> remount("/", False)
>>> a = open("/.env", "w")
>>> a.write("CIRCUITPY_WIFI_SSID='amogus'\n")
>>> a.write("CIRCUITPY_WIFI_PASSWORD='sussy'\n")
>>> a.write("CIRCUITPY_WEB_API_PASSWORD='baka'\n")
>>> a.write("CIRCUITPY_WEB_API_PORT=80")
>>> a.close()
>>> remount("/", True)
>>> import microcontroller
>>> microcontroller.reset()
```

bill88t, 2022-09-21:
This issue most certainly though affects all esp boards. Someone should update the tags too.

bill88t, 2022-09-21:
@saketvora 
> This bug is totally stalling my project on the Feather ESP32-S3.

A temporary workaround would be to simply force the board to always .reset() upon code completion as this bug occurs only on soft reloads.

> I don't seem to have the option for this platform to downgrade to CircuitPython 7.3.3 either.

You can try to build it if you'd like. Not gonna claim it will work, but you can try. [Building](https://learn.adafruit.com/building-circuitpython/build-circuitpython)
(make sure you clone the 7.3.x branch)

saketvora, 2022-09-22:
Thank you @bill88t 
Board: Adafruit ESP32-S3 4MB Flash 2MB PSRAM
Adafruit CircuitPython 8.0.0-beta.0-57-g32d8dd425 on 2022-09-16
WiFi network: 2.4GHz only, WPA2, Asus ZenWifi XT8 (mesh).
Location: 40ft from router through walls (i.e.., not close by or with line of sight)

Sorry, I just downloaded the latest WifiTester code from the GitHub link today but the results.json doesn't seem as well formatted for me. Here's the raw data:

`{"version": "8.0.0-beta.0-62-gb6f67be3e", "espmem_after_env": [216700, 215508, 215508, 215508, 215508, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612], "espmem_before": [224164, 223832, 223796, 223776, 223736, 223688, 223668, 223632, 223588, 223576, 223524, 223488, 223452, 223444, 223416, 223380, 223352, 223308, 223280, 223260, 223196, 223160, 223148, 223112, 223092, 223060, 223020, 222976, 222932, 222916, 222884, 222852, 222824, 222768, 222744, 222716, 222652, 222616, 222624, 222596, 222568, 222544, 222480, 222468, 222432, 222408, 222372, 222336, 222300, 222272], "PartialFail_env": 1, "CompleteFail": 0, "want_env": false, "CompleteFail_env": 0, "iterations": 101, "board": "Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM", "og_networks": 15, "cpu": "ESP32S3", "espmem_total": 324940, "Pass": 39, "espmem_after": [224116, 223796, 223760, 223740, 223692, 223652, 223632, 223596, 223552, 223544, 223492, 223452, 223416, 223408, 223380, 223344, 223316, 223280, 223244, 223216, 223160, 223128, 223112, 223076, 223056, 223024, 222988, 222932, 222896, 222884, 222848, 222808, 222792, 222736, 222708, 222680, 222616, 222580, 222588, 222560, 222536, 222508, 222444, 222436, 222400, 222372, 222336, 222292, 222272, 222244], "Pass_env": 48, "espmem_before_env": [216280, 216512, 215508, 215504, 215508, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216796, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612, 216612], "PartialFail": 11}`

bill88t, 2022-09-22:
@saketvora
The data is exported as a simple json by the wifi tester, I manually formatted them to be more readable.

Wait. `"CompleteFail": 0` and `PartialFail": 11`???
So ESP32-S3 behaves differently..
The fact that `CompleteFail` is 0 means that at none of the 50 .env-less runs did it go blind.
So the bug's different there?

> This bug is totally stalling my project on the Feather ESP32-S3.

From these results it means it can see at least *some* wifi's most of the times.
(PartialFails's on my runs were it seeing only wifi's VERY near it)
And in most cases it scanned just fine.

So *slightly* reduced range?
Can you not connect to them or something?

And since these results are vastly different from mine, could you please run it again with:
```
# configure how many runs
env_runs = 200
non_env_runs = 200
# each must be >0
```
set at the top of the script?

This will make it do many more runs.
(200 with env, 200 without env)

Note: To set the test to run again:
1. Open the REPL and stay at the `>>>` prompt
    (this is so the board does not autoreload during the file swaps)
    (If you do not, it may bootloop trying to remount rw, you can recover by entering [safe mode](https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#circuitpython-7-dot-0-0-and-later-2978455))
3. rename boot_disabled.py back to boot.py
4. remove the current results file
5. eject the board
6. Ctrl+D

charlien, 2022-09-24:
I have been playing with an esp32 proS3 and wifi has been almost completely unusable BUT only when I have the board connected to my desktop via the usb-c cable. At first I thought it was because I was screened into the board's terminal but it didn't matter in the end.

Using the same code, without change, it connects to WiFi almost 100% of the time after reboots. I have an ssd1306 connected which display the wifi status, which is how I confirm its connection (ip address) in addition I keep a ping active on another computer to the board.

UnexpectedMaker, 2022-09-30:
> 

What channel is it connecting to? Please try it again having it connect on a channel 9 or above.

charlien, 2022-09-30:
> > 
> 
> @charlien What channel is the ProS3 connecting to? Please try it again having it connect on a channel 9 or above.

I figured it out. At first I did attempt channel 9, but that didn't work so I had it iterate through all channels until it could find an address but that failed as well. 

The problem comes back to the same reason why I have to remote my logitech 2.4GHz mouse. USB interference (ref: https://www.usb.org/sites/default/files/327216.pdf). I should've known since the WiFi module only supports 2.4GHz. Thanks for checking in though!

This also solved the ability for it to scan other networks as well, for the same obvious reason.

UnexpectedMaker, 2022-09-30:
> > @charlien What channel is the ProS3 connecting to? Please try it again having it connect on a channel 9 or above.
> 
> I figured it out. At first I did attempt channel 9, but that didn't work so I had it iterate through all channels until it could find an address but that failed as well.
> 
> The problem comes back to the same reason why I have to remote my logitech 2.4GHz mouse receiver. USB interference (ref: https://www.usb.org/sites/default/files/327216.pdf). I should've known since the WiFi module only supports 2.4GHz. Thanks for checking in though @UnexpectedMaker, enjoying the small footprint board so far!
> 
> This also solved the ability for it to scan other networks as well, for the same reason.

Oh fantastic that you found the cause - bummer about what it is though!

askpatrickw, 2022-09-30:
This explains a lot of weird WIFI stuff on my workbench where I have a USB3.0 hub that I plug my boards into... 🤦‍♂️

anecdata, 2022-10-04:
Testing on:
`Adafruit CircuitPython 8.0.0-beta.1 on 2022-10-01; Adafruit QT Py ESP32S2 with ESP32S2`
`Adafruit CircuitPython 8.0.0-beta.1 on 2022-10-01; Adafruit QT Py ESP32S2 with ESP32S2`

Turning web workflow on makes the scanning (and connection) more reliable. The following code, executed before a wifi scan or wifi connect, also fixes the exceptions for me:
```py
wifi.radio.enabled = False
wifi.radio.enabled = True
```
It seems that something in the non-web-workflow path in CP 8.x isn't fully init-ing or de-init-ing wifi.

DJDevon3, 2022-10-05:
From my experience I agree with your findings. Scan only detected strong signals and only ones that were on the same channel. I had my AP's set on autoscan and the time they both showed up they were both on channel 6. All the other 10 or so weak SSID's that normally populate from my neighbors were absent.

anecdata, 2022-10-05:
Same caveat as my comment on issue [6791](https://github.com/adafruit/circuitpython/issues/6791#issuecomment-1267517793): the `wifi.radio.enabled` sequence isn't repeatable. I tested it somewhat extensively yesterday afternoon, but wasn't able to replicate it later. @KeithTheEE reported a similar [inconsistency](https://discord.com/channels/327254708534116352/327298996332658690/1027011198916501534). Some odd fluke that made it work for a while, but not again, so quite puzzling.

ghkites, 2022-10-05:
Similar observations here with an ESP32-S3 (4MB Flash, 2MB PSRAM) on 8.0.0 beta 1.  I switched my WiFi to channel 1 for the time being, and that has helped -- it was on 11 and I couldn't even always get a connection after a hard-reset (only 70% success rate, give or take), and even if it did connect, it never reconnected after a soft-reset.  With the lower channel, it seems consistently able to get a connection after a hard-reset. 

With web workflow disabled, initial connection is consistent, but WiFi scans get almost no results after a soft-reset - only strong signals, and only those on lower channels (I mostly only see channel 1).  If my WiFi is on channel 11, it's completely unable to connect after a soft-reset, even with the AP being 6 feet away.  Only channel 1, it'll usually reconnect after a soft-reset.

With web workflow enabled, it consistently connects and WiFi scans consistently return results even after a soft-reset (including some APs on channels 6 and 11).

bill88t, 2022-10-05:
`wifi.radio.enabled` is not a viable workaround even with sleep involved.
Feels like a good luck charm at best.

KeithTheEE, 2022-10-05:
(This'll cover most of the details I outline in the linked discord message that way the information is here in one place, and will have some added information that I hadn't finished putting together until now.)

Implementing `wifi.radio.enable` toggling worked for me consistently through multiple soft and hard resets, and multiple changes to the code, until I changed the version of circuit python on the board, after I did that I could not get back to making the toggle work even after returning to the original version of circuit python I had been using when it worked.

### While I had the `wifi.radio.enable` toggle working, 

Using an `Adafruit Metro ESP32-S2` I was on the nightly from `Adafruit CircuitPython 8.0.0-beta.0-41-g33a100611 on 2022-09-10`, tested the wifi radio enable toggle and it worked. Web workflow is not enabled on my device. 

I reliably got the same number of wifi access points as I did from a hard reset. There were some ap's that varied but they were the ones with a weak signal and they were inconsistently in the hard reset scan so it just seemed like it was consistently difficult to see signal. I tried to see if it saw different networks after a hard or soft reset, and just started unplugging and plugging the board in over and over, and looking at and comparing the networks it saw with what it saw after letting it run for a while with soft resets. It reliably saw the same networks regardless of hard or soft resets for me. There were a few networks it wouldn't reliably see, but this was true both on hard and soft resets. They were weak access points so this isn't much of a surprise. The hard resets would see those networks with the same reliability/variability.  

The wifi channels I'd see on soft resets after successfully toggling were not only channel 1. I reliably got channels 4, 5, and 11 (they are near enough and had a rssi that was usually > -70 but did have some variability). They made up around half of the access points the ESP32S2 saw. 

While the toggle did make the ESP32S2 see the correct number of access points, after running the script through enough soft resets the board still crashed due to the [same memory error](https://github.com/adafruit/circuitpython/issues/6866#issuecomment-1248649579) as before: the espidf use of memory and allocation fails. It still lost around 32 bytes of space after each soft reload. So while the toggle worked it didn't address the memory leak.

### Breaking the `wifi.radio.enable` toggle and trying to get it back

But then I wanted to make sure the 8 beta1 behaved the same, so I upgraded to that, and couldn't get the scan working off of soft resets even with `wifi.radio.enable` toggled between false and true. So then I erased flash and tried the beta, and toggling the wifi and it didn't work. Then I erased flash and went back to the sept10 nightly and the toggle was not working any more. 
So I can't duplicate what I was doing before with the same code on the same version of circuit python. 

Then on the off chance that there was some 7.3 variable that made the toggle work, (but really I was grasping at straws) I erased flash, installed that, then without erasing flash I tried the sept10 nightly. That didn't work. Trying the same approach of erasing flash, installing 7.3, I installed the 8 beta 1 without erasing flash and tried that. It didn't work either. 

So while the toggle bodge worked briefly, I can't make it work again and I don't know how to coax the platform into a state where it works again. Even while it was working the memory leak still persisted.

dhalbert, 2022-10-06:
I have now had time to test with a Metro ESP32-S2, using variants of @anecdata's test program:

```py
import time
import microcontroller
import supervisor
import wifi

def scan_wifi():
    for network in wifi.radio.start_scanning_networks():
        print("{0:02X}:{1:02X}:{2:02X}:{3:02X}:{4:02X}:{5:02X}".format(*network.bssid), end=" ")
        print("{:>2d}".format(network.channel), end=" ")
        print(network.rssi, network.authmode, network.country, network.ssid)
    wifi.radio.stop_scanning_networks()

while True:  # change to an if to do a reload or reset
    time.sleep(5)
    #wifi.radio.enabled = False
    #wifi.radio.enabled = True
    scan_wifi()
    time.sleep(5)
supervisor.reload()
#microcontroller.reset()
```
As others have seen, doing a full reset each time around shows all the available AP's. Or just repeatedly scanning every few seconds does the same, without a reload or reset. Changing to `supervisor.reload()` causes it to show the full complement the first time, and then none or occasionally one on subsequent passes, so there's some odd state thing going on. The `.enabled` toggle makes no difference.

I am not going to look at the ESP-IDF internal heap consumption until this is debugged, since I think fixing this may fix the storage leak.

DJDevon3, 2022-10-07:
Speaking of toggling states. I reported a bug on the rfm9x boards that might actually be related to this S2 bug. I'm using the lora board with an S2. https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/issues/81  It's possible since I'm using the lora featherwing that it's actually catching some type of transceiver toggle state happening. That is to say, there might be a specific pin for the RECEIVER part of the transceiver module inadvertently being toggled by the S2 bug. 

It's just an idea, I have no evidence to back up this hypothesis yet. Requires investigation of physical pinouts of the both the S2 chip and lora featherwing (which I will take on today). If I'm correct when the S2 hard reset bug happens it's knocking out the receiver of the lora featherwing at the same time. At least in the case of the lora featherwing the issue might be able to be traced to a physical pin state toggle on the S2 bare chip. 

If we fix the S2 hard reset bug I have a feeling it's instantly going to fix about 10-20 bug reports.

dhalbert, 2022-10-07:
(answered in https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/issues/81)
On Espressif chips, most pins are reset to high with a weak pullup when CircuitPython stops (or soft-restarts). So i I have not read that whole issue carefully, but that might be causing that particular problem.

DJDevon3, 2022-10-07:
Think I figured out the LORA issue. It's completely unrelated to the S2 bug. Sorry about that.

PR #7023 (2022-10-10): update esp-idf; allow start/stop channels in wifi scanning
- Fixes #6866.

---

- The snapshot of ESP-IDF we were using was out of date. Updating it fixed the wifi scanning issues in #6866. Added yet another new branch to  https://github.com/adafruit/esp-idf : https://github.com/adafruit/esp-idf/tree/release/v4.4-circuitpython. That branch is based on the current tip of https://github.com/espressif/esp-idf/tree/release/v4.4. Cherry-picked fixes from the  https://github.com/adafruit/esp-idf/tree/circuitpython8 branch, which we were using previously. Some fixes from `circuitpython8` were already [... cut]
closed 2022-10-10 by dhalbert