adafruit/circuitpython #6869: Web Workflow: network mDNS hostname often differs from local-device mDNS hostname (links & redirects don't work)

anecdata, 2022-09-06:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; Adafruit QT Py ESP32S2 with ESP32S2
```

### Code/REPL

```python
# From another device on the network, augmented names are found:
m = mdns.Server(wifi.radio)
for service in m.find(service_type="_circuitpython", protocol="_tcp", timeout=5):
    print(f"{all_the_things}")

"""
Adafruit QT Py ESP32S2-5
cpy-f66961-281.local
192.168.6.180:80
"""

# This does not work:
# curl -v -L http://cpy-f66961.local/cp/devices.json

# These work:
# curl -v -L http://192.168.6.180/cp/devices.json
# curl -v -L http://cpy-f66961-281/cp/devices.json
```

### Behavior

From the network perspective, CircuitPython mDNS hostnames often have a hyphen and integer appended to the hostname. CircuitPython mDNS `server.find` [code](https://gist.github.com/anecdata/06045da4eb109e7467d9e8068a8dffad) will expose these augmented hostnames (and instance names), and browsers and curl will find the devices using these augmented names.

However, the device itself [currently] does not recognize itself with these augmented names. The Web API will report something like:
```
Hostname:
[cpy-f66961](http://cpy-f66961.local/)
IP:
[192.168.6.180](http://192.168.6.180/)
```
and clicking the hostname link on the Web API will fail.

Redirects also fail since the device doesn't answer to its nominal hostname:
```
% curl -v -L http://circuitpython.local/cp/devices.json
*   Trying 192.168.6.180:80...
* Connected to circuitpython.local (192.168.6.180) port 80 (#0)
> GET /cp/devices.json HTTP/1.1
> Host: circuitpython.local
> User-Agent: curl/7.79.1
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 307 Temporary Redirect
< Connection: close
< Content-Length: 0
< Location: http://cpy-f66961.ocal/cp/devices.json
< Access-Control-Allow-Credentials: true
< Vary: Origin, Accept, Upgrade
< Access-Control-Allow-Origin: 
< 
* Closing connection 0
* Issue another request to this URL: 'http://cpy-f66961.local/cp/devices.json'
* Could not resolve host: cpy-f66961.local
* Closing connection 1
curl: (6) Could not resolve host: cpy-f66961.local
```
or simply:
```
% curl -v -L http://cpy-f66961.local/cp/devices.json 
* Could not resolve host: cpy-70e2ce.local
* Closing connection 0
curl: (6) Could not resolve host: cpy-f66961.local
```

### Additional information

Since the device knows its own IPv4 address, perhaps one solution could be to redirect to the IPv4 address instead of to the network-invalid mDNS local-device hostname.

makermelissa, 2022-09-09:
Ok, this sounds like it's not yet properly handling mDNS Hostname conflict naming. While in theory, every device should have a unique name because of using the MAC address, this isn't always the case such as with an improperly setup network where multiple mDNS resolvers keep trying to resolve the same host.

This is good info because there may need to be some updates to code.circuitpython.org to deal with this there too.

tannewt, 2022-11-29:
@anecdata How many devices do you have on the network? Are you relying on web workflow to start wifi or are you starting it everytime yourself in code.py?

anecdata, 2022-11-29:
Typically less than half a dozen CircuitPython devices with web workflow enabled. I don't have any collisions in the lower three octets of the MAC.

Currently on devices where web workflow is enabled, web workflow starts up wifi.

tannewt, 2022-11-30:
@anecdata Could you please give me more details of your network setup and what code is running on each device and what device it is? I was able to get the IDF to mangle the instance name but not the hostname.

anecdata, 2022-11-30:
Not sure what all is useful to you... Multiple industrial-strength APs, all with the same SSID(s). Relatively complex home network wired / wifi, but robust. A lot of devices on the network, a bunch of Apple devices, some Raspberry Pi, myriad other things. There are a lot of CircuitPython devices, though only a handful running web workflow (and none anymore doing non-web-workflow mDNS).

Typically I'm using Safari to interact with the web workflow pages, could be something to do with Mac cacheing. But accessing from Chromium on a Raspberry Pi also requires the numeric extension to the mDNS hostname (note that no devices ever show up in the `Here are other CircuitPython devices on your network:` list on that platform).

My assumption has been that mDNS is really just between the device advertising its hostname and instance name on an IPv4:port, and the device trying to access it (no central name server).

The web workflow devices are all ESP32-S2 (QT Py), now running 8.0.0-beta.3 or -beta.4. Four of them are running rudimentary adafruit_httpserver just for testing. Another is running a typical web client. Another is just chilling in the REPL. All are in Station mode, no APs.

Typical web workflow display as follows:
```
Here are other CircuitPython devices on your network:

[Adafruit QT Py ESP32S2-4 (cpy-d96468-22)](http://192.168.6.165/)
[Adafruit QT Py ESP32S2-6 (cpy-706b80)](http://192.168.6.197/)
[Adafruit QT Py ESP32S2-5 (cpy-a58fbc-83)](http://192.168.6.179/)
[Adafruit QT Py ESP32S2-2 (cpy-70e2de-12)](http://192.168.6.196/)
```
The `Here are other CircuitPython devices on your network:` rarely if ever finds all of the devices, typically 0-3.

A `[cpy-aaeb26](http://cpy-aaeb36.local/)` hostname on one device's web workflow page shows up to other devices' web workflow pages as `[Adafruit QT Py ESP32S2 (cpy-aaeb36-666)](http://192.168.6.194/)`.

Neradoc has also seen this behavior, various discussions on Discord.

tannewt, 2022-12-01:
Thank you both! I'll try again to replicate it today.

The collision should only happen if two devices share a hostname (which they shouldn't.) Instance names can have this happen too but it isn't critical for the API to work.

tannewt, 2022-12-01:
I think this change in the IDF will fix it: https://github.com/adafruit/esp-idf/pull/9

Thanks to @anecdata for finding that it is caused by a collision of `circuitpython.local` causing the mac based hostname to be mangled.

anecdata, 2024-09-20:
This issue of spontaneously-incrementing mDNS hostnames  (e.g., "blahblah-4") appears to be back in CP 9. CP 8.2.10 seems fine. @donutcat spotted it, and there's been discussion on Discord last evening and prior in #help-with-circuitpython.

`Adafruit CircuitPython 9.1.4 on 2024-09-17; Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3`

No `code.py` file.

`settings.toml`:
```py
CIRCUITPY_WIFI_SSID="Redacted"
CIRCUITPY_WIFI_PASSWORD="Redacted"
CIRCUITPY_WEB_API_PASSWORD="passw0rd"
```
(and a similar file on another S3)

Device 34, after reset:
```
% curl -iL http://cpy-****34.local
HTTP/1.1 200 OK
Content-Encoding: gzip
Content-Length: 644
Content-Type: text/html
```

Device 84, after reset:
```
% curl -iL http://cpy-****84.local
HTTP/1.1 200 OK
Content-Encoding: gzip
Content-Length: 644
Content-Type: text/html
```

Try `circuitpython.local`:
```
% curl -iL http://circuitpython.local
HTTP/1.1 307 Temporary Redirect
Connection: close
Content-Length: 0
Location: http://cpy-****34.local/
Access-Control-Allow-Credentials: true
Vary: Origin, Accept, Upgrade
Access-Control-Allow-Origin: *

HTTP/1.1 200 OK
Content-Encoding: gzip
Content-Length: 644
Content-Type: text/html
```

First time is fine, but after redirecting from `circuitpython.local`, the hostname has changed (though the redirect to the old name may still be cached... somewhere):
```
% curl -iL http://circuitpython.local
HTTP/1.1 307 Temporary Redirect
Connection: close
Content-Length: 0
Location: http://cpy-****34.local/
Access-Control-Allow-Credentials: true
Vary: Origin, Accept, Upgrade
Access-Control-Allow-Origin: *

curl: (6) Could not resolve host: cpy-****34.local
```
It's incremented occasionally even without visiting `http://circuitpython.local`.

Success to the incremented hostname:
```
% curl -iL http://cpy-****34-2.local
HTTP/1.1 200 OK
Content-Encoding: gzip
Content-Length: 644
Content-Type: text/html
```

[image]
[image]

Things get stranger if the hostname is changed in `code.py`, but that could be a second-order effect. Instance name also gets incremented.

closed 2024-10-10 by dhalbert

dhalbert, 2024-10-10:
`make latexpdf` worked with Sphinx 7.4.7. Then I updated did `pip install --upgrade -r requirements-doc.txt`, and Sphinx got upgraded to 8.1.0. Then `make latexpdf` failed. Checking further.

PR #7295 (2022-12-02): Fix MDNS hostname mangling
cpy-MAC hostnames were being mangled on circuitpython.local conflicts.

Fixes #6869
commit ded0ee48 (2022-12-02): Fix MDNS hostname mangling

cpy-MAC hostnames were being mangled on circuitpython.local
conflicts.

Fixes #6869
PR #9699 (2024-10-10): MDNS hostname match DHCP. Fix collision mangling
MDNS defaults to the hostname used by DHCP/LWIP since it is now unique. This makes it the same either way.

This also updates esp-protocols (used for mdns) with a patch to ensure that mangled names are the ones that collide. (circuitpython.local collides but was causing cpy- to be mangled.)

Fixes #6869 again.

@retiredwizard this changes the hostname slightly because it hit an assert when over 32 characters. cpy, board and mac are now dash separated too. That way the board's _ don't confusing things.

@anecdata Please test this for the MDNS col [... cut]
commit ed0e640f (2024-10-10): MDNS hostname match DHCP. Fix collision mangling

MDNS defaults to the hostname used by DHCP/LWIP since it is now
unique. This makes it the same either way.

This also updates esp-protocols (used for mdns) with a patch to
ensure that mangled names are the ones that collide.
(circuitpython.local collides but was causing cpy- to be mangled.)

Fixes #6869 again.