adafruit/circuitpython #6872: displayio.EPaperDisplay: support fast updates

pepijndevos, 2022-09-06:
PervasiveDisplay panels support a fast update mode, which requires that the first frame contains the old display data, and the second frame contains the new display data: https://docs.pervasivedisplays.com/epd-usage/epd-driving-sequence/small-epds/fast-update/ssf-upload-image-data/dtm1-and-dtm2-registers

I'm not sure if this is an easy change or a major overhaul. It'd basically require to double-buffer the whole display I think?

If the 128 u32 buffer fitted the entire screen, it'd be as simple as writing the old buffer to the first frame before updating it with the new data, but this seems to be only a partial linebuffer with all the subrectangle stuff going on. So the double-buffering would have to happen somewhere else, and not all circuitpython targets might have the RAM for that I imagine.

So I think it'd be a nontrivial change, but fast updates would allow a lot of applications that are currently out of reach without ditching circuitpython/displayio (PervasiveDisplays provides Arduino code for fast updates on request)

tannewt, 2022-11-15:
Does the panel lose the old memory between updates? We could write it after the update with the "new" data so that the next update has a reference.

pepijndevos, 2022-11-19:
Every update you need to write both tables with the old and new data as far as I understand.