adafruit/circuitpython #6877: Support for the Tiny pico Nano from UM

dhalbert, 2022-09-07:
Can we get support for the Tiny pico Nano from UM, added to the circuitpython supported boards?

_Originally posted by @sshinnawy in https://github.com/adafruit/circuitpython/issues/6685#issuecomment-1239739189_

ladyada, 2022-09-07:
@unexpectedmaker if ya can help out - it would be awesome :)

UnexpectedMaker, 2022-09-07:
I've not had a chance to look at the ESP32 support yet, I've been moving into a new workshop/factory, but I'll have a look at this today and add support for my TinyPICO and Nano boards :)

sshinnawy, 2022-09-16:
@UnexpectedMaker Hopefully you've settled into the new workshop. Please update me if the support for CircuitPython is going well or not, or if it's expected soon. Thanks for all you do!

UnexpectedMaker, 2022-09-17:
I have CP running on the TinyPICO, though it needs more testing and some tidying up before it's good to do a PR.
I've not looked at the TinyPICO Nano yet as that's going to be much harder to validate a fully compatible build.

UnexpectedMaker, 2022-09-28:
Support has been added - but no builds yet until the next version of 8 kicks off:
https://circuitpython.org/downloads?q=tinypico

I think we can close this now @dhalbert, cheers  :)

closed 2022-09-28 by dhalbert

sshinnawy, 2022-09-30:
> Support has been added - but no builds yet until the next version of 8 kicks off: https://circuitpython.org/downloads?q=tinypico
> 
> I think we can close this now @dhalbert, cheers :)

Thank you so much for this.

@dhalbert how to do I find out as soon as the CP v8 is released?

jepler, 2022-09-30:
You can subscribe to release notifications on github. Click the arrow next to "watch" at the top of the page, then select "custom", then turn on the "releases" checkbox, and finally click apply. Note that the update to circuitpython.org can still lag behind this, as it's a separate manual step of our release process

Even for boards that have not yet been in a tagged release, you can find the uf2 or other binary build on our s3: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/

saraceni-andrea, 2022-10-04:
Hi, is there the possibility of a port of CircuitPython 8 for **SQFMI Watchy**?

https://github.com/sqfmi/Watchy

It is equipped with **ESP32 PICO-D4**, like Unexpected Maker's TinyPICO, but **without PSRAM**.

https://watchy.sqfmi.com/docs/hardware
https://circuitpython.org/downloads?q=TinyPICO

Thanks!

UnexpectedMaker, 2022-10-04:
@saraceni-andrea but it's nothing like a TP Nano -  in term of IO used and functionality, so please create a new issue and post your request there as it's not on topic for here, sorry.

saraceni-andrea, 2022-10-04:
Ooops!

@UnexpectedMaker forgive me! :-)

https://github.com/adafruit/circuitpython/issues/6990