adafruit/circuitpython #6892: CircuitPython 8.X title-set ansi code sometimes ends up in stdin

bill88t, 2022-09-11:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; Raspberry Pi Pico with rp2040
Affects all boards, the chance of it occuring it higher on faster boards.
```

### Code/REPL

```python
Ljinux
```

### Behavior

[image]

### Description

Sometimes, when polling stdin, the console receives parts of the title-set ansi escape code sequence, circuitpython 8.x uses.
This means that at some point during the boot proccess, circuitpython decided to send the title-set, but sent it incompletely..

### Additional information

1. Ansi escape codes should be sent using stdout, NOT stdin..
2. Should you need help with the ansi code part I can help, however my C / C++ is mediocre at best.
3. Please provide a way to disable the constant title spam.
4. It's ok to spam more \007 just to be sure, they will never appear.

bill88t, 2022-09-11:
An easy mitigation for me was to simply create a bufferclear command and add it to the boot scripts, right after the serial connection occurs.

Also, it IS in stdin, it's treated as text in the buffer, and I can move my cursor to it.

dhalbert, 2022-09-11:
You can now disable the status bar: #6847. For example, in boot.py:
```py
import supervisor
supervisor.status_bar.console = False
```
> This means that at some point during the boot proccess, circuitpython decided to send the title-set, but sent it incompletely..

I don't think it is being sent incompletely, but the sending starts before the USB connection is fully up. This should be fixed.

dhalbert, 2022-09-28:
#6952 should fix this, on boards that have native USB. The status bar is not written out at all if, when the write starts, the serial connection is not available.

On boards that have an on-board serial-to-USB converter chip, the USB status pins on that chip may not be connected to anything, which is often the case. In that case, there is no way to know if the host is connected via USB, so there may still be situations where the first part of the status-bar write is missed, and the rest appears in-line. We might consider disabling the status bar by default on such boards, or delaying its enabling by several seconds.

(Re stdin: the status bar is written to the serial output like anything else, such as REPL output. Normally the escape code at the beginning would cause the terminal program to remove it from the what is displayed, but when the first part of the escape code is missed, it looks like "normal" output, and appears in the terminal window.)

bill88t, 2022-09-28:
Ι do not have any boards that require serial-to-usb so I cannot test that, however I will test the pr artifacts on my beetle-c3 and pico and provide any findings (if any) on the pr thread.

(Re Re stdin:
[image]
```
n = runtime.serial_bytes_available
if n > 0 or self.stdin is not None:
   i = None
   if self.stdin is not None: # fake stdin
          i = self.stdin # fake stdin
          self.stdin = None # fake stdin
   else:
          i = stdin.read(n)
```
You can read more about my fully non-blocking input implementation [here](https://github.com/bill88t/jcurses/blob/070fc1087bc57fd80ef410c86a92ee313f60a82c/jcurses.py#L287)
)

Leaving the issue open till merge.

bill88t, 2022-09-28:
Using artifact https://github.com/adafruit/circuitpython/actions/runs/3138352211 on raspberry_pi_pico
[image]
The bug is still there. It took 20 tries before I got it to show up.
It occured on a late connection.
In raw repl:
[image]
While it is still bugging, it doesn't go in stdin

bill88t, 2022-09-28:
@dhalbert

dhalbert, 2022-09-28:
I will re-test on RPi Pico now that it's merged.

bill88t, 2022-10-05:
Btw this bug is still alive and kicking in daily builds
[image]

dhalbert, 2022-10-05:
@bill88t what are you using as a terminal program?

bill88t, 2022-10-05:
If you are wondering, yes it does support ansi escape.
I can vouch for it, as I developed all of jcurses with it.
Jcurses uses most ansi escape sequences and I would have noticed if something was broken.

Also might I suggest a potential workaround?
Wouldn't it be possible to test if a terminal was present with a cursor query?
Like I mean asking (with ansi escape) the cursor position..
If it responds, it's connected. If it doesn't, it's not.
Only when it answers to size queries should the title begin to be set.
The logic could be hollow. Send the query bytes, `if stdin: it's here.`

If the terminal is at a goodnough state to reply to that, surely it's ready for a title set

dhalbert, 2022-10-05:
> Kde Konsole.

That's the terminal program you're using, but what are you using to connect to /dev/ttyACM0, or whatever? Are you using `tio`, or `picoom`, or something else? What is the exact command line, and what settings (including an initialization file, if any) are you suing?

E.g., I am using `gnome-terminal` and `tio`.

bill88t, 2022-10-05:
I am using GNU/screen for that `screen /dev/ttyACM0 115200`

PR #7041 (2022-10-12): delay first serial write to allow host setup time
Fixes #6892 and other cases in which serial output (usually USB output) is fed back into the REPL input, as noted on Linux. (@bill88t take note.)

@jepler's diagnosis of this is that serial terminal programs on Linux first open the serial channel with "echo mode" on, in which serial output is echoed back to the serial input. They then immediately turn echo off with an `ioctl()`, but there is a race condition in which one or more characters may get echoed before the echo gets turned off.

This PR adds 50 ms of delay before the very first write a [... cut]