adafruit/circuitpython #6893: CPY V7.3.3 on ESP32-S3-BOX hard crash

PaulskPt, 2022-09-12:
### CircuitPython version

```python
Initially:
Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; ESP32-S3-Box-2.5 with ESP32S3
Board ID:espressif_esp32s3_box
UID:C7FD1A1E7C45

After lots of 'internal watchdog timer expired' errors I downgraded CPY to:
Adafruit CircuitPython 7.3.3 on 2022-08-29; ESP32-S3-Box-2.5 with ESP32S3
Board ID:espressif_esp32s3_box
```

### Code/REPL

```python
# SPDX-FileCopyrightText: Copyright (c) 2022 Tim Cocks for Adafruit Industries
#
# SPDX-License-Identifier: MIT
# See: https://github.com/FoamyGuy/Adafruit_DisplayIO_FlipClock
# Added microcontroller and watchdog to try prevent 'watchdog timer expired' error (in CPY V8.0.0-beta0)
# Now running CPY V7.3.3
import time
import sys
import board
import wifi as wifi
import socketpool as socketpool
from displayio import Group
import adafruit_imageload
import adafruit_ntp
import adafruit_requests

from adafruit_displayio_flipclock.flip_clock import FlipClock
# 4 lines added by @PaulskPt to prevent watchdog timer expired errors
import microcontroller
import watchdog
wdt = microcontroller.watchdog
wdt.timeout = 40 # set watchdog timeout to 20 seconds
wdt.mode = watchdog.WatchDogMode.RAISE # Set watchdog mode
wdt.feed()

# microcontroller.disable_interrupts()

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

ANIMATION_DELAY = 0.01
TRANSPARENT_INDEXES = range(11)
BRIGHTER_LEVEL = 0.99
DARKER_LEVEL = 0.5
MEDIUM_LEVEL = 0.9
UTC_OFFSET = -5

# Get our username, key and desired timezone
aio_username = secrets["aio_username"]
aio_key = secrets["aio_key"]
location = secrets.get("timezone", None)
TIME_URL = "https://io.adafruit.com/api/v2/%s/integrations/time/strftime?x-aio-key=%s" % (aio_username, aio_key)
TIME_URL += "&fmt=%25Y-%25m-%25d+%25H%3A%25M%3A%25S.%25L+%25j+%25u+%25z+%25Z"

wifi.__init__()

print("My MAC addr:", [hex(i) for i in wifi.radio.mac_address])

print("Available WiFi networks:")
for network in wifi.radio.start_scanning_networks():
    print("\t%s\t\tRSSI: %d\tChannel: %d" % (str(network.ssid, "utf-8"),
            network.rssi, network.channel))
wifi.radio.stop_scanning_networks()

print("Connecting to %s"%secrets["ssid"])
try:
    wifi.radio.connect(secrets["ssid"], secrets["password"])
    pool = socketpool.SocketPool(wifi.radio)
    ntp = adafruit_ntp.NTP(pool, tz_offset=UTC_OFFSET)
except Exception as e:
    print("Error: ", e)
    raise

ipv4 = ipaddress.ip_address("8.8.8.8")
res = wifi.radio.ping(ipv4)
if isinstance(res, type(None)):
    print("Ping to {} failed".format(ipv4))
else:
    print("Ping google.com: {} mS".format(res))
wdt.feed()

pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

print("Fetching text from", TIME_URL)
response = requests.get(TIME_URL)
if response is not None:
    print("-" * 40)
    print(response.text)
    print("-" * 40)

display = board.DISPLAY

static_spritesheet, static_palette = adafruit_imageload.load("static_sheet.bmp")
static_palette.make_transparent(0)

top_animation_spritesheet, top_animation_palette = adafruit_imageload.load("grey_top_animation_sheet.bmp")
bottom_animation_spritesheet, bottom_animation_palette = adafruit_imageload.load("grey_bottom_animation_sheet.bmp")

SPRITE_WIDTH = static_spritesheet.width // 3
SPRITE_HEIGHT = (static_spritesheet.height // 4) // 2

clock = FlipClock(
    static_spritesheet, static_palette,
    top_animation_spritesheet, top_animation_palette,
    bottom_animation_spritesheet, bottom_animation_palette,
    SPRITE_WIDTH, SPRITE_HEIGHT, anim_delay=ANIMATION_DELAY, transparent_indexes=TRANSPARENT_INDEXES,
    brighter_level=BRIGHTER_LEVEL, darker_level=DARKER_LEVEL, medium_level=MEDIUM_LEVEL
)

clock.anchor_point = (0.5, 0.5)
clock.anchored_position = (display.width//2, display.height//2)

main_group = Group()
main_group.append(clock)
board.DISPLAY.show(main_group)
print()
print("@Foamyguy\'s \"Adafruit_DisplayIO_FlipClock test on an ESP32-S3-BOX trial")
while True:
    try:
        wdt.feed() # Feed the watchdog
        cur_time = ntp.datetime
        clock.first_pair = str(cur_time.tm_hour)
        clock.second_pair = str(cur_time.tm_min)
    except watchdog.WatchDogTimeout as e:
        print("Error:", e)
        raise
    except (OSError, ValueError):
        # no problem, try again next time.
        pass
    except KeyboardInterrrupt:
        print("KeyboardInterrupt. Exiting...")
        sys.exit()

    time.sleep(10)
```

### Behavior

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Crikey!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.3.3 on 2022-08-29; ESP32-S3-Box-2.5 with ESP32S3

### Description

Because, on ```https://circuitpython.org/board/espressif_esp32s3_box/``` there was no bootloader download for the ESP32-S3-BOX (I posted a note for that on circuitpython.org),
I downloaded the bootloader from: ```https://circuitpython.org/board/espressif_esp32s3_devkitc_1_n8r8/``` and flashed it.
The output.txt contents of this bootloader: TinyUF2 Bootlader 0.10.2 - tinyusb (0.12.0-203-ga4cfd1c69)
Model: Espressif ESP32S3 DevKitC 1
Board-ID: ESP32S3-DevKitC-v1.0 

Initially installed CircuitPython V8.0.0-beta0

### Additional information

When removing the 5Volt from the device and after re-connecting the device to (usb) 5Volt source, I saw in mu-editor some commands being executed, until (and including) the result of 'wifi.radio.start_scanning_networks()' . Immediately after the device went into a reset and next showing in REPL the hard crash info.

[image]

dhalbert, 2022-10-21:
@PaulskPt is this issue fixed in the 8.0.0 builds?

PaulskPt, 2022-10-21:
No.  I had totally forgotten that I had created this issue. Today I created anouther one (#7093) .

dhalbert, 2022-10-21:
We'll use that one.

Closing -- superceded by #7093.

closed 2022-10-21 by dhalbert

PaulskPt, 2022-10-21:
OK, tnx!