adafruit/circuitpython #6896: Brightness is zero on startup on FunHouse, Feather ESP32-S3, etc.

dhalbert, 2022-09-12:
After #6894, on-board displays are dark on initial boot on various boards, including FunHouse and Feather ESP32-S3. Not a problem on PyPortal.

gamblor21, 2022-09-12:
Set brightness is being called from the constructor and it does seem that (on my device) the brightness cannot be changed so the bug may be in set brightness.

dhalbert, 2022-09-12:
I neglected to say that in the REPL, it's possible to set `board.DISPLAY.brightness`, and it will work. It just comes up dark.

dhalbert, 2022-09-12:
I believe the problem is that `board_init()`, which sets up the display and the PWM, is called before `reset_port()` in `main()` (in `main.c`). So `pwmout_reset()` gets called after the initial brightness setting and gets messed up. Before #6894, brightness was set again, and the `PWMOut` was restored, but that isn't happening anymore. I will try some reordering.

dhalbert, 2022-09-12:
More investigation: `pwmout_reset()` appears to be buggy.

PR #6898 (2022-09-14): Espressif pwmout reset fixes
- Fixes #6896.

- The logic in `pwmout_reset()` was not checking the "never reset" flags for PWMOut correctly. It was partly the inverse of what it should have been. Fix this. The display brightness PWMOut was being reset multiple times due to this.
- `esp32-camera` code was resetting LEDC channel 0 clock by mistake, causing dark display on ESP32-S2 TFT.
- Removed some power-pin-reset bug workaround code which may or may not have been necessary due to this bug.
- Reordered display initialization in `main()` to avoid `pwmout_reset()` calls in `board.c` files.
- Remo [... cut]
closed 2022-09-14 by dhalbert