adafruit/circuitpython #6901: on circuitpython.org download page for device ESP32-S3-BOX no bootloader.zip available

PaulskPt, 2022-09-13:
The webpage:    ```https://circuitpython.org/board/espressif_esp32s3_box/ ```    doesn't have a section ```Install, Repair, or Update UF2 Bootloader``` like it's 'brother' on page: ```https://circuitpython.org/board/espressif_esp32s3_devkitc_1_n8r8/``` does.

For this reason I tried to build a port for the ESP32-S3-BOX in micropython/ports/espressif however unfortunately did not succeed. The build succeeded without error but did not create the .bin neither .uf2 files.

Neradoc, 2022-09-13:
Hi, the content of circuitpython.org is managed at https://github.com/adafruit/circuitpython-org
The bootloader section depends of the board having a bootloader_id [in its description](https://github.com/adafruit/circuitpython-org/blob/main/_board/espressif_esp32s3_box.md).
And for that the board must have [a board definition in the tinyuf2 repo](https://github.com/adafruit/tinyuf2/tree/master/ports/espressif/boards). Which it doesn't right now.

Neradoc, 2022-09-13:
I don't have the board, but note that adding it to tinyuf2 is pretty much similar to adding it to Circuitpython, copy an existing similar board (say the S3 TFT feather) and make a few changes, using the CP board files to help sometimes.

PaulskPt, 2022-09-13:
I just also was thinking about doing something like that.

PaulskPt, 2022-09-13:
in the circuitpython repo /ports/expressif/boards there is a board definition for the ESP32-S3-BOX. See the (edited) screenshot combo below. I guess that a board definition could be added to the tinyuf2 repo, isn't it ?

[image]

dhalbert, 2022-09-13:
Right, you could extract some or all of the info you need from the CircuitPython definition.

Closing this as it isn't a CircuitPython issue. If you have build issues, there is a Discussions section in the tinuf2 repo:
https://github.com/adafruit/tinyuf2/discussions

closed 2022-09-13 by dhalbert