adafruit/circuitpython #6916: audiomp3.MP3Decoder decodes Tags instead of skipping them

GeorgBisseling, 2022-09-17:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; Raspberry Pi Pico with rp2040
and a Pimoroni picoaudio board, but that may be irrelevant for this bug sind it works with the mp3 files provided with the examples.
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2020 Jeff Epler for Adafruit Industries
#
# SPDX-License-Identifier: MIT
#
# straight from the CircuitPython examples and adapted to the picoaudio board

"""CircuitPython Essentials Audio Out MP3 Example"""
import board
import digitalio
import audiobusio

from audiomp3 import MP3Decoder

try:
    from audioio import AudioOut
except ImportError:
    try:
        from audiopwmio import PWMAudioOut as AudioOut
    except ImportError:
        pass  # not always supported by every board!

button = digitalio.DigitalInOut(board.A1)
button.switch_to_input(pull=digitalio.Pull.UP)

# The listed mp3files will be played in order
mp3files = ["wdr5.mp3"]

# You have to specify some mp3 file when creating the decoder
mp3 = open(mp3files[0], "rb")
decoder = MP3Decoder(mp3)
#audio = AudioOut(board.A0)
# initialize picoaudio board
audio = audiobusio.I2SOut(board.GP10, board.GP11, board.GP9)

while True:
    for filename in mp3files:
        # Updating the .file property of the existing decoder
        # helps avoid running out of memory (MemoryError exception)
        decoder.file = open(filename, "rb")
        audio.play(decoder)
        print("playing", filename)

        # This allows you to do other things while the audio plays!
        while audio.playing:
            pass

#        print("Waiting for button press to continue!")
#        while button.value:
#            pass
```

### Behavior

Sound plays but contains noise. See below.

### Description

When playing a lame-encoded mp3 or the start of an internet radio stream the sound is intersected with periodical noice that seems to be generated by the interspersed meta data tags

https://user-images.githubusercontent.com/2944744/190852016-d5440531-14cd-4a5f-8226-dd884274cde0.mp4

.

For building an internet radio preprocessing the input data is not an option.
I suggest to have an optional argument to MP3Decoder() to enable skipping the embedded tags.

The attached sample file has to be renamed to wdr5.mp3

GeorgBisseling, 2022-09-17:
Related to
https://github.com/adafruit/circuitpython/issues/6850
since we all want to build an Internet Radio based on the Pico W. ;-)

jepler, 2022-09-17:
We are using https://github.com/adafruit/Adafruit_MP3 as our MP3 decoder. Can you see whether that project's Arduino or standalone examples have the same behavior, and if so report it there?

I found it tricky to get github to let me download your attachment properly. I eventually did, so here's a zip with the mp3 inside it:
[wdr5.zip](https://github.com/adafruit/circuitpython/files/9591665/wdr5.zip)

GeorgBisseling, 2022-09-17:
I do not have any compatible setup ready to try that out. I could however try to setup an Arduino Uno, Nano or Mega to try this. Give me some time to understand how to make that work.

But I played their test.mp3 on my setup and it worked like expected.