adafruit/circuitpython #6920: Very early code.py remounts go through, even though they should not.

bill88t, 2022-09-18:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0-41-g33a100611 on 2022-09-10; Waveshare ESP32-S2-Pico with ESP32S2
```

### Code/REPL

```python
import storage
storage.remount("/", False)
```

### Behavior

It goes through, even though there is no boot.py disabling usb access.
This results to the usb access being read only.

### Description

It should raise RuntimeError.
It does not.

### Additional information

If sleep till usb inits fully, it RuntimeError's just fine.

dhalbert, 2022-09-21:
We can't tell whether USB is going to be available or not. When the board powers up, we don't wait for USB, since we might wait forever. It's just not available at the time the `remount()` runs, so the `remount()` succeeds. We probably don't want to restrict `remount()` to being in `boot.py` only, because there are use cases for being in `code.py`, on, say, a battery-powered device.

bill88t, 2022-09-21:
Remounts in code.py should I not be mistaken are meant to be done only when one of the following is true:
 - the board only supports web-workflow
 - boot.py disabled usb access

In any other cases we should be getting a RuntimeError, yes?

If this is not the case it introduces a new issue:
We have no way of knowing if usb access is enabled from code.py.

It can take up to 3s for usb to init.
The code.py in that time cannot know.
And an early remount (even if used to test usb) can make the usb access readonly, which for a non wireless board can be kinda problematic, no?

A bool `usb_enabled` under storage, that is set to True by default and becomes false when boot.py has run disable_usb_drive would be more than enough.
```
def remount(...):
    if we_are_in_codepy and not usb_enabled:
         # do le remount
    else:
         raise RuntimeError
```

dhalbert, 2022-09-21:
You can use `supervisor.runtime.usb_connected` to check whether USB is available: https://docs.circuitpython.org/en/latest/shared-bindings/supervisor/index.html#supervisor.Runtime

bill88t, 2022-09-21:
That is a different thing but fine.. (usb_connected != usb_enabled)
 
So are the early remounts (along with the read-only bug) the intended behaviour?

dhalbert, 2022-09-21:
> So are the early remounts (along with the read-only bug) the intended behaviour?

We cannot tell whether USB connection is going to happen or not. So it's working as intended, but the usual idea is to do the remount in `boot.py`, not `code.py`.

bill88t, 2022-09-21:
Alright then. This closing this issue.

closed 2022-09-21 by bill88t