adafruit/circuitpython #6921: Add Microchip SAM4 support

skerr92, 2022-09-18:
I am adding this issue as a way to track and formalize an effort I plan to take to bring circuitpython support to the Microchip SAM4. The SAM4 is a great candidate for CircuitPython as it has lots of RAM, FLASH, Native USB, and plenty of functionality. 

I do imagine that some of the modules from the ATMEL-SAMD port can be used. With this new support, I would suggest a chance to the `port/atmel-samd` directory to be `port/atmel-sam` but this can be taken care of later.

Some questions I have before starting this effort:

1. Has this been considered before, if so is there a working example of CircuitPython running on the SAM4?
2. Does anyone have any SAM4 boards/would like SAM4 boards to help with development for this? I am designing a feather around the chip and hope to have a prototype ready before the end of the year.

dhalbert, 2022-09-19:
We haven't considered the SAM4 line strongly because the maximum RAM is 128kB, and we had more capable chips in the pipeline when this might have come up. Which chip are you targeting?

skerr92, 2022-09-19:
I know 128kB is a bit low on the RAM side, I am targeting this variant: https://www.digikey.com/en/products/detail/ATSAM4S16BA-MUR/3593650 

there are some things that probably are not necessarily a priority to have in its core which would probably help make it fit.

skerr92, 2022-10-07:
Given that there is no support for SAM4S in ASF4 at all, I'm going to close this since support efforts would be greater than the overall benefit.

closed 2022-10-07 by skerr92