adafruit/circuitpython #6924: dir() breaks under exec() (sometimes)

bill88t, 2022-09-19:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0-61-g5593d23a0 on 2022-09-19; DFRobot Beetle ESP32-C3 with ESP32-C3FN4
```

### Code/REPL

```python
>>> import board
>>> dir(board)
['__class__', '__name__', 'A0', 'A1', 'A2', 'A3', 'A4', 'A5', 'A6', 'A7', 'D0', 'D1', 'D10', 'D2', 'D3', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'I2C', 'IO0', 'IO1', 'IO10', 'IO2', 'IO20', 'IO21', 'IO3', 'IO4', 'IO5', 'IO6', 'IO7', 'IO8', 'IO9', 'LED', 'MISO', 'MOSI', 'RX', 'SCK', 'SCL', 'SDA', 'SPI', 'TX', 'UART', 'board_id']
>>> print(dir(board))
['__class__', '__name__', 'A0', 'A1', 'A2', 'A3', 'A4', 'A5', 'A6', 'A7', 'D0', 'D1', 'D10', 'D2', 'D3', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'I2C', 'IO0', 'IO1', 'IO10', 'IO2', 'IO20', 'IO21', 'IO3', 'IO4', 'IO5', 'IO6', 'IO7', 'IO8', 'IO9', 'LED', 'MISO', 'MOSI', 'RX', 'SCK', 'SCL', 'SDA', 'SPI', 'TX', 'UART', 'board_id']
>>> exec(print(dir(board)))
['__class__', '__name__', 'A0', 'A1', 'A2', 'A3', 'A4', 'A5', 'A6', 'A7', 'D0', 'D1', 'D10', 'D2', 'D3', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'I2C', 'IO0', 'IO1', 'IO10', 'IO2', 'IO20', 'IO21', 'IO3', 'IO4', 'IO5', 'IO6', 'IO7', 'IO8', 'IO9', 'LED', 'MISO', 'MOSI', 'RX', 'SCK', 'SCL', 'SDA', 'SPI', 'TX', 'UART', 'board_id']
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: object with buffer protocol required
>>>
```

### Behavior

[image]

### Description

Under ljinux it only sometimes errors out.

dhalbert, 2022-09-19:
Re `exec(print(dir(board)))`: `print()` returns` None`. Did you mean `exec(str(dir(board))))` ?

bill88t, 2022-09-19:
Ah sorry, during debugging my brain acted funky and forgot exec needs string.