adafruit/circuitpython #6930: accurately document partial BLE support on espressif

tlyu, 2022-09-20:
The documentation pages for `_bleio` don't clearly indicate that there is only partial BLE support for the espressif port. (Implementation is in progress and tracked by #5926.) This should be documented clearly in more places to avoid disappointing users who might purchase these devices hoping to have full BLE support in CircuitPython. I don't know offhand whether it will be easier to add a note to the support status of each board in the `_bleio` module or whether it makes more sense to add a warning near the top of the text of the `_bleio` documentation.

The relevant product pages should also be updated, but that probably gets done elsewhere.

dhalbert, 2022-09-21:
I think this is a more general problem of finding a consistent way of noting what is not implemented and what works.

 In general we have noted these kinds of limitations in the Learn Guides for each board. Many pages are shared or "mirrored" so this does not make guide maintenance onerous as it sounds. Sometimes we forget to note such things. 

We could make these notes in readthedocs, which comes from shared-bindings. But many beginners do not know about readthedocs. Or we could perhaps make it in some notes that are per port, and possibly per board in some cases. Perhaps there would be "limitations" file. Then that info has to get published somewhere. I think this is worth discussing maybe in a Monday meeting, and I will also bring it up among staff.

mlg556, 2022-09-30:
just wanted to add that I was bitten by this recently. I bought a ESP32-S3-DevKitC-1-N8 to build a BLE HID keypad, thinking it would work since I saw "_bleio" in the available modules, but I'm getting a NotImplementedError when trying to initialize a Service.

of course I'm sure it will be supported _eventually_, and I can always try micropython, but it was a bit of a let-down 😅

dhalbert, 2022-10-03:
I have generalized the title of this bug.

@jepler @kattni @tlyu @caternuson I wrote a draft page for the "Welcome to CircuitPython" guide (it does not have to stay there) that documents port and board-specific limitations, mostly of built-in modules. I found all these by looking for placeswhere `NotImplementedError` was called, and then added a few more things. The page is not published, but you can view it at this temporary link: https://learn.adafruit.com/welcome-to-circuitpython/board-and-port-specific-limitations?preview_token=WsdVgHQVBTr9iHLerIfyWA. 

Does a Guide page make sense for this kind of info? These limitations could also be documented in RTD, but a lot of people don't dig into that initially. As we add new features, this page will need to be revised, but that's true of anything.

I originally thought this might be some `.md` or `.rst` files in the source tree, in each port, that got collected somewhere into RTD, but it was easier to do as a guide page, and the information doesn't necessarily all divide neatly into ports.

We can discuss this in our internal meeting, in "In the Weeds" in the weekly meeting, and/or in #circuitpython-dev.

anecdata, 2022-10-03:
This page is really good. A couple of questions:

> DHCP client not yet available.
Something beyond this?
```
>>> wifi.radio.ipv4_address
192.168.6.201
>>> wifi.radio.ipv4_gateway
192.168.4.1
>>> wifi.radio.ipv4_subnet
255.255.252.0
```

> IPV4 DNS not yet available.

DNS lookups are available, but not setting custom DNS server(s).

kairos0ne, 2023-05-14:
Why do you restrict the import of the module adafruit_ble if the board is not supported. It seems strange to do such a thing given the market offering of the airlift coprocessor ? 

```
Traceback (most recent call last):
  File "main.py", line 16, in <module>
  File "adafruit_ble/__init__.py", line 25, in <module>
ImportError: no module named '_bleio'
```

dhalbert, 2023-05-14:
> The page is not published, but you can view it at this temporary link: https://learn.adafruit.com/welcome-to-circuitpython/board-and-port-specific-limitations?preview_token=WsdVgHQVBTr9iHLerIfyWA.

We ended up not doing this. Instead, there are some limitations listed in https://learn.adafruit.com/welcome-to-circuitpython/frequently-asked-questions.  In the ReadTheDocs pages, there are now **Limitations:** sections describing various port-specific limitations.

dhalbert, 2023-05-14:
> Why do you restrict the import of the module adafruit_ble if the board is not supported. It seems strange to do such a thing given the market offering of the airlift coprocessor

`adafruit_ble` depends on `_bleio`. There are several implementations of `_bleio`:
1. nRF native implementation.
2. Espressif native implementation. This implementation is incomplete: see https://learn.adafruit.com/welcome-to-circuitpython/frequently-asked-questions#faq-3129409
3. BLE HCI implementation done by a co-processor, e.g. the AirLift co-processor.

"3." `_bleio` is omitted from boards that don't have space for it. Which board are you using?

kairos0ne, 2023-05-14:
I have tried with the featherS2 and featherS3 by unexpected maker and I’ve also tried the adafruit Rp2040 feather I’ve had no such luck with any of them but the strangest thing is I had it working on 6.x circuit python with the rp2040 but not amnymore on the later versions.

kairos0ne, 2023-05-14:
Obviously I meant along side the Airlift Featherwing, plz excuse my tone I’m a little frustrated that I bought three boards and two airlifts – my project requires BLE only to find that its not supported for these boards. The Airlift coprocessor is advertised as a BLE solution for any feather based board and if there are three that it in fact doesn’t work with then people should know that before they spend money…

kairos0ne, 2023-05-15:
Yeah RP2040 seems to give me a different error, however I think I'm getting the gist of what you wrote and whats been written: 

Circuit Python Serial Monitor

```
Traceback (most recent call last):
  File "main.py", line 55, in <module>
  File "adafruit_airlift/esp32.py", line 199, in start_bluetooth
_bleio.BluetoothError: Timeout waiting to write HCI request
```

dhalbert, 2023-05-15:
For the RP2040, this sounds like a support problem. Could you open a thread in https://forums.adafruit.com/viewforum.php?f=60 ? Include the simplest program that causes the error -- it sounds like either the firmware on the AirLift is out of date or there is a wiring problem or a pin setup problem.

Also in the support thread we can discuss the Espressif issue: it turns out getting AirLift `_bleio` working on Espressif is not so easy.

karavelioglu, 2024-04-13:
Hey folks,. I am getting a _bleio problem on HUZZA32 Feather board. Is there a workaround?

https://forums.adafruit.com/viewtopic.php?p=1011498#p1011498