adafruit/circuitpython #6934: Add support for the Wemos D1 R32 Development board.

mirkokral, 2022-09-22:
It is an Arduino-shaped esp32 dev board, and i use the [the basic esp32 board](https://micropython.org/download/esp32/) when i use micropython.

mirkokral, 2023-02-07:
UPDATE: Looks like if i use the Adafruit Feather HUZZAH32 board it works like a charm.
Should i close or not?

dhalbert, 2023-02-07:
If it's basically the same, yes, let's close. Each board we add increases the time spent on every build and PR.

mirkokral, 2023-03-26:
I have made my own implementation based on the doit devkit, just added pins, built it and it worked like a charm!

closed 2023-03-26 by mirkokral

eadmaster, 2024-06-14:
which binary do you suggest to flash on this board?

it is still not listed in the [downloads page.](https://circuitpython.org/downloads)

RetiredWizard, 2024-06-14:
> which binary firmware do you suggest to flash on this board?
> 
> it is still not listed in the [downloads page.](https://circuitpython.org/downloads)
> 
> EDIT: the board seems to have wifi issues with the Esp32 firmwares i've tried (not from circuitpython) . Not sure if it is fixable at firmware-level.

According to @mirkokral above:

> UPDATE: Looks like if i use the Adafruit Feather HUZZAH32 board it works like a charm. Should i close or not?

eadmaster, 2024-07-14:
just tried the Adafruit Feather HUZZAH32 build, no issues found so far.

btw having a ["generic" ESP32 build like micropython](https://micropython.org/download/ESP32_GENERIC/) coulde be a good idea for boards like this (more obvious pick over a board that looks totally different).

dhalbert, 2024-07-15:
> just tried the Adafruit Feather HUZZAH32 build, no issues found so far.
> 
> btw having a ["generic" ESP32 build like micropython](https://micropython.org/download/ESP32_GENERIC/) coulde be a good idea for boards like this (more obvious pick over a board that looks totally different).

The `adafruit_huzzah32_breakout` might be the most generic. We might need multiple generic builds based on the different available ESP32 modules.