adafruit/circuitpython #6938: `ulab` module missing from current 7.3.3 Matrix Portal release

CedarGroveStudios, 2022-09-23:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Matrix Portal M4 with samd51j19
```

### Code/REPL

```python
import ulab
```

### Behavior

```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Matrix Portal M4 with samd51j19
>>> import ulab
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'ulab'
>>> ```

### Description

7.3.2 contains the `ulab` module as does 8.0.0-beta.0

circuitpython.org isn't listing the `ulab` module for the MatrixPortal 7.3.0:

<img width="643" alt="Screen Shot 2022-09-22 at 7 29 05 PM" src="https://user-images.githubusercontent.com/29906257/191881876-79bd6399-b6c1-410c-a20a-a3865a069536.png">

CedarGroveStudios, 2022-09-23:
7.3.2 contains the `ulab` module as does 8.0.0-beta.0

circuitpython.org isn't listing the `ulab` module for the MatrixPortal 7.3.0:

[image]

Neradoc, 2022-09-23:
Yep that is intentional and was the subject of a discussion. I'm sure it can be revised.
https://github.com/adafruit/circuitpython/pull/6779
https://github.com/adafruit/circuitpython/pull/6809

CedarGroveStudios, 2022-09-23:
Thanks @Neradoc. I dropped a comment/question in https://github.com/adafruit/circuitpython/pull/6779

CedarGroveStudios, 2022-09-23:
I’ll stick with 7.3.2 and will place a note in PaletteFader to warn users.

closed 2022-09-23 by CedarGroveStudios