adafruit/circuitpython #6941: pool.getaddrinfo() raises OSError of -2 when attempting to resolve invalid DNS name

jdimpson, 2022-09-23:
The behavior below is correct, but the Exception type / error message could be more helpful.

Example problem code and results

```
Adafruit CircuitPython 7.3.2 on 2022-07-20; Adafruit QT Py ESP32S2 with ESP32S2
>>> 
>>> import wifi
>>> import socketpool
>>> from secrets import secrets
>>> wifi.radio.connect(secrets["ssid"], secrets["password"])
>>> pool = socketpool.SocketPool(wifi.radio)
>>> server="nosuchhostname"
>>> pool.getaddrinfo(host=server, port=443)[0][4][0]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -2
>>> 
```

Expected result:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SocketPoolNameError: No such domain name
```

or at least some text describing what OSError of -2 means, but since DNS failure is a common error, it would be nicer to be able to do, say:
```
try:
  pool.getaddrinfo(host=server, port=443)[0][4][0]
except SocketPoolNameError:
  server = false  # mark unavailable
```

rather than special-casing the exception every time you do a DNS lookup:
```
try:
  pool.getaddrinfo(host=server, port=443)[0][4][0]
except OSError as e:
  if int(str(e)) == -2:
    server = false  # mark unavailable
  else:
    raise e
```

Thank you for your consideration and all your hard work!

DJDevon3, 2022-09-29:
Ran into something similar on my S3 feather tonight with OSError -2 pointing at `response = requests.get(DATA_SOURCE).json()` but more specifically line 509 in adafruit_requests.py (8.0.0-beta.0-67) for _get_socket. This happened as I lost power during hurricane ian. 

No clue if this is helpful, first time i've seen an actual error instead of the hard fault handler crash so wanted to document it somewhere, this seems appropriate.

jepler, 2022-11-28:
For compatibility with 7.x, code can catch `OSError` and check for `e.errno == -2`

PR #7269 (2022-11-29): Add SocketPool.gaierror
gaierror(-2) is raised in the failure case of getaddrinfo. This is compatible with cpython's socket module.

Typical session:
```
>>> import socketpool
>>> import wifi
>>> socket = socketpool.SocketPool(wifi.radio)
>>> try: socket.getaddrinfo("boo", 0)
... except socket.gaierror as e: ee = e
...
>>> type(ee)
<class 'gaierror'>
>>> ee.errno == socket.EAI_NONAME
True
>>> ee.strerror
'Name or service not known'
>>> raise ee
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 1, in <module>
gaierror: (-2, 'Name or service not know [... cut]
commit aef55b7e (2022-11-29): Add SocketPool.gaierror

gaierror(-2) is raised in the failure case of getaddrinfo. This is
compatible with cpython's socket module.

Typical session:
```
>>> import socketpool
>>> import wifi
>>> socket = socketpool.SocketPool(wifi.radio)
>>> try: socket.getaddrinfo("boo", 0)
... except socket.gaierror as e: ee = e
...
>>> type(ee)
<class 'gaierror'>
>>> ee.errno == socket.EAI_NONAME
True
>>> ee.strerror
'Name or service not known'
>>> raise ee
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 1, in <module>
gaierror: (-2, 'Name or service not kno [... cut]
closed 2022-11-29 by MicroDev1