adafruit/circuitpython #6954: Hardfault on esp32s3 when trying to release either a asyncio lock or event

oldfartradioman, 2022-09-27:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0-30-g14498f793 on 2022-09-04; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
```

### Code/REPL

```python
import  asyncio
from asyncio import Lock

async def task(i, lock):
    while 1:
        await lock.acquire()
        print("Acquired lock in task", i)
        await asyncio.sleep(0.5)
        lock.release()

async def main():
    lock = asyncio.Lock()  # The Lock instance
    for n in range(1, 4):
        asyncio.create_task(task(n, lock))
    await asyncio.sleep(10)

asyncio.run(main())  # Run for 10s
```

### Behavior

%Run -c $EDITOR_CONTENT
Acquired lock in task 1

Connection lost (EOF)

Use Stop/Restart to reconnect.

### Description

Have identical issue with Event

dhalbert, 2022-09-30:
If you have time, could you try to replicate this on MicroPython? We use their implementation with very few changes.

rbedia, 2022-10-03:
I reported a similar issue here: adafruit/Adafruit_CircuitPython_asyncio#29

The root cause appears to be the change from `yield` to `await core.sleep(0)` in `acquire()`. Using yield takes the lock acquiring task out of the task queue and it only gets put back into the task queue by the lock releasing task. Using sleep causes the acquiring task to put itself back into the queue after the sleep finishes and then acquire the lock even though it wasn't yet released because there isn't a check if the lock is available.

oldfartradioman, 2022-10-03:
Everything works fine on Micropython.

closed 2022-10-18 by dhalbert