adafruit/circuitpython #6955: Hardfault when after 8-10 pings to n IP every 200msec on ESP32-S2/S3

todbot, 2022-09-28:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; Adafruit QT Py ESP32-S3 no psram with ESP32S3
Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; S2Mini with ESP32S2-S2FN4R2
Adafruit CircuitPython 7.3.3 on 2022-08-29; S2Mini with ESP32S2-S2FN4R2
Adafruit CircuitPython 7.3.0-rc.1 on 2022-05-18; Adafruit QT Py ESP32-S3 no psram with ESP32S3
```

### Code/REPL

```python
import time
import wifi
import ipaddress
from secrets import secrets
ip_to_ping = "1.1.1.1"
print("connecting...")
wifi.radio.connect(ssid=secrets['ssid'],password=secrets['password'])

print("my IP addr:", wifi.radio.ipv4_address)
print("pinging ",ip_to_ping)
ip1 = ipaddress.ip_address(ip_to_ping)
while True:
    print("ping:", wifi.radio.ping(ip1))
    time.sleep(0.2)
```

### Behavior

```
[18:05:04.211] Waiting for tty device..
[18:05:05.222] Connected
my IP addr: 192.168.42.189
pinging  1.1.1.1
ping: 0.0
ping: 0.0
ping: 0.017
ping: 0.026
ping: 0.037
ping: 0.048
ping: 0.017
ping: 0.013

[18:05:11.276] Disconnected
[18:05:12.293] Connected
Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; Adafruit QT Py ESP32-S3 no psram with ESP32S3
>>>
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

Above is the serial output via `tio` for one example run.  You can ping about 8-10 times, then it HardFaults.  

The same code with `time.sleep(1)` does not seem to crash.

### Additional information

This was originally reported by @Wind-stormger here https://github.com/todbot/circuitpython-tricks/issues/10

Wind-stormger, 2022-09-28:
Thanks for testing again.

anecdata, 2022-09-28:
Is this distinct from https://github.com/adafruit/circuitpython/issues/5980?

Wind-stormger, 2022-09-28:
@anecdata It did not enter safe mode, as can be observed, it just restarted.

todbot, 2022-09-28:
Yep, this is a dupe of 5980, with the only addition is that it is still a problem in 8.0 and on ESP32-S2 and ESP32-S3. 

I will close this issue.  Apologies, I didn't search hard enough in existing issues before submitting this one.

closed 2022-09-28 by todbot

anecdata, 2022-09-28:
Tod would you mind commenting with the updated findings?

There's an older ping issue on Espressif, but it's closed https://github.com/espressif/esp-idf/issues/5202

todbot, 2022-09-28:
You mean leave a comment in 5980 with 8.0 & S2/S3 info?

anecdata, 2022-09-28:
Please, yes, that would bring the comments up to date.