adafruit/circuitpython #6956: Feather S3 hard faulting denies use of REPL to any other board while connected via USB.

DJDevon3, 2022-09-28:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0-67-gd44edc4a0 on 2022-09-25; FeatherS3 with ESP32S3
Board ID:unexpectedmaker_feathers3
# Hard faulting board, connected via USB

Adafruit CircuitPython 8.0.0-beta.0-72-ga7b10d41b on 2022-09-26; Adafruit Feather ESP32S2 with ESP32S2
Board ID:adafruit_feather_esp32s2
# Attempting to work with this board on different project
```

### Code/REPL

```python
[Contents of my Feather S3](https://github.com/DJDevon3/My_Circuit_Python_Projects/tree/main/Boards/espressif/Unexpected%20Maker%20Feather%20S3/Online-Offline%20Weatherstation)
```

### Behavior

```py
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues
```

### Description

2 boards plugged into USB.
1 Adafruit Feather S2
1 UM Feather S3

Open code.py manually on Feather S2 in Mu and open serial. While the Feather S3 is hard faulting it denies use of repl to any other board. The only way around this issue is to disconnect the S3 while working on the S2.

### Additional information

There are a lot of reports on the feather s3 hard faults. This is different and unexpected behavior vs the hard faults I was getting while trying to use the Feather S3. I feel this one might be a clue and deserves its own bug report just in case it's unrelated.

dhalbert, 2022-09-30:
What is the host OS?

Each board has its own serial port. If you connect explicitly to the correct serial port for the non hard-faulting board, are you saying that repeated hard faults on the "bad board" block talking to the "good" board?

One thing I have noticed on Linux is that USB enumeration of a new device can block the USB system in general, but that would be an issue only if the board is repeatedly rebooting. That does not sound like the problem you are describing.

DJDevon3, 2022-09-30:
Correct. Windows 10.  Starting to get the feeling I have some kind of rogue/conflicting USB device. I need to conduct some unplugging tests, have about 10 USB devices. Mouse/KB/Printer/Webcam/Boards/etc..  Also my PC is wireless so maybe that's conflicting somehow. Too many times reporting issues with others saying their devices are working fine, points to my PC as the common denominator at least for some of it I think.  Others are experiencing hard faults specifically with the S3 but I'm also having issues on an RP2040, NRF52840, and ESP32-S2... while others cannot replicate my issues. I think this S3 bug in particular does exists but there's also something else going on with my setup.

dhalbert, 2022-09-30:
Consider using the "USB Device Cleanup Tool" discussed here, which may or may not help:
https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#device-errors-or-problems-on-windows-3094694

DJDevon3, 2022-09-30:
@dhalbert Sure enough one of my USB Host Controllers got corrupted. Port 12 wasn't enumerating correctly. I don't have a Port 12. Removed the controller which took out my mouse and keyboard. Hard shut down, restarted, and voila.

Still have an issue with the S3 not getting a new SSID on soft reboot... requires hard reboot.. but so far no safe mode or hard fault. Fingers crossed.

dhalbert, 2022-09-30:
Based on your testing this afternoon, should we close this?

DJDevon3, 2022-10-01:
Got a hard fault this morning putting the S3 through its paces.  Issue persists. Connected an S2, opened Mu, manually loaded S2 code.py, opened REPL, hard fault error disallows use of REPL. Unplug the hard faulting S3 and Mu immediately loads REPL.  Again unsure if this is a Mu problem for hard fault causing a USB issue.  No errors in windows device manager.

dhalbert, 2022-10-01:
You might try a different terminal program, like [Tera Term](https://ttssh2.osdn.jp/index.html.en) or putty. (I like Tera Term better.) This will connect to a specified COM port. You can open multiple instances at once.

I don't think that Mu is necessarily causing the hard fault, but the errors might be confusing Mu.

DJDevon3, 2022-10-01:
It does NOT exhibit these symptoms while used with a QT PY S3 and stemma BME380 sensor. Running a small ampule server in an attempt to do my own datalogging to a locally hosted webpage.  Written by danh apparently. :P
```py
# SPDX-FileCopyrightText: 2022 Dan Halbert for Adafruit Industries
#
# SPDX-License-Identifier: Unlicense

import board
import wifi
import socketpool
import ampule
import adafruit_bmp3xx
import os

i2c = board.STEMMA_I2C()  # uses board.SCL and board.SDA
bmp = adafruit_bmp3xx.BMP3XX_I2C(i2c)

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("Secrets File Import Error")
    raise

ssid = secrets["ssid"]
print("Connecting to", ssid)
wifi.radio.connect(ssid, secrets["password"])
print("Connected to", ssid)
print(f"Listening on http://{wifi.radio.ipv4_address}:80")

# Connect to WiFi
print("\n===============================")
print("Connecting to {}...".format(secrets["ssid"]))
wifi.radio.connect(secrets["ssid"], secrets["password"])
print("Connected!\n")

bmp.pressure_oversampling = 8
bmp.temperature_oversampling = 2
pressure = "Pressure: {:6.4f}".format(bmp.pressure)
temperature = "{:.1f}".format(bmp.temperature)

# Collect details about the system
details = {
    "Machine": os.uname().machine,
    "CP Version": os.uname().release,
    "Platform": os.uname().sysname,
    "Temperature": (temperature),
}

def static_file():
    """Load the web page from the CIRCUITPY drive."""
    with open('demo.html') as local_file:
        content = local_file.read()
    return content

pool = socketpool.SocketPool(wifi.radio)
socket = pool.socket()
socket.bind(['0.0.0.0', 80])
socket.listen(1)
print("Connected to {}, http://{}:80".format(secrets["ssid"], wifi.radio.ipv4_address))
print(temperature)

while True:
    ampule.listen(socket)
pass
```
Hit reload as many times as I want on the QT Py and REPL works fine even with the hard faulting S3 attached. Maybe it's an issue of the S2 & S3 combination?

DJDevon3, 2022-10-12:
This issue is related to #6791. Closing this one because it's a duplicate in a way.

closed 2022-10-12 by DJDevon3