adafruit/circuitpython #6977: Ussl module for Unix port

brianlinuxing, 2022-09-30:
Is there a broad timescale for the inclusion of the ussl module  in the Unix port?

upip, etc are non-functional without it.

It would be very useful to prototype apps within a Unix port and then transfer them to the devices, and it would really reduce debugging time, etc

Thank you, Brian :)

PS: I am very happy to help test any Unix port!

[image]

dhalbert, 2022-09-30:
We don't use the Unix port, except for testing. It still uses MicroPython conventions, such as `u*` names, unlike CircuitPython. What are you using it for?

brianlinuxing, 2022-10-09:
Thanks for the reply Dan,

I have an app that uses a framework, which in turn imports a lot of modules, as a consequence I need to trim them but that is next to impossible on the Pico as they won't load (no space). I have tried it.

Therefore, I would simulate on the Unix port as much as is possible (leaving out any Pico specific material), trim down the modules to a sensible size and then implement on the Pico.

So having a Unix port which is very similar to the Pico implementation, overall, would be very beneficial.

PS: I'll happily have any form of ssl, etc as long as I can get upip working, etc

o-murphy, 2025-07-31:
unix port really needed to simify development