adafruit/circuitpython #6978: Set strong drive mode on pwm pins on SAMD

deshipu, 2022-09-30:
The `DigitalInOut` sets pins to the strong drive mode, but the `PWMOut` doesn't. It would be nice to have it consistent, especially since PWM might be driving small LEDs and similar things.

I was hoping to make a pull request for this myself, but I'm not recently able to contribute much, so I thought I will report this, because it may potentially be a good first issue. One has only to look at the `DigitalInOut` code and copy the relevant lines to the `PWMOut` code.

diamant3, 2022-10-08:
can i take this issue and help me along the way, thanks!

dhalbert, 2022-10-08:
Please do! This should be simple.

deshipu, 2023-07-29:
@diamant3 are you still working on this, or is it free to be taken by other people?

diamant3, 2023-07-29:
> @diamant3 are you still working on this, or is it free to be taken by other people?

Yes it's free!, sorry.