adafruit/circuitpython #6983: TinyUSB *** PANIC ***  ep 0 out was already available

anecdata, 2022-10-03:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.1 on 2022-10-01; Raspberry Pi Pico W with rp2040
# also seen on Pico w/ 7.3.3 & Xiao 2040 w/ 7.3.3
```

### Code/REPL

```python
# unknown; anecdotally associated with _some_ USB / serial activity
```

### Behavior

> *** PANIC ***
> 
> ep 0 out was already available
> 

### Description

A couple of Discord users reported this:
https://discord.com/channels/327254708534116352/537365702651150357/1023410511276159077
(scroll up for the 2nd report; click @Neradoc's link for the 1st report.

Seems to come from TinyUSB here:
https://github.com/hathach/tinyusb/blob/111515a29cce668ea435904a397a1f0117114122/src/portable/raspberrypi/rp2040/rp2040_usb.c#L91

I saw this today trying to put some code at the beginning of `code.py` to check for whether serial was connected (`supervisor.runtime.serial_connected`). Even after removing that code, the PANIC persisted across several short power cycles. Finally went away after I powered down the Pico W for 15-30 seconds - perhaps some interaction with macOS was persisting it previously

dhalbert, 2022-10-03:
@hathach May be of interest to you.

dhalbert, 2023-04-14:
@anecdata TinyUSB has been updated with significant fixes since this report. If you have time, could you re-test?

anecdata, 2023-04-17:
Tried various combinations of printing as fast as possible before serial is connected, and calls to `supervisor.runtime.usb_connected` and `supervisor.runtime.serial_connected`.

Not able to reproduce with:
`Adafruit CircuitPython 8.1.0-beta.1-25-g4d7e54310 on 2023-04-16; Raspberry Pi Pico W with rp2040`
Though I don't think we ever saw a reproducible procedure. I also haven't seen this in Discord since the two previous occurrences mentioned above.

I think this can be closed.

Neradoc, 2023-04-17:
Same, haven't heard of it since.

closed 2023-04-17 by Neradoc