adafruit/circuitpython #6992: Gathering data on the CircuitPython memory weirdness.

bill88t, 2022-10-04:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.1 on 2022-10-01; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
from gc import mem_free, collect
from sys import exit

targ_mem = None
mem_offs = 69420 # offset to add, value set to allocate the needed ram

# tests

# for loop test
for_t = False

# int test
int_t = False

mem_offs = 0 # revert the value to 0, so we can actually use it

# script begin

def memstep(fail=True):
    global targ_mem, mem_offs
    collect()
    f = mem_free()
    diff = targ_mem - f + mem_offs
    if diff:
        print("-> Diff: " + str(diff), end="")
        if fail:
            print(" | This is not fine.\n\n--- FAIL ---", end="")
            exit(1)
        else:
            print(" | This is fine.",end="")
        print()
    del f, fail, diff
    collect()

collect()
targ_mem = mem_free()
print(f"\nTarget mem is set to {targ_mem}.")
# fstrings with vars leave collectable garbage
print("\nInitial       test")
memstep()

if for_t:
    print("Before  \"for\" test")
    memstep()
    
    print("In      \"for\" test")
    mem_offs -= 16
    
    for i in range(30):
        memstep()

    mem_offs += 16
    collect()
    print("After   \"for\" test")
    memstep()

if int_t:
    print("In      \"int\" test")
    i = 69420531221
    #collect()
    mem_offs -= 16
    memstep()
    
    print("Del     \"int\" test")
    del i
    mem_offs += 16
    #collect()
    memstep()

print("End           test")
collect()
memstep()

print("--- PASS ---")
exit(0)
```

### Behavior

As is, it works. This means the `memstep` function and the prints aren't leaving anything behind.

If we now `for_t = True`, we instead fail:
```
Target mem is set to 158560.

Initial       test
Before  "for" test
In      "for" test
After   "for" test
-> Diff: 16 | This is not fine.

--- FAIL ---
Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
if the offset wasn't there we would fail within the "for" test.
The outcome doesn't change if I replace the `memstep` with a `pass` inside the for loop.

If we then `for_t = False` and `int_t = True`, we yet again fail:
```
Target mem is set to 158560.

Initial       test
In      "int" test
Del     "int" test
-> Diff: 16 | This is not fine.

--- FAIL ---
Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
What even holds those 16 bytes???????????

dhalbert, 2022-10-05:
If you can come up with a simpler example that just prints `gc.mem_free()`, this might be able to be narrowed down. I am not sure if you are expecting `del` to free storage immediately: it does not. Note that the REPL uses storage for its command history.

bill88t, 2022-10-05:
> If you can come up with a simpler example

Ok sure I will need a few days though

> I am not sure if you are expecting del to free storage immediately

Isn't a gc.collect() gonna force it?

> Note that the REPL uses storage for its command history.

This should not affect code.py? Or are you talking about scrollback?

dhalbert, 2022-10-05:
> This should not affect code.py? Or are you talking about scrollback?

I am talking about up-arrow in the REPL

bill88t, 2022-10-06:
Focusing on the for loop allocation for now.
```
from gc import mem_free, collect
from sys import stdout

def primem():
    collect()
    a = mem_free()
    stdout.write(str(a))
    del a # a is used to not account for the str() in ram

collect()

stdout.write("\nStarting mem test with ")
primem()
stdout.write(" bytes free\n")

stdout.write("Using primem or printing will not change the usage. ")
primem()
stdout.write("\n")
stdout.write("Using primem or printing will not change the usage. ")
primem()
stdout.write("\nPretty sure the prints are loaded on script start..\n\n")

stdout.write("A for loop will however for the remaining of the script reduce the memory free by 16 bytes.")
for i in range(10):
    del i
    collect()
    primem()
    stdout.write("\n")
stdout.write("\n")

stdout.write("Look, There are NO NEW VARIABLES, but we have lost 16 bytes. ")
collect()
primem()
```
This is as simple as it gets for this one.
Save this as a code.py and let it run.
My output:
```
code.py output:

Starting mem test with 158224 bytes free
Using primem or printing will not change the usage. 158224
Using primem or printing will not change the usage. 158224
Pretty sure the prints are loaded on script start..

A for loop will however for the remaining of the script reduce the memory free by 16 bytes.158208
158208
158208
158208
158208
158208
158208
158208
158208
158208

Look, There are NO NEW VARIABLES, but we have lost 16 bytes. 158208
Code done running.
```

dhalbert, 2023-01-09:
@bill88t Is this still an issue for you?

bill88t, 2023-01-10:
Always has been, and always will be unless we do something about it.
I have tried to resolve it by myself in the code, however was unable to.
When I find time I need to present further data.

dhalbert, 2024-01-24:
I think I will close this for now, especially as memory consumption is even more confusing with 9.0.0 due to the dynamic heap growth.

closed 2024-01-24 by dhalbert