adafruit/circuitpython #6994: PinAlarm not working properly with value=True

kattni, 2022-10-04:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.1-6-gb62f8b35e on 2022-10-04; Adafruit Feather ESP32S2 with ESP32S2
```

### Code/REPL

```python
import board
import alarm

print("Alarm:", alarm.wake_alarm)

pin_alarm = alarm.pin.PinAlarm(pin=board.D12, value=True, pull=True)
print("Entering deep sleep.")
alarm.exit_and_deep_sleep_until_alarms(pin_alarm)
```

### Behavior

While connected to the serial console, it prints `Entering deep sleep.` and then nothing. The `Pretending to deep sleep....` message never shows up, and you can no longer interact with the serial console with CTRL+D/C. If you want to save and reload your code, you must press the reset button to do so.

### Description

If you connect the trigger pin to GND before reloading the board, and then connect it to 3V, it at least wakes up the board when connected to 3V, but it still seems to be a bit wonky.

If you update `value` to `False`, the issue is resolved, and everything responds appropriately. 

### Additional information

Narrowed the issue down to this after testing #6676 (the TimeAlarm+PinAlarm issue was a red herring, they work fine together.)

dhalbert, 2022-10-24:
I tested this on the same board. I added a little code to the test program to turn on the LED for two seconds before sleeping to make sure I knew when the program was running.

- If I set `value=True`, jumper D12 to GND, and hard reset, then, as reported, the "Pretending to deep sleep..." does not show up until after I move the jumper to 3V to wake things up. If I leave the jumper on 3V then it loops waking up and "Pretending", as expected.
- If I set `value=False`, jump D12 to 3V, and hard reset, then I see "Pretending to deep sleep" right away. I then wake it up jumpering to GND. But if I then leave the jumper on GND, it does not loop as I would expect. I have to pull the jumper out to wake it up.

Still investigating.

jepler, 2022-10-25:
Is this an edge vs level triggered interrupt problem?

PR #7125 (2022-10-26): Fix setup of Espressif PinAlarms
- Fixes #6994.
---
- Code to wait for a pull to happen was wrong -- it only waited in one direction, and it waited essentially forever. Not a good idea anyway, since the pin might be tied firmly to one value or the other. Instead, add a brief delay after setting all pulls.
- Make sure RTC is powered up during sleep.
- Unrelated corrections. to Metro ESP32-S2 debug UART pins.
closed 2022-10-26 by dhalbert