adafruit/circuitpython #7027: Raspberry Pi Zero 2W BLE problem

pietru2004, 2022-10-10:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; Raspberry Pi Zero 2W with bcm2837
```

### Code/REPL

```python
from adafruit_ble import BLERadio
ble = BLERadio()
ble.name
```

### Behavior

```python
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "adafruit_ble/__init__.py", line 162, in __init__
RuntimeError: No adapter available
```

### Description

I tried using Raspberry Pi Zero 2W in build BLE, but it is not being detected and sends error + I bought HM-10 BLE module, but it is still undetected by the board.

pietru2004, 2022-10-10:
Well I wanted to just have drag and drop for code so I would not have to run hole linux (laziness), is there way to mayby use external BLE module ?

dhalbert, 2022-10-10:
We support the AirLift ESP32 module via SPI, but that runs only as a BLE peripheral right now: https://learn.adafruit.com/adafruit-airlift-breakout/circuitpython-ble.

The filesystem on the RPi's is not that stable right now: https://github.com/adafruit/circuitpython/issues/5628. This port is currently highly experimental: you'd be better off with an nRF board for the best BLE support.

pietru2004, 2022-10-10:
Well I just wanted to make more advanced tank steering car using rasbery pi. Well mayby I can try to connect the HM-10 I mentioned above to arduino and then setup some passthrou to rasbery pi.

Anyway thank you for your time. 🙂

closed 2022-10-10 by pietru2004