adafruit/circuitpython #7028: Unexpected Maker FeatherS3 + RGB Matrix M0 Featherwing rgb_pins[0] is not on the same port as clock

DJDevon3, 2022-10-10:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.1 on 2022-10-01; FeatherS3 with ESP32S3
Board ID:unexpectedmaker_feathers3
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2022 DJDevon3 for Adafruit Industries
#
# SPDX-License-Identifier: MIT
# Used with Unexpected Maker FeatherS3
# Coded for Circuit Python & 8.0.0-beta.1 libraries

import time
from math import sin
import board
import displayio
import rgbmatrix
import framebufferio
import adafruit_imageload
import terminalio
from adafruit_display_text.label import Label

displayio.release_displays()
matrix = rgbmatrix.RGBMatrix(
    width=64, height=32, bit_depth=6,
    rgb_pins=[board.D6, board.D5, board.D9, board.D11, board.D10, board.D12],
    addr_pins=[board.A5, board.A4, board.A3, board.A2],
    clock_pin=board.D13, latch_pin=board.RX, output_enable_pin=board.TX,
    doublebuffer=True)

# Associate the RGB matrix with a Display so we can use displayio
display = framebufferio.FramebufferDisplay(matrix, auto_refresh=False)

# Quick Colors for Labels
text_black = 0x000000
text_blue = 0x0000FF
text_cyan = 0x00FFFF
text_gray = 0x8B8B8B
text_green = 0x00FF00
text_lightblue = 0x90C7FF
text_magenta = 0xFF00FF
text_orange = 0xFFA500
text_purple = 0x800080
text_red = 0xFF0000
text_white = 0xFFFFFF
text_yellow = 0xFFFF00

g = displayio.Group()
b, p = adafruit_imageload.load("images/Adafruit_32px_RGB_Matrix_Logo.bmp")
t = displayio.TileGrid(b, pixel_shader=p)
t.x = 20
g.append(t)

l = Label(text="Adafruit\nFeather ESP32-S3", font=terminalio.FONT, color=text_magenta, line_spacing=1.0)
g.append(l)

display.show(g)

target_fps = 10
ft = 1/target_fps
now = t0 = time.monotonic_ns()
deadline = t0 + ft

p = 1
q = 17
while True:
    tm = (now - t0) * 1e-9
    x = l.x - 1
    if x < -100:
        x = 63
    y =  round(12 + sin(tm / p) * 6)
    l.x = x
    l.y = y
    display.refresh(target_frames_per_second=target_fps, minimum_frames_per_second=0)
    while True:
        now = time.monotonic_ns()
        if now > deadline:
            break
        time.sleep((deadline - now) * 1e-9)
    deadline += ft
```

### Behavior

```py
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.1\Traceback (most recent call last):
  File "code.py", line 23, in <module>
ValueError: rgb_pins[0] is not on the same port as clock
]0;🐍Wi-Fi: off | 23@code.py ValueError | 8.0.0-beta.1\
Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

This code has been tested and confirmed working (by me) on the Adafruit Feather ESP32-S2 & S3.

Error on UM Feather S3:
rgb_pins[0] is not on the same port as clock

There are 2 versions of the RGB Matrix Featherwing. 1 is meant for the M0/M4/RP2040/ESP32-S2/ESP32-S3 and the other is meant for the NRF5240 feather.

UM doesn't provide a brd schematic so I can't trace the physical pins to ensure this M0 Featherwing is intended to be compatible instead of the NRF52840 RGB Matrix Featherwing. 

The board schematic UM provides is confusing since the chip layout in the schematic doesn't match a real world chip layout. There is no way to ensure the traces are going to the correct pins I need. There's no way to know which RGB Matrix Featherwing to use with the UM FeatherS3 board. I have the M0 version of the RGB Matrix Featherwing.

It's reasonable to assume the M0 featherwing should work instead of the NRF52840 version. Since a .brd file has not been made publicly available to see the chip-to-pin-header trace routing there's no way to confirm.

Let's assume all the pins are routed correctly then it's a circuit python library issue not working working with the FeatherS3 unless I'm missing something simple like using a pullup on every rgb pin, setting digitalio, or something like that. The Adafruit Feather doesn't do that so I can only assume it's just not working as intended.

dhalbert, 2022-10-17:
I think this is not a circuitpython core issue?

dhalbert, 2022-10-18:
I think this is a question of which chip/module pins are mapped to which Feather edge pins. It sounds like the FeatherS3 does not match pins with the capabilities needed with what the FeatherWing expects. We discussed this in discord. I _think_ you have the pin info you need now, so closing. Please reopen if this is an issue that can be addressed in the circuitpython core.

closed 2022-10-18 by dhalbert

DJDevon3, 2022-10-18:
Core? I thought I submitted it to the RGB Matrix library. My fault!