adafruit/circuitpython #7032: Inconsistent resolution of FQDN & mDNS between espressif & raspberrypi

anecdata, 2022-10-11:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.1-25-gec855a2b1 on 2022-10-10; Adafruit Feather ESP32-S2 TFT with ESP32S2
Adafruit CircuitPython 8.0.0-beta.1-20-g66f84f5f9 on 2022-10-07; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import socketpool
import wifi
import ipaddress
from secrets import secrets

HOSTS = ("adafruit.com", "adafruit.com.", "circuitpython.local", "circuitpython.local.",)

pool = socketpool.SocketPool(wifi.radio)

wifi.radio.connect(secrets["ssid"], secrets["password"])

for host in HOSTS:
    try:
        print(f"{ipaddress.ip_address(pool.getaddrinfo(host, 443)[0][4][0])}")
    except Exception as e:
        print(f"{e}")
```

### Behavior

`espressif`:
```
Adafruit CircuitPython 8.0.0-beta.1-25-gec855a2b1 on 2022-10-10; Adafruit Feather ESP32-S2 TFT with ESP32S2
>>> import code
104.20.38.240
-2
192.168.6.196
-2
```

`raspberrypi`:
```
Adafruit CircuitPython 8.0.0-beta.1-20-g66f84f5f9 on 2022-10-07; Raspberry Pi Pico W with rp2040
>>> import code
104.20.38.240
104.20.38.240
[Errno 2] No such file/directory
[Errno 2] No such file/directory
```

### Description

`espressif` resolves typical DNS and mDNS relative domain names; resolves mDNS relative domain names without importing the `mdns` module.

`espressif` does not resolve any FQDNs with the final dot, either DNS or mDNS.

`raspberrypi` resolves DNS relative domain names, and FQDNs with the final dot.

`raspberrypi` does not resolve any mDNS. This may be [expected](https://github.com/adafruit/circuitpython/pull/6933#issue-1381554442) as currently unimplemented.

### Additional information

See: [Multicast DNS](https://datatracker.ietf.org/doc/html/rfc6762#section-21)

> Most computer users neglect to type the trailing dot at the end of a fully qualified domain name, making it a relative domain name

This is probably not a significant issue in the big scheme of things. There may be occasions where FQDNs are linked or cut-and-pasted from a browser for use in CircuitPython code, and the final dot is easy to overlook.

jepler, 2022-10-11:
Thanks, this is good information!

I think it would be better to treat this as two different issues, because it will probably best be resolved through separate pull requests.

Please transfer some of your findings to https://github.com/adafruit/circuitpython/issues/7033 which I just filed, and then retitle this one to reflect that it's about the resolution of names with trailing dots, affecting espressif only (and trim the initial comment text too)? Thanks for doing that.

anecdata, 2022-10-18:
As of #7079,  `raspberrypi` now resolves all 4 combinations correctly.

`espressif` does not handle FQDNs with the final dot.

jepler, 2022-10-19:
As of our main branch, espressif's version of lwip lacks the required bug fix,
https://github.com/lwip-tcpip/lwip/commit/ef76bbe3f0b97e695dff4a8574fe1647a8fe07ed -- marking as upstream.

jepler, 2022-10-19:
Reported with Espressif: https://github.com/espressif/esp-idf/issues/10013

PR #7281 (2022-11-30): handle domain with trailing dot
Closes: #7032
commit 286efc18 (2022-11-30): handle domain with trailing dot

Closes: #7032
closed 2022-11-30 by dhalbert