adafruit/circuitpython #7042: Tab key support in code.circuitpython.org

domdfcoding, 2022-10-12:
(I couldn't find an issue tracker for the editor itself, so I'm posting here)

In the CircuitPython web editor, pressing the "tab" key on my keyboard moves focus to the "Editor" button at the bottom of the screen, instead of indenting the code (with a tab *or* 4 spaces, whichever you prefer), as a desktop editor would. Whilst it's common to navigate controls on a website with the tab key (and may be necessary for some users) it feels out of place in a source code editor.

Would it be possible to add an option to have the tab key indent the code, perhaps alongside the buttons at the top or bottom of the editor? It could even be tri-state: 4 spaces, a tab, or off.

DerBroader71, 2022-10-12:
The issues are around the use of the tab key are varied.
Should tab be used for indents, auto completion or navigation (WCAG)?
In a development environment, I think the ability to turn auto completion on/off should be customisable (ala Thonny) and there should also be the option to switch between \t and a customisable number spaces for indentation (2 or 4)
The main decision comes down to whether the accessibility requirement of the ecosystem outweigh those of aims of the web editor component.
IMHO Tab in an editor = 4 spaces (please don't start a war)

Neradoc, 2022-10-12:
Hi, the repository for code.circuitpython.org is here: https://github.com/circuitpython/web-editor

dhalbert, 2022-10-17:
@domdfcoding Could you repost this issue and the comment in https://github.com/circuitpython/web-editor ? I will close it here.