adafruit/circuitpython #7047: picow: increased power-consumption during light-sleep and deep-sleep

bablokb, 2022-10-13:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.1 on 2022-10-01; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
# see https://github.com/bablokb/pico-sleepcurrent/
# directory cp-timer
```

### Behavior

Compared to the Pico, the Pico-W has substantially higher current consumption. You can find some graphs here: https://github.com/bablokb/pico-sleepcurrent/.

- light-sleep: about 21mA vs. 17mA
- deep-sleep: about 17mA vs. 7mA

(both figures with TimerAlarm). I suppose this is due to the wifi-coprocessor. Is there anything that can be done here?

anecdata, 2022-10-13:
You can try the [power-save](https://docs.circuitpython.org/en/latest/shared-bindings/cyw43/index.html) modes. There are four constants available, but you can also supply your own integer (hex).

jepler, 2022-10-13:
It might also be worth making sure that when the board enters deep sleep mode, WL_ON aka GPIO23 is set to power off the wireless module. This probably requires a core change since GPIO23 is not available from Python code.

bablokb, 2022-10-14:
My test-program now shows 6mA in deep-sleep with active TimeAlarm. No changes with  light-sleep, but that was expected due to the nature of the patch.

Thanks for all of your effort and work!

jepler, 2022-10-15:
Thank you for testing and reporting back!

PR #7050 (2022-10-14): picow: Turn off wifi co-processor regulator when entering deep sleep
This reduces power consumption during true deep sleep.

In my measurements with ppk2 and a program that _irrevocably_ entered deep sleep (no time alarm or pin alarm), power usage as measured on a ppk2 decreased from ~10mA to ~1mA.

Closes: #7047 (or at least improves on the status quo)

ppk graph, before (right hand side represents deep sleep):
[image]
zoomed in on the right:
[image]

after, zoomed in on deep sleep:
[image]

Using pin alarms or time alarms still incurs a power cost. Light sleep and fake deep sleep are not imp [... cut]