adafruit/circuitpython #7067: wifi module on Raspberry Pi 4 B (bare metal)

ComCon2022, 2022-10-16:
I've recently installed Circuitpython on my Raspberry Pi 4 B (the "bare metal" version). The Raspberry Pi 4 B comes with Wi-Fi capabilities but Circuitpython doesn't support the Wi-Fi module. Are there any plans in the future to add this? Thanks

dhalbert, 2022-10-16:
There are no definite plans. The bare metal `broadcom` ports are in a very early stage, and have significant bugs.

tannewt, 2022-11-15:
The WiFi module is a relative of the one on the Pico W. However, it is connected through SDIO rather than PIO. The module code is also specifically licensed for use with an RP2040. https://github.com/georgerobotics/cyw43-driver

criptych, 2023-02-22:
I may be reading it wrong, but the license seems to apply to any Raspberry Pi device.  Emphasis mine:

[README.md](https://github.com/georgerobotics/cyw43-driver/blob/main/README.md)
> This cyw43-driver is also available for use with ***Raspberry Pi Ltd semiconductor devices*** under different terms (see LICENSE.RP in this directory).

[LICENSE.RP](https://github.com/georgerobotics/cyw43-driver/blob/main/LICENSE.RP)
> Raspberry Pi Ltd (Licensor) hereby grants to you a non-exclusive license to use this software solely with the Licensor's microcontroller chip (RP2040) ***or any other semiconductor device produced by the Licensor***.

bill88t, 2023-02-22:
Broadcom SoC, this would work for zero 2w though.

cleverca22, 2023-08-29:
https://iosoft.blog/zerowi/
this is a series of blog posts, on how to interface with the wifi on a pi-zero
all of the rpi devices (both broadcom and rp2040) use wifi devices from the same family, so you should be able to adapt this blog to other SDIO based chips

the rp2040 is an exception, because i think the wifi chip needs a unique firmware blob, to re-route things over SPI

tannewt, 2024-02-16:
This is a duplicate of #5633

closed 2024-02-16 by tannewt