adafruit/circuitpython #7068: Update nRF submodules for upcoming nRF7xxx

skerr92, 2022-10-16:
There isn't really a timeline right now of when the nRF7xxx chips are set to be released, but there have been some big updates to the nRF software baseline that will need to be updated in the nRF submodules that would enable it's support.

nRF in the latest nRFx added preliminary support for Matter and a preview to the nRF7 series chips. 

This is more of a long term task, and here are a few things that I think would be in that process up bring support for the nRF7 series into CircuitPython:

- [ ] updating nRF submodules to latest. This could be a breaking change for the nRF52 boards, so it's probably best for CP9 release 
- [ ] Create Adafruit Matter Submodule

I'm open to expanding these tasks. I haven't looked too deep at the Matter preview.

dhalbert, 2022-10-16:
We use our own version of `nrfx`, which uses the SoftDevice. My understanding was that Nordic has moved to zephyr, and away from the SoftDevice concept, for new chips (nrf5340 is an existing example). So just updating nrfx I don't think would support the new nRF7xxx (wifi) chips. I have not looked at Matter and what it uses internally.

skerr92, 2022-10-16:
nRFX might not be right, let me look and update the main issue.

skerr92, 2022-10-16:
I guess to extrapolate on the updated issue description;

the nRF7002 is just a companion/coprocessor so running circuitpython on it isn't really possible from my view. With this in mind, for there to be support for using the nRF7002, a specific firmware would need to be made to be loaded on the nRF7002 that is separate from circuitpython and not dependent on circuitpython. This would be to avoid the Zephyr issues with circuitpython.

Additionally, a SPI library would need to be defined so that when a circuitpython device wanted to communicate, it's as simple as using the SPI bus and sending the data. I don't like the word simple here, but I'm not sure how else to describe it.

Finally, there appears to be some [python libraries for Matter available](https://github.com/home-assistant-libs/python-matter-server) which might be forkable into a circuitpython library.

My main question would be, is this something that would want to be a Frozen module that could be included in builds if for instance boards with the nRF7002 attached are submitted to be added to circuitpython?