adafruit/circuitpython #7083: WPA2-Enterprise support ??

one0410, 2022-10-18:
Hi,
I found that there is no support for WPA2-Enterprise currently.
I really need this to connect to my customers enterprise network.
Do you have a plan to support it ??

dhalbert, 2022-10-21:
Which chip family do you want this on?

one0410, 2022-10-23:
I'm sorry for that I didn't mention for using ESP32-S3 currently.

MaxMyzer, 2023-01-19:
Any chance of Pi Pico W (CYW_43) support?

dhalbert, 2023-01-19:
We depend on https://github.com/georgerobotics/cyw43-driver for the Pico W, so we'd await that. It's not clear to me whether just defining some more authmode constants will work or it needs further support.

one0410, 2024-03-08:
Hi, any update for support enterprise wifi for esp32 ??

anecdata, 2024-03-08:
Looks like different methods have different levels of implementation (and use) complexity
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/wifi-security.html#wi-fi-enterprise
Which method(s) are needed?

one0410, 2024-03-09:
> Looks like different methods have different levels of implementation (and use) complexity https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/wifi-security.html#wi-fi-enterprise Which method(s) are needed?

I think that most company will use PEAP and EAP-TTLS most. These methods can be used with username & password.
Could you help to implement these methods first? Thank you very much.

anecdata, 2024-03-09:
I was just clarifying requirements. I'm not the one to do the implementation. I've never used enterprise and have no access to test.

one0410, 2024-03-09:
Thanks for your quickly response.
So do you know is there any plans or possibilities to implement these recently?

dhalbert, 2024-03-09:
We have no immediate plans, but it doesn't mean we don't want to. It's not so easy to come up with a test situation at home.

one0410, 2024-03-09:
Got it.
Wish we can have these features in near future when you team have time.
Thank you again.

mlamparter, 2026-02-04:
Just checking in to see if there has been any progress on this.  I have contributed zero to this project so I am incredibly grateful for the hard work others have put into.  Just want to make it clear I'm respectfully checking in, not making a demand :)

Not sure if it matters or offers any help, but it looks like MicroPython may provide support for this in their [upcoming 1.28 release on March 9](https://github.com/micropython/micropython/pull/17234).

I work at a university, so if having access to a test bed would be helpful for this feature I may be able to offer some assistance.