adafruit/circuitpython #7084: Configurable option to skip the safe mode yellow blinks boot sequence

billedluh, 2022-10-18:
I've got a project that goes in to deep sleep and waits for a button press to do something, but having to wait for 1 second to make sure I don't want to boot in to safe mode kills the project.  It would be really great to make the safe mode boot behavior configurable to give the option for projects that spring to life more when they wake from sleep or even when they get connected to power.   For most of my projects there's an early phase when I'm very thankful for features like this that protect me from myself , then a later phase when most everything is done and it makes sense to take off the caution tape.  (It also seems redundant with the bootloader hardware switch on my RP2040 board, but maybe this is more critical with other boards)

Thanks for all the great work you do around here!

dhalbert, 2022-10-18:
If you do come up with a custom build for this, as we discussed in discord, you could make it configurable at compile time, or even add some API to make it user-configurable, and we'd be happy to consider a PR.

deshipu, 2022-10-18:
There is already a compile-time option that you can use for this: `CIRCUITPY_SKIP_SAFE_MODE_WAIT`.

Neradoc, 2022-10-18:
I thought the safe mode blink was not enabled when coming out of deep sleep. That would make sense, after all if you need safe mode after deep sleep you can just press reset then enter safe mode.

dhalbert, 2022-10-18:
> I thought the safe mode blink was not enabled when coming out of deep sleep. That would make sense, after all if you need safe mode after deep sleep you can just press reset then enter safe mode.

Hmm, it does appear that way.`wait_for_safe_mode_reset()` in `supervisor/shared/safe_mode.c` determines whether to wait or not for the safe mode reset. It returns immediately in these circumstances:
https://github.com/adafruit/circuitpython/blob/b56d5d97d2ecdd11adb270770beabd0ed96c9c15/supervisor/shared/safe_mode.c#L62-L68.

`reset_reason` would be `RESET_REASON_DEEP_SLEEP_ALARM` after waking up from a deep sleep. So it should return right away.

@billedluh The delay you are seeing may be due to other things, or there may be a bug in the wakeup code for the RP2040.

billedluh, 2022-10-19:
It looks like microcontroller.cpu.reset_reason is returning

billedluh, 2022-10-19:
I should clarify that when connected to usb it simulates sleep and doesn't actually reset, so the reset reason doesn't get updated.  But when I disconnect and run over battery it consistently reports the reason as being SOFTWARE upon being awoken from deep sleep, regardless of the previous reset reason.  I'll let y'all decide if that's a bug or not.  

I'll poke at using the existing in compile time option, which looks like it should work.

rdlaner, 2023-01-02:
@billedluh, did changing `CIRCUITPY_SKIP_SAFE_MODE_WAIT` work well enough for reducing your boot time? I recently tried setting `#define CIRCUITPY_SKIP_SAFE_MODE_WAIT (1)` in my board config header, but not seeing the speed up in boot time that I was hoping for. Still seems to be around 2 seconds for me with an esp32-s2 (Magtag).

tannewt, 2023-01-10:
I'm going to close this because we have `CIRCUITPY_SKIP_SAFE_MODE_WAIT` and the safe mode blinks are skipped after waking from deep sleep.

@rdlaner I believe your delay may be coming from display initialization. (I'm seeing the delay on an ESP32-S3 USB OTG.) Please file an issue if you have a proposal for display behavior when waking from deep sleep. Thanks!

closed 2023-01-10 by tannewt