adafruit/circuitpython #7085: set_ipv4_address issue pico w

talavrahami99, 2022-10-18:
Getting the following error on set_ipv4_address line:
TypeError: extra positional arguments given

Code is below. Creating ipaddress objects as specified in the documentation for the wifi module. ipaddress objects are created correctly.

`dic = {}
with open('PCATS_WID_CONFIG.ini') as f:
    lines = f.readlines()
    for line in lines:
        if line[0] is not '#':
            line = line.strip()
            key = line.split(':')[0].lower()
            value = line.split(': ')[1]
            dic[key] = value

ip = ipaddress.ip_address(dic["ip"])
netmask = ipaddress.ip_address(dic["netmask"])
gateway = ipaddress.ip_address(dic["gateway"])
#print(ip,netmask,gateway)

wifi.radio.set_ipv4_address(ip,netmask,gateway) # LINE ERROR OCCURS ON
wifi.radio.connect(dic["ssid"],dic["password"])`

talavrahami99, 2022-10-20:
> Hi! Thanks for your report. This method is actually unimplemented on the
> Pico W at this time. However, the reason you did not get a NotImplemented
> error appears to be a simple usage error.
>
> I'll explain the usage error but changing your code will only get you the
> NotImplemented error, it won't fix the problem.
>
>     def set_ipv4_address(
>         self,
>         *,
>         ipv4: ipaddress.IPv4Address,
>         netmask: ipaddress.IPv4Address,
>         gateway: ipaddress.IPv4Address,
>         ipv4_dns: Optional[ipaddress.IPv4Address]
>     ) -> None: ...
>
> The "*" means that the following arguments are "keyword only", that is,
> they have to be of the form ipv4=ip in the argument list, not just ip.
>
> So, the corrected code would be:
>
> wifi.radio.set_ipv4_address(ipv4=ip, gateway=gateway, netmask=netmask)

Hi,

Thank you for the response. Is there a timeline for when this function
might be implented on the Pico W?

Kind regards,
Tal

closed 2022-10-20 by dhalbert

jepler, 2022-10-20:
It's implemented now, if you grab the very latest builds or wait for the next beta release. If you do please let us know if it worked for you, either here or (it it's not working for you) in a fresh issue

talavrahami99, 2022-10-26:
I've now tested this function on beta.3 and it seems to be working fine.
Thank you.

Kind regards,
Tal

PR #7088 (2022-10-20): pico w: implement static configuration methods in wifi.Radio
Closes: #7085