adafruit/circuitpython #7086: non-blocking tcp sockets not working on pico w

justbuchanan, 2022-10-19:
### CircuitPython version

```python
Current main branch. https://github.com/adafruit/circuitpython/commit/b56d5d97d2ecdd11adb270770beabd0ed96c9c15
```

### Code/REPL

```python
# Using adafruit_httpserver.py from https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/ba2da4352290b880592ae5ab5db83ee50c045244/adafruit_httpserver.py

from secrets import secrets
import socketpool
import wifi
from adafruit_httpserver import HTTPServer, HTTPResponse

ssid = secrets['WIFI_SSID']
wifi.radio.connect(ssid, secrets['WIFI_PASSWORD'])
print("Connected to", ssid)
print(f"Listening on http://{wifi.radio.ipv4_address}:80")

pool = socketpool.SocketPool(wifi.radio)
server = HTTPServer(pool)

@server.route("/")
def base(request):
    return HTTPResponse(content_type="text/html", body="<html><body>Hello from pico w</body></html>")

# Never returns
server.serve_forever(str(wifi.radio.ipv4_address))
```

### Behavior

When I make an http request to the pi with `curl`, I get `curl: (56) Recv failure: Connection reset by peer` and my route handler is not called. I'm able to get it to work by modifying `adafruit_httpserver.py` to set the socket to blocking.

Changing `False` to `True` here fixes things and the http endpoint responds as expected:

https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/ba2da4352290b880592ae5ab5db83ee50c045244/adafruit_httpserver.py#L334

### Description

I'm aware that the pico w wifi code is under active development (thanks for the great work so far!), so I'm not sure if non-blocking sockets haven't been fully implemented yet or if this is a bug. Or maybe I'm using it wrong.

If you have any pointers, I'd be happy to look into debugging this. I'm unclear on how non-blocking sockets are supposed to work though. If I call `socket.accept()`, then `socket.recv_frominto()`, at what point is the socket's `pbuf` supposed to be populated?

dhalbert, 2022-10-21:
@jepler Can you take a look at this?

jepler, 2022-10-21:
I will, though it'll probably be next week.

jepler, 2022-10-21:
```
        self._sock.bind((host, port))
        self._sock.listen(10)
        self._sock.setblocking(False)  # non-blocking socket
```
The nonblocking request in question is for the listening socket, as opposed to the connected socket created by accept().  This is just a note for myself.

foxy82, 2022-10-25:
In case it helps i'm using non-blocking server sockets outside of the HttpServer and they work fine:

`
import socketpool
import wifi
import time

from secrets import secrets

# Connect to WIFI
print()
ssid = secrets['WIFI_SSID']
print("Connecting to Wifi")
wifi.radio.connect(ssid, secrets['WIFI_PASSWORD'])
print(f"Connected: IP address is {wifi.radio.ipv4_address}")

HOST = str(wifi.radio.ipv4_address)
PORT = 80  # Port to listen on

print("Creating server socket")
pool = socketpool.SocketPool(wifi.radio)
server_socket = pool.socket(pool.AF_INET, pool.SOCK_STREAM)
server_socket.setblocking(False)
server_socket.bind((HOST, PORT))
server_socket.listen(10)
print(f"Accepting connections on {HOST}:{PORT}")

# A list of incoming sockets
inputs = []

buff = bytearray(128)

while True:
	for s in inputs:
		try:
			numbytes = s.recvfrom_into(buff)
			received_message = buff[: numbytes[0]]
			print(f"Received {received_message}")
		except OSError as e:
			# print("No data ", e)
			pass

	try:
		connection, client_address = server_socket.accept()
		if connection is not None:
			print("Connected by", client_address)
			connection.setblocking(False)
			inputs.append(connection)
	except OSError as e:
		# print(e)
		pass
		
	time.sleep(0.01)

`

closed 2022-10-25 by anecdata