adafruit/circuitpython #7093: esp32-s3-box with cpy 8.0.0-b.2 loops in reboot as soon as .env file is uploaded

PaulskPt, 2022-10-21:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.2 on 2022-10-14; ESP32-S3-Box-2.5 with ESP32S3
```

### Code/REPL

```python
Contents .env file:

`# To auto-connect to Wi-Fi
CIRCUITPY_WIFI_SSID=YOUR_SSID
CIRCUITPY_WIFI_PASSWORD=YOUR_PASSWORD
# Following are variables used by code.py`

To not publish my router's WiFi login details I show the contents here with 'YOUR SSID' and 'YOUR_PASSWORD' placeholders instead.
```

### Behavior

Immediately after copying the .env file to the  CIRCUITPY disk, the MCU starts with a looping reboot cycle.

### Additional information

Note that a reboot often also happens when, while using mu-editor, press CTRL+S to save the code.py file I'm working on,  at moments that the script is running.

Various times it happened that the device turned  CPY into 'Safe Mode'. So, I had to reset is manually to reboot and return to a 'normal' state.

This usually does not happen whenever I first issue a CTRL+C (KeybiardInterrupt) to stop the running script. and next when I have back the command prompt '>>' in mu-editor REPL, and issue then a 'save file' (CTRL+S) command.

PaulskPt, 2022-10-21:
I just upgraded CPY to V8.0.0-beta.3. Same result.

[image]

dhalbert, 2022-10-21:
> I just upgraded CPY to V8.0.0-beta.3. Same result.

Yes, there isn't anything in beta.3 that I would have expected to fix this. (Btw, could you paste text instead of screenshots? It makes it easier to copy/paste or quote.)

Neradoc, 2022-10-21:
I believe this is an issue with wifi connection.
Instead of `.env`, connect to wifi from code.py, I have the board reboot into safe mode.
Since safe mode does not disable wifi autoconnect from .env, that would cause a boot loop, but it's really a reboot caused by the wifi connection.
```py
import wifi
from secrets import secrets
print("Connecting")
wifi.radio.connect(ssid=secrets["ssid"], password=secrets["password"])
socket_pool = socketpool.SocketPool(wifi.radio)
```

```
code.py output:
Connecting

[16:05:50.333] Disconnected
[16:05:51.338] Warning: Could not open tty device (No such file or directory)
[16:05:51.338] Waiting for tty device..
[16:05:53.348] Connected
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.

Press any key to enter the REPL. Use CTRL-D to reload.
```

PaulskPt, 2022-10-21:
@Neradoc  I ran your example. Here is it and the REPL output:

import wifi
from secrets import secrets
print("Connecting")
wifi.radio.connect(ssid=secrets["ssid"], password=secrets["password"])
socket_pool = socketpool.SocketPool(wifi.radio)

REPL output:
    
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.

Press any key to enter the REPL. Use CTRL-D to reload.

jepler, 2022-10-21:
Is there anything unusual about your wifi setup?  For instance, are the ssid or password strings long? Do they contain non-alphanumeric characters? I see your latest test appears to exclude the `.env` file altogether (assuming you removed or renamed the file), which is good information to have.

PaulskPt, 2022-10-21:
Next I copied a part that I recently successfully used with an UM PROS3:
But it crashed also into safe mode with the 'internal watchdog' error:

```
import time
import wifi
import socketpool
import ipaddress
from secrets import secrets

use_ping = True 

print("Connecting")

try:
    wifi.radio.connect(ssid=secrets["ssid"], password=secrets["password"])
except ConnectionError as e:
    print("WiFi connection Error: \'{}\'".format(e))

ip = wifi.radio.ipv4_address

pool = socketpool.SocketPool(wifi.radio)

if ip:
    s_ip = str(ip)
if s_ip is not None and s_ip != '0.0.0.0':
    print("s_ip= \'{}\'".format(s_ip))
    print("connected to %s!"%ssid_to_connect)
    print("IP address is", ip)
    if use_ping:
        open_socket()
        addr_idx = 1
        addr_dict = {0:'LAN gateway', 1:'google.com'}
        info = pool.getaddrinfo(addr_dict[addr_idx], 80)
        addr = info[0][4][0]
        print(f"Resolved google address: \'{addr}\'")
        ipv4 = ipaddress.ip_address(addr)
        for _ in range(10):
            result = wifi.radio.ping(ipv4)
            if result:
                print("Ping google.com [%s]: %.0f ms" % (addr, result*1000))
                break
            else:
                print("Ping no response")
        free_socket()

while True:
    time.sleep(5)  ```

Result on serial output:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x15e8
load:0x403b6000,len:0xbf8
load:0x403ba000,len:0x31c8
entry 0x403b621c
I (24) boot: ESP-IDF v4.4.2-378-g9269a536ac 2nd stage bootloader
I (25) boot: compile time 11:36:39
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 16MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (71) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (86) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (93) boot:  4 uf2              factory app      00 00 00410000 00040000
I (101) boot:  5 ffat             Unknown data     01 81 00450000 00bb0000
I (108) boot: End of partition table
I (613) esp_image: segment 0: paddr=00010020 vaddr=3c110020 size=2e108h (188680) map
I (647) esp_image: segment 1: paddr=0003e130 vaddr=3fc9ae40 size=01ee8h (  7912) load
I (648) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=10d36ch (1102700) map
I (820) esp_image: segment 3: paddr=0014d394 vaddr=3fc9cd28 size=02fc4h ( 12228) load
I (822) esp_image: segment 4: paddr=00150360 vaddr=40374000 size=16e34h ( 93748) load
I (843) esp_image: segment 5: paddr=0016719c vaddr=600fe000 size=0002ch (    44) load
I (844) esp_image: segment 6: paddr=001671d0 vaddr=600fe030 size=01010h (  4112) load
I (859) boot: Loaded app from partition at offset 0x10000
I (859) boot: Disabling RNG early entropy source...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378f40
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x15e8
load:0x403b6000,len:0xbf8
load:0x403ba000,len:0x31c8
entry 0x403b621c
I (29) boot: ESP-IDF v4.4.2-378-g9269a536ac 2nd stage bootloader
I (29) boot: compile time 11:36:39
I (29) boot: chip revision: 0
I (32) qio_mode: Enabling default flash chip QIO
I (37) boot.esp32s3: Boot SPI Speed : 80MHz
I (42) boot.esp32s3: SPI Mode       : QIO
I (47) boot.esp32s3: SPI Flash Size : 16MB
W (52) boot.esp32s3: PRO CPU has been reset by WDT.
W (57) boot.esp32s3: APP CPU has been reset by WDT.
I (63) boot: Enabling RNG early entropy source...
I (68) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (79) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (87) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (94) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (102) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (109) boot:  4 uf2              factory app      00 00 00410000 00040000
I (117) boot:  5 ffat             Unknown data     01 81 00450000 00bb0000
I (124) boot: End of partition table
I (629) esp_image: segment 0: paddr=00010020 vaddr=3c110020 size=2e108h (188680) map
I (658) esp_image: segment 1: paddr=0003e130 vaddr=3fc9ae40 size=01ee8h (  7912) load
I (659) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=10d36ch (1102700) map
I (830) esp_image: segment 3: paddr=0014d394 vaddr=3fc9cd28 size=02fc4h ( 12228) load
I (833) esp_image: segment 4: paddr=00150360 vaddr=40374000 size=16e34h ( 93748) load
I (854) esp_image: segment 5: paddr=0016719c vaddr=600fe000 size=0002ch (    44) load
I (855) esp_image: segment 6: paddr=001671d0 vaddr=600fe030 size=01010h (  4112) load
I (870) boot: Loaded app from partition at offset 0x10000
I (870) boot: Disabling RNG early entropy source...

PaulskPt, 2022-10-21:
@jepler No. both SSID as PASSWORD use normal characters, one dash '-' and one underscore '_'.
These I used in many CPY scripts without any problem.

PaulskPt, 2022-10-22:
A few minutes ago I re-flashed the esp-box with tinyuf2 combined.bin (I discovered that I flashed b4 using SPI mode 'DIO' and this afternoon I discovered that the serial output of the MCU says the SPI mode is 'QIO'. I also took the chance to copy the latest CPY version dated today. Unfortunately a script to test the WiFi connection with the AP here in the room resulted in 'WDT expired'. See images below.

[image]

[image]

PaulskPt, 2022-10-22:
`Saturday 2022-10-22 18h57 utc+1

Adafruit CircuitPython 8.0.0-beta.3-6-g5a6fffca4 on 2022-10-22; ESP32-S3-Box-2.5 with ESP32S3
Board ID:espressif_esp32s3_box
UID:C7FD1A1E7C45

Serial output:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378f28
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x15e8
load:0x403b6000,len:0xbf8
load:0x403ba000,len:0x31c8
entry 0x403b621c
I (29) boot: ESP-IDF v4.4.2-378-g9269a536ac 2nd stage bootloader
I (29) boot: compile time 11:36:39
I (29) boot: chip revision: 0
I (32) qio_mode: Enabling default flash chip QIO
I (37) boot.esp32s3: Boot SPI Speed : 80MHz
I (42) boot.esp32s3: SPI Mode       : QIO
I (47) boot.esp32s3: SPI Flash Size : 16MB
W (52) boot.esp32s3: PRO CPU has been reset by WDT.       <<<================================  Crash / reset
W (57) boot.esp32s3: APP CPU has been reset by WDT.       <<<================================  -- idem --
I (63) boot: Enabling RNG early entropy source...
I (68) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (79) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (87) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (94) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (102) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (109) boot:  4 uf2              factory app      00 00 00410000 00040000
I (117) boot:  5 ffat             Unknown data     01 81 00450000 00bb0000
I (124) boot: End of partition table
I (629) esp_image: segment 0: paddr=00010020 vaddr=3c110020 size=2b5c0h (177600) map
I (656) esp_image: segment 1: paddr=0003b5e8 vaddr=3fc9ab70 size=0448ch ( 17548) load
I (659) esp_image: segment 2: paddr=0003fa7c vaddr=40374000 size=0059ch (  1436) load
I (662) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=1037f8h (1062904) map
I (831) esp_image: segment 4: paddr=00143820 vaddr=4037459c size=165d4h ( 91604) load
I (849) esp_image: segment 5: paddr=00159dfc vaddr=600fe000 size=0002ch (    44) load
I (849) esp_image: segment 6: paddr=00159e30 vaddr=600fe030 size=01010h (  4112) load
I (865) boot: Loaded app from partition at offset 0x10000
I (865) boot: Disabling RNG early entropy source...`

Neradoc, 2022-10-22:
I'm getting that with a DEBUG=1 build, it doesn't look like it gives any useful info to me, it just... stops when connecting.
```py
>>> import wifi
```
```
D (28309) esp_netif_lwip: LwIP stack has been initialized
D (28309) esp_netif_lwip: esp-netif has been successfully initialized
D (28309) event: created task for loop 0x3fcebd90
D (28309) event: running task for loop 0x3fcebd90
D (28309) event: created event loop 0x3fcebd90
D (28319) esp_netif_objects: esp_netif_add_to_list 0x3fceccbc
D (28329) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (28339) esp_netif_objects: esp_netif_add_to_list 0x3fced088
D (28339) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 2)
I (28349) pp: pp rom version: e7ae62f
I (28349) net80211: net80211 rom version: e7ae62f
I (28369) wifi:wifi driver task: 3fcafc98, prio:23, stack:3584, core=0
I (28369) system_api: Base MAC address is not set
I (28369) system_api: read default base MAC address from EFUSE
D (28379) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 8 bit
D (28379) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 0 bit
D (28389) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 24 bit
D (28399) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 16 bit
D (28409) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 8 bit
D (28409) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 0 bit
I (28419) wifi:wifi firmware version: f1d33cc
I (28419) wifi:wifi certification version: v7.0
I (28429) wifi:config NVS flash: disabled
I (28429) wifi:config nano formating: disabled
I (28439) wifi:Init data frame dynamic rx buffer num: 8
I (28439) wifi:Init management frame dynamic rx buffer num: 8
I (28449) wifi:Init management short buffer num: 32
I (28449) wifi:Init static tx buffer num: 16
I (28459) wifi:Init tx cache buffer num: 16
I (28459) wifi:Init static tx FG buffer num: 2
I (28459) wifi:Init static rx buffer size: 1600
I (28469) wifi:Init static rx buffer num: 4
I (28469) wifi:Init dynamic rx buffer num: 8
I (28479) wifi_init: rx ba win: 6
I (28479) wifi_init: tcpip mbox: 32
I (28479) wifi_init: udp mbox: 6
I (28489) wifi_init: tcp mbox: 6
I (28489) wifi_init: tcp tx win: 2880
I (28499) wifi_init: tcp rx win: 2880
I (28499) wifi_init: tcp mss: 1440
I (28499) wifi_init: WiFi IRAM OP enabled
I (28509) wifi_init: WiFi RX IRAM OP enabled
D (28509) ADC: Wi-Fi takes adc2 lock.
I (28519) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
D (28519) phy_init: loading PHY init data from application binary
D (28529) nvs: nvs_open_from_partition phy 0
E (28539) phy_init: esp_phy_load_cal_data_from_nvs: NVS has not been initialized. Call nvs_flash_init before starting WiFi/BT.
W (28549) phy_init: failed to load RF calibration data (0x1101), falling back to full calibration
D (28559) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 8 bit
D (28559) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 0 bit
D (28569) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 24 bit
D (28579) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 16 bit
D (28589) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 8 bit
D (28589) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 0 bit
D (28629) nvs: nvs_open_from_partition phy 1
D (28629) phy_init: esp_phy_store_cal_data_to_nvs: failed to open NVS namespace (0x1101)
D (28629) wifi:filter: set rx policy=0
I (28629) wifi:mode : sta (7c:df:a1:**:**:**)
I (28639) wifi:enable tsf
D (28639) wifi:filter: set rx policy=1
D (28639) wifi:connect status 0 -> 0
D (28649) event: running post WIFI_EVENT:2 with handler 0x4207dc14 and context 0x3fcecf34 on loop 0x3fcebd90
D (28659) wifi_init_default: wifi_start esp-netif:0x3fceccbc event-id2
D (28659) wifi_init_default: WIFI mac address: 7c df a1 ** ** **
D (28669) esp_netif_handlers: esp_netif action has started with netif0x3fceccbc from event_id=2
[... 272 lines cut ...]
D (4253) ledc: LEDC_PWM CHANNEL 0|GPIO 45|Duty 0000|Time 0
I (4263) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4263) gpio: GPIO[1]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4273) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4283) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4293) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4303) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4313) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4323) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4333) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4343) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4353) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4363) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4373) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4383) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[... cut ...]
I (4413) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4413) gpio: GPIO[46]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4423) gpio: GPIO[47]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (4433) I2S: i2s_driver_uninstall(2047): I2S port 0 has not installed
E (4443) I2S: i2s_driver_uninstall(2047): I2S port 1 has not installed
I (4543) espidf: Reserving 0 bytes of psram
D (4543) intr_alloc: Connected src 38 to int 9 (cpu 1)
```

dhalbert, 2022-10-22:
@PaulskPt We talked about doing a similar test on your UM ESP32-S3 ProS3. Do I understand that a similar script did _not_ fail on that board? So it seems to Box-specific?

Neradoc, 2022-10-23:
I can't reproduce it with the Feather ESP32-S3 TFT, it seems to run all fine, `wifi.radio.connect` or `.env`.

PaulskPt, 2022-10-23:
@Neradoc  thank you for your investigation.
@danh. I soldered the pin-strips to my new UM ProS3. Initial tests OK. Then I re-flashed it with tthe tinyuf2 bootloader for CPY. Loaded the really latest (from 2022-10-22) CPY V8. Wrote a similar script to connect to the same WiFi AP in the room here. It works great. I copied output of it to the dropbox folder.

I just copied also an .env file to the CIRCUITPY disk of the UM ProS3. I see that it works. After boot the statusbar shows an IP-address:

```]0;🐱92.168.1.111 | code.py | 8.0.0-beta.3-7-g17bda551c\]0;🐍192.168.1.111 | code.py | 8.0.0-beta.3-7-g17bda551c\```

Yes, it looks that it is the esp-box (Note it is not an esp-box-lite !) specific.

PaulskPt, 2022-10-23:
Here REPL output in mu-editor from the WiFi test with the new UM ProS3.
Btw: it's programmed to request date & time string from AIO Time Service every 2 minutes.

```
WiFi test for UM ProS3
Waiting for mu-editor etc. getting ready
wifi.radio.hostname= 'UMProS3'
wifi.radio.mac_address= f4:12:fa:42:4:ec
pr_scanned_ap(): 
Available WiFi networks (sorted on: 'rssi')
	AP nr 15	SSID: Vodafone-287778_EXT             	RSSI: -52	Channel:  9
	AP nr 12	SSID: MEO-AFM_EXT                     	RSSI: -66	Channel:  3
	AP nr  1	SSID: devolo-1a9                      	RSSI: -73	Channel:  6
	AP nr 16	SSID: Vodafone-287778                 	RSSI: -76	Channel:  9
	AP nr 14	SSID: MEO-WiFi                        	RSSI: -80	Channel:  3
	AP nr 13	SSID: MEO-AFM                         	RSSI: -80	Channel:  3
	AP nr  5	SSID: MEO-WiFi                        	RSSI: -80	Channel: 11
	AP nr  4	SSID: MEO-FEEE80                      	RSSI: -80	Channel: 11
	AP nr  2	SSID: MEO-WiFi                        	RSSI: -84	Channel:  6
	AP nr  3	SSID: MEO-FIBRA                       	RSSI: -85	Channel:  6
	AP nr  6	SSID: MEO-WiFi                        	RSSI: -88	Channel: 11
	AP nr  7	SSID: MEO-8997C0                      	RSSI: -89	Channel: 11
	AP nr  8	SSID: ThomsonDAA08B                   	RSSI: -90	Channel: 11
	AP nr 17	SSID: MEO-78CEB0                      	RSSI: -92	Channel:  9
	AP nr 19	SSID: Vodafone-FEE539                 	RSSI: -92	Channel: 10
	AP nr  9	SSID: Vodafone-48D3E0                 	RSSI: -92	Channel: 11
	AP nr 10	SSID: MEO-WiFi                        	RSSI: -92	Channel: 11
	AP nr 11	SSID: MEO-WiFi                        	RSSI: -93	Channel: 11
	AP nr 18	SSID: Mateus                          	RSSI: -93	Channel:  4
connected to Vodafone-287778_EXT!
IP address is 192.168.1.111
Resolved google address: '142.250.185.14'
Ping no response
Ping no response
Ping no response
Ping no response
Ping google.com [142.250.185.14]: 18 ms
Entering loop
elapsed_t= 10. WiFi status: connected
elapsed_t= 20. WiFi status: connected
elapsed_t= 30. WiFi status: connected
elapsed_t= 40. WiFi status: connected
elapsed_t= 50. WiFi status: connected
elapsed_t= 60. WiFi status: connected
elapsed_t= 70. WiFi status: connected
elapsed_t= 80. WiFi status: connected
elapsed_t= 90. WiFi status: connected
elapsed_t=100. WiFi status: connected
elapsed_t=110. WiFi status: connected
elapsed_t=120. WiFi status: connected
get_time_fm_AIO(): ip= 192.168.1.111
----------------------------------------
Time=  2022-10-23 01:33:08.599 296 7 +0100 WEST
----------------------------------------
```

PaulskPt, 2022-10-24:
I, from my end, restarted my investigation.  For what it is worth:
The file: [mpconfigboard.mk](https://github.com/adafruit/circuitpython/blob/main/ports/espressif/boards/espressif_esp32s3_box/mpconfigboard.mk) is the file most recently modified: 18 days ago.  In this file the flash mode is defined as 'dio' however a serial output at boot of the esp32-s3-box states flash mode 'QIO'. See composite screenshot below:

[image]

PaulskPt, 2022-10-24:
As a comparison: the board that is similar: Unexpected Maler esp32s3 pros3, which does not go into a boot-loop when trying to connect to a WiFi-Access Point, has in the mpconfigboard.mk the flash mode as 'qio', the same as in its serial output during boot. See the composite screenshot below:

[image]

PaulskPt, 2022-10-24:
Since Seon Rozenblum (Expected Maker) recently sent me updated replacement for the three models of his esp32s3 series of boards, I am able to compare the serial output at boot time with the flash definitions in the respective mpconfigboard.mk files of the latest edition of CPY V8. Here is the comparison for the board UM tinys3. As you can see: the same difference as with the esp32s3_box: dio in mpconfigboard.mk versus qio in the serial output at boot time.

[image]

PaulskPt, 2022-10-24:
The same situation for the board unexpectedmaker_feathers3:

[image]

PaulskPt, 2022-10-24:
Just flasthed the UM FeatherS3 with CPY V8 latest. Here is the serial output at boot time while running the first WiFi test on this board:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1644
load:0x403b6000,len:0xf1c
load:0x403ba000,len:0x31c8
entry 0x403b634c
I (25) boot: ESP-IDF v4.4.2-378-g9269a536ac 2nd stage bootloader
I (25) boot: compile time 11:37:31
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling QIO for flash chip WinBond
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (72) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (79) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (86) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (94) boot:  4 uf2              factory app      00 00 00410000 00040000
I (101) boot:  5 ffat             Unknown data     01 81 00450000 00bb0000
I (109) boot: End of partition table
I (613) esp_image: segment 0: paddr=00010020 vaddr=3c110020 size=2bef0h (179952) map
I (645) esp_image: segment 1: paddr=0003bf18 vaddr=3fc9a520 size=04100h ( 16640) load
I (649) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=103bd4h (1063892) map
I (813) esp_image: segment 3: paddr=00143bfc vaddr=3fc9e620 size=00358h (   856) load
I (813) esp_image: segment 4: paddr=00143f5c vaddr=40374000 size=1651ch ( 91420) load
I (836) esp_image: segment 5: paddr=0015a480 vaddr=600fe000 size=0002ch (    44) load
I (836) esp_image: segment 6: paddr=0015a4b4 vaddr=600fe030 size=01010h (  4112) load
I (851) boot: Loaded app from partition at offset 0x10000
I (852) boot: Disabling RNG early entropy source...
```

The WiFi test was successful. Received an IP-address. Positive ping on 'google.com'.

PaulskPt, 2022-10-24:
To make it complete: just flashed also the new UM TinyS3 with CPY latest:
```Adafruit CircuitPython 8.0.0-beta.3-11-gedce717cf on 2022-10-23; TinyS3 with ESP32S3```
Then I ran the same WiFi test. Result: successfull. Received an IP-address. Also a positive ping on 'google.com'.
Below the serial output at boot:

```

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1644
load:0x403b6000,len:0xe74
load:0x403ba000,len:0x31c8
entry 0x403b632c
I (25) boot: ESP-IDF v4.4.2-378-g9269a536ac 2nd stage bootloader
I (25) boot: compile time 11:37:42
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (71) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (86) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (93) boot:  4 uf2              factory app      00 00 00410000 00040000
I (101) boot:  5 ffat             Unknown data     01 81 00450000 003b0000
I (108) boot: End of partition table
I (613) esp_image: segment 0: paddr=00010020 vaddr=3c110020 size=2bf20h (180000) map
I (645) esp_image: segment 1: paddr=0003bf48 vaddr=3fc9a520 size=040d0h ( 16592) load
I (649) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=10420ch (1065484) map
I (813) esp_image: segment 3: paddr=00144234 vaddr=3fc9e5f0 size=00388h (   904) load
I (813) esp_image: segment 4: paddr=001445c4 vaddr=40374000 size=1651ch ( 91420) load
I (836) esp_image: segment 5: paddr=0015aae8 vaddr=600fe000 size=0002ch (    44) load
I (836) esp_image: segment 6: paddr=0015ab1c vaddr=600fe030 size=01010h (  4112) load
I (852) boot: Loaded app from partition at offset 0x10000
I (852) boot: Disabling RNG early entropy source...```

dhalbert, 2022-10-24:
> Regarding the CPY port definitions for the board 'esp32s3_box', the file: [mpconfigboard.mk](https://github.com/adafruit/circuitpython/blob/main/ports/espressif/boards/espressif_esp32s3_box/mpconfigboard.mk) is the file most recently modified: 18 days ago. In this file the flash mode is defined as `dio` however a serial output at boot of the esp32-s3-box states flash (SPI) mode `qio`.

Those three `CIRCUITPY_ESP_FLASH_*` constants are only used to construct the flags to pass to esptool.py in `ports/espressif/Makefile`. They are not used for any compilation variations. It does sound like it should be changed, but I don't think it's causing this particular problem.

PaulskPt, 2022-10-24:
Thank you. I also expected an answer like this however I wanted to mention it because it is not correct.

PaulskPt, 2022-10-25:
Any news for me? Some development?

dhalbert, 2022-10-25:
Nope, sorry, I am working on ESP32-nn sleep issues at the moment.

PaulskPt, 2022-10-25:
Thank you. I just managed to create a build of CPY 8.0.0-beta.3-12-gf67d2794c-dirty\  with DEBUG=1.
I think I have some interesting serial output that will help us further. 
However I don't understand everything from the errors printed in red but you and @Neradoc maybe will.
I made three screenshots because they show you the real problem points in red color.
Beside that I also  copied the text output version. See all below please.

[image]

[image]

[image]

```
Tuesday 2022-10-25 20h40 utc+1
Platform: ESP32-S3-Box 
Circuitpython V8.0.0-beta.3-11-

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x15e8
load:0x403b6000,len:0xbf8
load:0x403ba000,len:0x31c8
entry 0x403b621c
I (24) boot: ESP-IDF v4.4.2-378-g9269a536ac 2nd stage bootloader
I (25) boot: compile time 11:36:39
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 16MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (71) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (86) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (93) boot:  4 uf2              factory app      00 00 00410000 00040000
I (101) boot:  5 ffat             Unknown data     01 81 00450000 00bb0000
I (108) boot: End of partition table
I (613) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=4c8cch (313548) map
I (665) esp_image: segment 1: paddr=0005c8f4 vaddr=3fc9c620 size=03724h ( 14116) load
I (668) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=114920h (1132832) map
I (843) esp_image: segment 3: paddr=00174948 vaddr=3fc9fd44 size=01d40h (  7488) load
I (845) esp_image: segment 4: paddr=00176690 vaddr=40374000 size=1861ch ( 99868) load
I (868) esp_image: segment 5: paddr=0018ecb4 vaddr=600fe000 size=0002ch (    44) load
I (868) esp_image: segment 6: paddr=0018ece8 vaddr=600fe030 size=01010h (  4112) load
I (884) boot: Loaded app from partition at offset 0x10000
I (885) boot: Disabling RNG early entropy source...
I (897) opi psram: vendor id : 0x0d (AP)
I (897) opi psram: dev id    : 0x02 (generation 3)
I (897) opi psram: density   : 0x03 (64 Mbit)
I (901) opi psram: good-die  : 0x01 (Pass)
I (905) opi psram: Latency   : 0x01 (Fixed)
I (910) opi psram: VCC       : 0x01 (3V)
I (915) opi psram: SRF       : 0x01 (Fast Refresh)
I (920) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (926) opi psram: BurstLen  : 0x01 (32 Byte)
I (931) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (937) opi psram: DriveStrength: 0x00 (1/1)
W (942) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
D (952) MSPI Timing: tuning success, best point is index 5
I (957) spiram: Found 64MBit SPI RAM device
I (962) spiram: SPI RAM mode: sram 80m
I (967) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (974) cpu_start: Pro cpu up.
I (977) cpu_start: Starting app cpu, entry point is 0x4037856c
[... 464 lines cut ...]
D (3432) ledc: In slow speed mode, using clock 1
D (3432) ledc: LEDC_PWM CHANNEL 0|GPIO 45|Duty 0000|Time 0
I (3442) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3442) gpio: GPIO[1]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3452) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3462) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3472) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3482) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3492) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3502) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3512) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3522) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3532) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3542) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3552) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3562) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3572) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3582) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3592) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3592) gpio: GPIO[46]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3602) gpio: GPIO[47]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (3612) I2S: i2s_driver_uninstall(2047): I2S port 0 has not installed
E (3622) I2S: i2s_driver_uninstall(2047): I2S port 1 has not installed
I (3722) espidf: Reserving 0 bytes of psram
D (3722) intr_alloc: Connected src 38 to int 9 (cpu 1)```

PaulskPt, 2022-10-25:
About the HW TIMER errors: the error is reported four times. The hardware timers are described here: [General Purpose Timers](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/timer.html). The doc starts to describe that there are 2 Timer groups. Each group having 2 timers. When I google'd for the error I found about four hits. Some had to do with IR-applications.

PaulskPt, 2022-10-25:
About the errors ```I2S port 0/1 has not installed```.  The [doc about I2S](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32s3/api-reference/peripherals/i2s.html) says: 

```ESP32-S3 contains two I2S peripheral(s). These peripherals can be configured to input and output sample data via the I2S driver.``` [...] 

Driver uninstall: 

```If the established communication is no longer required, the driver can be removed to free allocated resources by calling [i2s_driver_uninstall()](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32s3/api-reference/peripherals/i2s.html#_CPPv420i2s_driver_uninstall10i2s_port_t).```

dhalbert, 2022-10-25:
I believe we see these errors in the log output of all builds. It would be worth comparing the log output of a board that works with the Box output.

PaulskPt, 2022-10-25:
OK. However I don't understand your last sentence. Can you rephrase pse.

dhalbert, 2022-10-25:
I'm saying that you will probably see these same log errors on output from a board that works, such as the UM FeatherS3. It would be worth verifying that is the case, and if you see error reports that differ, to note that. For instance, I am see the `HW TIMER NEVER INIT ERROR` right now on a test program on a Feather ESP32-S2 that is not using wifi at all.

PaulskPt, 2022-10-25:
Thank you for the elaboration. I'll try to build a debug version for the UM FeatherS3.

PaulskPt, 2022-10-26:
I managed to build a CPY V8b3-11 for the UM ESP32 FeatherS3 with DEBUG=1.
As soon as I try to issue wifi.radio.contact(ssid=ssid, password=password) the CPU halts and disk is lost. I have to reflash the bootloader.
The same happens as soon as I copy the file .env to the drive .

See the serial output below:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1644
load:0x403b6000,len:0xf1c
load:0x403ba000,len:0x31c8
entry 0x403b634c
I (25) boot: ESP-IDF v4.4.2-378-g9269a536ac 2nd stage bootloader
I (25) boot: compile time 11:37:31
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling QIO for flash chip WinBond
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (72) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (79) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (86) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (94) boot:  4 uf2              factory app      00 00 00410000 00040000
I (101) boot:  5 ffat             Unknown data     01 81 00450000 00bb0000
I (109) boot: End of partition table
I (613) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=4d0d4h (315604) map
I (666) esp_image: segment 1: paddr=0005d0fc vaddr=3fc9bd40 size=02f1ch ( 12060) load
I (669) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=114b94h (1133460) map
I (844) esp_image: segment 3: paddr=00174bbc vaddr=3fc9ec5c size=02250h (  8784) load
I (846) esp_image: segment 4: paddr=00176e14 vaddr=40374000 size=17d3ch ( 97596) load
I (868) esp_image: segment 5: paddr=0018eb58 vaddr=600fe000 size=0002ch (    44) load
I (868) esp_image: segment 6: paddr=0018eb8c vaddr=600fe030 size=01010h (  4112) load
I (884) boot: Loaded app from partition at offset 0x10000
I (884) boot: Disabling RNG early entropy source...
I (897) spiram: Found 64MBit SPI RAM device
I (897) spiram: SPI RAM mode: sram 80m
I (897) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (902) cpu_start: Pro cpu up.
I (906) cpu_start: Starting app cpu, entry point is 0x4037847c
I (0) cpu_start: App cpu up.
I (1830) spiram: SPI SRAM memory test OK
D (1830) efuse: In EFUSE_BLK2__DATA4_REG is used 2 bits starting with 0 bit
D (1831) efuse: In EFUSE_BLK2__DATA4_REG is used 8 bits starting with 13 bit
D (1838) efuse: In EFUSE_BLK1__DATA3_REG is used 3 bits starting with 24 bit
D (1845) efuse: In EFUSE_BLK2__DATA4_REG is used 2 bits starting with 0 bit
D (1852) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 20 bit
D (1859) efuse: In EFUSE_BLK1__DATA5_REG is used 8 bits starting with 3 bit
D (1866) efuse: In EFUSE_BLK1__DATA5_REG is used 5 bits starting with 11 bit
D (1873) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 13 bit
D (1880) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 20 bit
D (1887) efuse: In EFUSE_BLK1__DATA4_REG is used 5 bits starting with 27 bit
D (1894) efuse: In EFUSE_BLK1__DATA5_REG is used 3 bits starting with 0 bit
D (1901) efuse: In EFUSE_BLK1__DATA5_REG is used 8 bits starting with 3 bit
D (1908) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 13 bit
D (1915) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 20 bit
D (1923) efuse: In EFUSE_BLK1__DATA4_REG is used 5 bits starting with 27 bit
D (1930) efuse: In EFUSE_BLK1__DATA5_REG is used 3 bits starting with 0 bit
D (1937) efuse: In EFUSE_BLK1__DATA5_REG is used 8 bits starting with 3 bit
D (1952) clk: RTC_SLOW_CLK calibration value: 3635379
I (1960) cpu_start: Pro cpu start user code
I (1960) cpu_start: cpu freq: 240000000
I (1961) cpu_start: Application information:
[... 264 lines cut ...]
D (6535) wifi:filter: set rx policy=5
I (6535) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
D (6545) wifi:connect_op: status=0, auth=5, cipher=3
D (6555) gdma: new group (0) at 0x3fcb81dc
D (6555) gdma: new pair (0,0) at 0x3fcb8218
D (6555) gdma: new tx channel (0,0) at 0x3fcb8090
D (6565) gdma: new rx channel (0,0) at 0x3fcb8238
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). 
Exception was unhandled.  <<<================  .env present in Drive

Core  0 register dump:
PC      : 0x40056ea0  PS      : 0x00060130  A0      : 0x8002f1fd  A1      : 0x3fcedfe0
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3fcee083
A6      : 0x4002eb04  A7      : 0x00000000  A8      : 0x3fcf0ec5  A9      : 0x3fcf0ec6
A10     : 0x3fceea54  A11     : 0x00000020  A12     : 0x3fcf0c44  A13     : 0x3fcee083
A14     : 0x3fcee059  A15     : 0x00ffffff  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000001  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x40056e9d:0x3fcedfe0 |<-CORRUPTED

ELF file SHA256: 7adcc34a72169957```

Next, when I reflash tthe tinyuf2 bootloader and then copy the 'normal' file (adafruit-circuitpython-unexpectedmaker_feathers3-en_US-20221023-edce717.uf2), downloaded from 'circuitpython.org',
then the script runs OK and automatic connection to WiFi, using the credentials from file '.env' works OK.
The CPY statatusbar showed an IP-address.

PaulskPt, 2022-10-26:
```
Wednesday 2022-10-26 11h48 utc+1

board: Unexpected Maker FeatherS3

Flashed
1) Adafruit tinyuf2 bootloader;
2) Own build: Adafruit CircuitPython 8.0.0-beta.3-12-gf67d2794c-dirty   with DEBUG=1

Serial output using PuTTY on desktop PC MS Windows 11 Pro:

1st boot:
---------

Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1644
load:0x403b6000,len:0xf1c
load:0x403ba000,len:0x31c8
entry 0x403b634c
I (25) boot: ESP-IDF v4.4.2-378-g9269a536ac 2nd stage bootloader
I (25) boot: compile time 11:37:31
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling QIO for flash chip WinBond
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (72) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (79) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (86) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (94) boot:  4 uf2              factory app      00 00 00410000 00040000
I (101) boot:  5 ffat             Unknown data     01 81 00450000 00bb0000
I (109) boot: End of partition table
I (613) esp_image: segment 0: paddr=00010020 vaddr=3c110020 size=2bef0h (179952) map
I (645) esp_image: segment 1: paddr=0003bf18 vaddr=3fc9a520 size=04100h ( 16640) load
I (649) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=103bd4h (1063892) map
I (813) esp_image: segment 3: paddr=00143bfc vaddr=3fc9e620 size=00358h (   856) load
I (813) esp_image: segment 4: paddr=00143f5c vaddr=40374000 size=1651ch ( 91420) load
I (836) esp_image: segment 5: paddr=0015a480 vaddr=600fe000 size=0002ch (    44) load
I (836) esp_image: segment 6: paddr=0015a4b4 vaddr=600fe030 size=01010h (  4112) load
I (851) boot: Loaded app from partition at offset 0x10000
I (852) boot: Disabling RNG early entropy source...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download                                         # | <<<=== Put board in DFU mode 

Started the Espressif Flash Download Tool V3.9.3
-------------------------------------------------
Settings: 
SPI SPEED: 40 MHz
SPI MODE: QIO
DoNotChgBin: checked
After pressing 'START' button the following info appeared:

- In 'DetectedInfo':
flash vendor:
EFh: W
flash devID:
4018h
QUAD; 128Mbit
crystal:
40 Mhz

[... 965 lines cut ...]
D (2961741) wifi:new_bss=0x3fca9c40, cur_bss=0, new_chan=<9,0>, cur_chan=1
D (2961751) wifi:filter: set rx policy=5
I (2961751) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
D (2961761) wifi:connect_op: status=0, auth=5, cipher=3
D (2961761) gdma: new group (0) at 0x3fcb8374
D (2961771) gdma: new pair (0,0) at 0x3fcb7fd8
D (2961771) gdma: new tx channel (0,0) at 0x3fcb8344
D (2961771) gdma: new rx channel (0,0) at 0x3fcb83b0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40056ea0  PS      : 0x00060930  A0      : 0x8002f1fd  A1      : 0x3fcee0b0
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3fcee153
A6      : 0x4002eb04  A7      : 0x00000000  A8      : 0x3fcf0ec8  A9      : 0x3fcf0ec9
A10     : 0x3fceeb30  A11     : 0x00000020  A12     : 0x3fcf0c44  A13     : 0x3fcee153
A14     : 0x3fcee129  A15     : 0x00ffffff  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000001  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x40056e9d:0x3fcee0b0 |<-CORRUPTED

ELF file SHA256: 7adcc34a72169957

CPU halted.
-------------------------------------------------------------
```

PaulskPt, 2022-10-26:
So, with the 'own build' for the FeatherS3, trying to establish a WiFi connection 'by hand', from the REPL, the board's CPU halted.
When flashing the 'normal' CPY V8 b3, downloaded from 'circuitpython.org', WiFi contact establishes without problem on the UM FeatherS3.

OK, all this is 'noise'. The problem is with the Espressif ESP32-S3-Box.

PaulskPt, 2022-10-27:
During a Ubuntu-Linux session, I saw that the '/circuitpython/ports/espressif/esp-idf' version is 4.4.2 while the documentation of espressif expresses that for the esp32s3 esp-idf 5.0 is needed. Perhaps this is the source of the problem. Can I clone esp-idf 5.0 into ' /circuitpython/ports/espressif' or is that esp-idf a special version prepared for circuitpython ?

PaulskPt, 2022-10-27:
I copied 'esp-idf' (V 5.0) from elsewhere into /circuitpython/ports/espressif. It worked. firmware.uf2 etc built.

PaulskPt, 2022-10-27:
@MicroDev1  Here is the serial output from the firmware.uf2 that you creaated / sent to me:

```
Thursday 2022-10-27 11h55 utc+1
Board: (Adafruit) Espressif ESP32-S3-BOX

Flashed with Circuitpython firmware.uf2 received from @MicroDev

Adafruit CircuitPython 8.0.0-beta.3-20-g2bba6a68e on 2022-10-27; ESP32-S3-Box-2.5 with ESP32S3

Serial Output (after Reset) via PuTTY at Com5  (desktop PC with MS Windows 11 Pro)

ESP-ROM:esp32s3-2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x15e8
load:0x403b6000,len:0xbf8
load:0x403ba000,len:0x31c8
entry 0x403b621c
I (24) boot: ESP-IDF v4.4.2-378-g9269a536ac 2nd stage bootloader
I (25) boot: compile time 11:36:39
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 16MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (71) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (86) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (93) boot:  4 uf2              factory app      00 00 00410000 00040000
I (101) boot:  5 ffat             Unknown data     01 81 00450000 00bb0000
I (108) boot: End of partition table
I (613) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=4cd18h (314648) map
I (666) esp_image: segment 1: paddr=0005cd40 vaddr=3fc9c620 size=032d8h ( 13016) load
I (668) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=114960h (1132896) map
I (843) esp_image: segment 3: paddr=00174988 vaddr=3fc9f8f8 size=021dch (  8668) load
I (845) esp_image: segment 4: paddr=00176b6c vaddr=40374000 size=1861ch ( 99868) load
I (868) esp_image: segment 5: paddr=0018f190 vaddr=600fe000 size=0002ch (    44) load
I (868) esp_image: segment 6: paddr=0018f1c4 vaddr=600fe030 size=01010h (  4112) load
I (885) boot: Loaded app from partition at offset 0x10000
I (885) boot: Disabling RNG early entropy source...
I (897) opi psram: vendor id : 0x0d (AP)
I (897) opi psram: dev id    : 0x02 (generation 3)
I (897) opi psram: density   : 0x03 (64 Mbit)
I (901) opi psram: good-die  : 0x01 (Pass)
I (906) opi psram: Latency   : 0x01 (Fixed)
I (910) opi psram: VCC       : 0x01 (3V)
I (915) opi psram: SRF       : 0x01 (Fast Refresh)
I (920) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (926) opi psram: BurstLen  : 0x01 (32 Byte)
I (931) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (937) opi psram: DriveStrength: 0x00 (1/1)
W (942) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
D (953) MSPI Timing: tuning success, best point is index 5
I (957) spiram: Found 64MBit SPI RAM device
I (962) spiram: SPI RAM mode: sram 80m
I (967) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (974) cpu_start: Pro cpu up.
I (978) cpu_start: Starting app cpu, entry point is 0x4037856c
I (0) cpu_start: App cpu up.
I (1410) spiram: SPI SRAM memory test OK
D (1411) efuse: In EFUSE_BLK2__DATA4_REG is used 2 bits starting with 0 bit
D (1411) efuse: In EFUSE_BLK2__DATA4_REG is used 8 bits starting with 13 bit
D (1418) efuse: In EFUSE_BLK1__DATA3_REG is used 3 bits starting with 24 bit
D (1425) efuse: In EFUSE_BLK2__DATA4_REG is used 2 bits starting with 0 bit
[... 456 lines cut ...]
I (3442) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3442) gpio: GPIO[1]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3452) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3462) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3472) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3482) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3492) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3502) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3512) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3522) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3532) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3542) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3552) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3562) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3572) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3582) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3592) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3592) gpio: GPIO[46]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3602) gpio: GPIO[47]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (3612) I2S: i2s_driver_uninstall(2047): I2S port 0 has not installed
E (3622) I2S: i2s_driver_uninstall(2047): I2S port 1 has not installed
I (3722) espidf: Reserving 0 bytes of psram
D (3722) intr_alloc: Connected src 38 to int 9 (cpu 1)

``

PaulskPt, 2022-10-27:
Here is the code.py I used:

```
###
#
# Circuitpython V8b3-11 for ESP32-S3-Box (with DEBUG=1)
# WiFi test
#

import wifi
import sys
import time
#import ssl
import ipaddress
import socketpool
from secrets import secrets
#import adafruit_requests
#from rtc import RTC

my_debug = False

ssid = secrets["ssid"]
password = secrets["password"]
use_ping = True
pool = None
connected = False
ip = None
s_ip = None

print(f"\nTrying to connect to \'{ssid}\'")
wifi.__init__()

wifi.radio.connect(ssid=ssid, password=password)

ip = wifi.radio.ipv4_address

if ip:
    s_ip = str(ip)
    le_s_ip = len(s_ip)

if s_ip is not None and len(s_ip) > 0 and s_ip != '0.0.0.0':
    connected = True
else:
    connected = False

if connected:
    s2 = ''
else:
    s2 = "Not"

print(f"{s2} connected to: \'{ssid}\'")

if my_debug:
    print("s_ip= \'{}\'".format(s_ip))

if use_ping and connected:
    if not pool:
        pool = socketpool.SocketPool(wifi.radio)
    addr_idx = 1
    addr_dict = {0:'LAN gateway', 1:'google.com'}
    info = pool.getaddrinfo(addr_dict[addr_idx], 80)
    addr = info[0][4][0]
    print(f"Resolved google address: \'{addr}\'")
    ipv4 = ipaddress.ip_address(addr)
    for _ in range(10):
        result = wifi.radio.ping(ipv4)
        if result:
            print("Ping google.com [%s]: %.0f ms" % (addr, result*1000))
            break
        else:
            print("Ping no response")

print("Exiting...")
sys.exit()

```

PaulskPt, 2022-10-27:
Hardware: (Adafruit ESP32-S3-Box and Adafruit CP2102N.

[image]

[image]

PaulskPt, 2022-10-27:
Thank you Dan. I cannot find back yet the page where I read "needed". I am searching for it.

PaulskPt, 2022-10-27:
Ok Dan. You aare correct by stating that 'This unit is not made by Adafruit', however I bought it from Adafruit. (May 3, 2022). I doubt the unit is defective. It worked when I received it. The factory demo, using WiFi worked OK; speech command recognition (that sends the voice commands to a server of Espressif): worked. II am going to flash the factory demo again and make a short video to demonstrate. Maybe a defect is there now. We'll find out.

I see that you have a lot 'on your plate' so to speak. I am greatful that you are willing to test your own sample of the Box to see if you can reproduce the problem.

Here an image I took of the prodoct label of Adafruit on the packaging of the ESP-S3-Box on the day I received it. The order was first sent to the address of my son who lives in Goshen (near Louisville), KY. He then sent the ordered items to me.

[image]

PaulskPt, 2022-10-27:
https://user-images.githubusercontent.com/9107950/198345164-c3ef8258-39c4-4fd5-b217-8a0cb5f1f44e.mp4

PaulskPt, 2022-10-27:
Just for the record. Serial output after reset with the factory demo flashed.

```

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x17b4
load:0x403b6000,len:0xe0c
load:0x403ba000,len:0x3078
entry 0x403b626c
I (25) boot: ESP-IDF v4.4-264-g2bdea81b2a 2nd stage bootloader
I (25) boot: compile time 10:06:42
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 16MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 sec_cert         Unknown data     01 06 0000d000 00003000
I (71) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (86) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (93) boot:  4 fctry            WiFi data        01 02 00019000 00006000
I (101) boot:  5 ota_0            OTA app          00 10 00020000 0039d000
I (108) boot:  6 storage          Unknown data     01 82 003bd000 0028a000
I (116) boot:  7 model            Unknown data     01 82 00647000 007e9000
I (124) boot: End of partition table
I (128) esp_image: segment 0: paddr=00020020 vaddr=3c140020 size=1fc3ach (2081708) map
I (452) esp_image: segment 1: paddr=0021c3d4 vaddr=3fca57c0 size=03c44h ( 15428) load
I (460) esp_image: segment 2: paddr=00220020 vaddr=42000020 size=13d6a4h (1300132) map
I (657) esp_image: segment 3: paddr=0035d6cc vaddr=3fca9404 size=01688h (  5768) load
I (659) esp_image: segment 4: paddr=0035ed5c vaddr=40378000 size=1d7c0h (120768) load
I (686) esp_image: segment 5: paddr=0037c524 vaddr=50000000 size=00014h (    20) load
I (698) boot: Loaded app from partition at offset 0x20000
I (698) boot: Disabling RNG early entropy source...
I (699) opi psram: vendor id : 0x0d (AP)
I (703) opi psram: dev id    : 0x02 (generation 3)
I (708) opi psram: density   : 0x03 (64 Mbit)
I (713) opi psram: good-die  : 0x01 (Pass)
I (718) opi psram: Latency   : 0x01 (Fixed)
I (723) opi psram: VCC       : 0x01 (3V)
I (728) opi psram: SRF       : 0x01 (Fast Refresh)
I (733) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (738) opi psram: BurstLen  : 0x01 (32 Byte)
I (743) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (749) opi psram: DriveStrength: 0x00 (1/1)
W (754) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (765) spiram: Found 64MBit SPI RAM device
I (769) spiram: SPI RAM mode: sram 80m
I (774) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (781) cpu_start: Pro cpu up.
I (784) cpu_start: Starting app cpu, entry point is 0x403794f0
```

dhalbert, 2022-10-27:
OK, if I disable PSRAM completely, it works. So there is something about the PSRAM setup that is causing trouble.

I am going to hide a bunch of comments in this issue to expose the ones that are the most relevant.

dhalbert, 2022-10-27:
OK, I think this is a dupe of #6206. Closing this in favor of that bug, which gets to the root issue faster. @PaulskPt wait up before trying more things, as I am on the trail of the bug.

PR #7140 (2022-10-27): correct SPIRAM settings for ESP32-S3-BOX
Fixes #7093 and #6206.

Fix sdkconfig flags for ESP32-S3-BOX SPIRAM. The original flags were wrong in one or more ways, copied from someplace else, I think. I did not try to diagnose each wrong setting, but instead came up with a simpler set.

Note that `CONFIG_SPIRAM` subsumes `ESP32S3_SPIRAM_SUPPORT`.
from `esp32s3/Kconfig`:
```
    # Hint: to support SPIRAM across multiple chips, check CONFIG_SPIRAM instead
    config ESP32S3_SPIRAM_SUPPORT
        bool "Support for external, SPI-connected RAM"
        default "n"
        select SPIRAM
        help
  [... cut]
closed 2022-10-27 by dhalbert