adafruit/circuitpython #7105: CPX: sound meter: RuntimeError: All sync event channels in use

chickenbit, 2022-10-22:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.1 on 2022-10-01; Adafruit CircuitPlayground Express with samd21g18
Board ID:circuitplayground_express

UF2 Bootloader v3.14.0 SFHWRO
Model: CPlay Express
Board-ID: SAMD21G18A-CPlay-v0
```

### Code/REPL

```python
while True:
    mic.record(samples, len(samples))
    magnitude = normalized_rms(samples)
    # You might want to print this to see the values.
    # print(magnitude)

    # Compute scaled logarithmic reading in the range 0 to NUM_PIXELS
    c = log_scale(constrain(magnitude, input_floor, input_ceiling),
                  input_floor, input_ceiling, 0, NUM_PIXELS)

    # Light up pixels that are below the scaled and interpolated magnitude.
    pixels.fill(0)
    for i in range(NUM_PIXELS):
        if i < c:
            pixels[i] = volume_color(i)
        # Light up the peak pixel and animate it slowly dropping.
        if c >= peak:
            peak = min(c, NUM_PIXELS - 1)
        elif peak > 0:
            peak = peak - 1
        if peak > 0:
            pixels[int(peak)] = PEAK_COLOR
    pixels.show()
```

### Behavior

**Expected Behavior**

1. Able to edit _sound meter_ ([Example 1 Source](https://learn.adafruit.com/adafruit-circuit-playground-express/playground-sound-meter), [Example 2 Source from ](https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/download/20221021/adafruit-circuitpython-bundle-8.x-mpy-20221021.zip)
2. Save
3. Reload
4. No exceptions

**Actual behavior**: RuntimeError exception. Press reset button to bring CPX back to good state

circuitplayground_sound_meter.py example periodically fails with:

```python
code.py output:
]0;🐍code.py | 8.0.0-beta.1\Traceback (most recent call last):
  File "code.py", line 94, in <module>
RuntimeError: All sync event channels in use
```
or (depends on source example)

```python
]0;🐍code.py | 8.0.0-beta.1\Traceback (most recent call last):
  File "code.py", line 38, in <module>
RuntimeError: All sync event channels in use
]0;🐍38@code.py RuntimeError | 8.0.0-beta.1\
Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

Steps to reproduce:

1. Update code.py with contents of adafruit-circuitpython-bundle-8.x-mpy-20220930\examples\circuitplayground_sound_meter.py and save
2. Update code.py with contents of [Playground Sound Meter @ Adafruit]((https://learn.adafruit.com/adafruit-circuit-playground-express/playground-sound-meter) example and save

### Description

The sound meter examples were always flaky from my experience. I have been going through the examples the past weeks. Initially I was not monitoring the serial port. I would use the sound meter examples as-is and the neopixels just blink red. Initially I thought maybe it was a doc bug and the CPX was not supported. But as I kept reading and trying things again, it ended up working after a reset but was flakey after that.

Since updating boot loader and Circuit Python to 8.x I found another source for the sound_meter example and am able to reproduce using the two reference examples that have no modifications.

Power Source: USB connected to computer
CPX is only connected to USB and battery-pack (switch in off position)

reset via reset button on CPX brings CPX back to working state.

### Additional information

Filing issue after discussion on adafruit forum: https://forums.adafruit.com/viewtopic.php?p=944454

- Originally hit with Circuit Python 7.3.3 but asked to reproduce on 8.x
- https://circuitpython.org/board/circuitplayground_express/
- No specific notes on running this exact example, but another example I suspect can be used to reproduce issue: https://docs.circuitpython.org/projects/circuitplayground/en/latest/examples.html#id29

petewarden, 2022-11-07:
Just a note that I'm seeing this intermittently too, on a CPX when calling `cpx.play_file()`. I haven't prepared a minimal repro case or bootloader/version info yet, I'm crunching to get https://github.com/usefulsensors/person_sensor_tv_remote/blob/main/code.py done unfortunately and it's not a blocker for me, but I wanted to at least mention that it comes up outside of the original context.

chickenbit, 2022-11-22:
thank you @tannewt!

tannewt, 2022-11-22:
@dhalbert fixed it. I just reviewed and merged. :-)

chickenbit, 2022-11-22:
> @dhalbert fixed it. I just reviewed and merged. :-)

noticed that later…thanks @dhalbert for the fix!

PR #7217 (2022-11-16): samd21: port_disable_tick() should disable event channel
Fixes #7105.

On SAMD21, `port_disable_tick()` did not finish the job of disabling the `_tick_event_channel`.

Also added a check to avoid doing anything if `_tick_event_channel` was already disabled.

Before this change, editing a file and causing an autoreload would use up a bunch of event channels:
(this is from some temporary logging I added:
```
[code.py edited here]
Code stopped by auto-reload. Reloading soon.
>port_disable_tick, _tick_event_channel: 0
<port_enable_tick, _tick_event_channel: 0
>port_disable_tick, _tick_event_channe [... cut]