adafruit/circuitpython #7107: Possible manifestation of memory move bug after imports. Rotaryio/Displayio - Crashes board out.

xgpt, 2022-10-23:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; Seeeduino XIAO RP2040 with rp2040
Board ID:seeeduino_xiao_rp2040
```

### Code/REPL

```python
BROKEN:

#code.py

import gc
from AxDisp import *
print(gc.mem_free())
from AxInput import *
print(gc.mem_free())
last_text_top='test'
text_top(last_text_top)
import time
while True:
    print(gc.mem_free())
    if Ax1Check():
        last_text_mid='True'
        text_mid(last_text_mid)
    else:
        last_text_mid='False'
        text_mid(last_text_mid)

```python
#AxDisp.py
import board
import adafruit_ssd1306
import busio
i2c=busio.I2C(board.SCL, board.SDA)#, frequency=800_000)
oled = adafruit_ssd1306.SSD1306_I2C(128, 32, i2c)
def off():
    oled.fill(0)
    oled.show()
def on():
    oled.fill(1)
    oled.show()
def text_top(x):
    oled.fill(0)
    oled.text(x,0,0,1)
    oled.show()
def text_mid(x):
    oled.fill(0)
    oled.text(x,0,8,1)
    oled.show()
def text_low(x):
    oled.fill(0)
    oled.text(x,0,16,1)
    oled.show()
```
```python
#AxInput.py
import board
import digitalio
import rotaryio

Ax1 = digitalio.DigitalInOut(board.D6)
Ax1.switch_to_input(pull=digitalio.Pull.UP)
Ax2 = digitalio.DigitalInOut(board.D7)
Ax2.switch_to_input(pull=digitalio.Pull.UP)
Ax3 = digitalio.DigitalInOut(board.D8)
Ax3.switch_to_input(pull=digitalio.Pull.UP)
Ax4 = digitalio.DigitalInOut(board.D9)
Ax4.switch_to_input(pull=digitalio.Pull.UP)
Ax5= digitalio.DigitalInOut(board.D10)
EncSw=digitalio.DigitalInOut(board.D1)
EncSw.switch_to_input(pull=digitalio.Pull.UP)
Ax5.switch_to_input(pull=digitalio.Pull.UP)
Enc= rotaryio.IncrementalEncoder(board.D3, board.D2, 2)

#def EncPosCheck():
#    if Enc.position>1:
#        Enc.position=0
#        return 1
#    elif Enc.position<1:
#        Enc.position=0
#        return -1
#    else:
#        return 0
def Ax1Check():
    if not Ax1.value:
        return 1
    elif  Ax1.value:
        return 0
def Ax2Check():
    if not Ax2.value:
        return 1
    elif  Ax2.value:
        return 0
def Ax3Check():
    if not Ax3.value:
        return 1
    elif  Ax3.value:
        return 0
def Ax4Check():
    if not Ax4.value:
        return 1
    elif  Ax4.value:
        return 0
def Ax5Check():
    if not Ax5.value:
        return 1
    elif  Ax5.value:
        return 0
```

Neradoc fixed:
```python
#code.py
import gc
from AxDisp import *
from AxInput import *
disp_init()
input_init()

last_text_top='test'
text_top(last_text_top)
import time
while True:
    print(gc.mem_free())
    print(gc.collect())
    if Ax1Check():
        last_text_mid='True'
        text_top(last_text_mid)
    else:
        last_text_mid='False'
        text_mid(last_text_mid)
    text_low(str(EncPosCheck()))
```
```python
#AxDisp.py

import board
import adafruit_ssd1306
import busio

oled = None
def disp_init():
    global oled
    i2c = busio.I2C(board.SCL, board.SDA)#, frequency=800_000)
    oled = adafruit_ssd1306.SSD1306_I2C(128, 32 , i2c)

def off():
    oled.fill(0)
    oled.show()
def on():
    oled.fill(1)
    oled.show()
def text_top(x):
    oled.fill(0)
    oled.text(x,0,0,1)
    oled.show()
def text_mid(x):
    oled.fill(0)
    oled.text(x,0,10,1)
    oled.show()
def text_low(x):
    oled.fill(0)
    oled.text(x,0,20,1)
    oled.show()
```
```python
#AxInput.py
import board
import digitalio
import rotaryio

Ax1 = digitalio.DigitalInOut(board.D6)
Ax1.switch_to_input(pull=digitalio.Pull.UP)
Ax2 = digitalio.DigitalInOut(board.D7)
Ax2.switch_to_input(pull=digitalio.Pull.UP)
Ax3 = digitalio.DigitalInOut(board.D8)
Ax3.switch_to_input(pull=digitalio.Pull.UP)
Ax4 = digitalio.DigitalInOut(board.D9)
Ax4.switch_to_input(pull=digitalio.Pull.UP)
Ax5 = digitalio.DigitalInOut(board.D10)
EncSw = digitalio.DigitalInOut(board.D1)
EncSw.switch_to_input(pull=digitalio.Pull.UP)
Ax5.switch_to_input(pull=digitalio.Pull.UP)
Enc = None

def input_init():
    global Enc
    Enc = rotaryio.IncrementalEncoder(board.D3, board.D2, 2)

def EncPosCheck():
   if Enc.position>1:
       Enc.position=0
       return 1
   elif Enc.position<0:
       Enc.position=0
       return -1
   else:
       return 0
def Ax1Check():
    if not Ax1.value:
        return 1
    elif  Ax1.value:
        return 0
def Ax2Check():
    if not Ax2.value:
        return 1
    elif  Ax2.value:
        return 0
def Ax3Check():
    if not Ax3.value:
        return 1
    elif  Ax3.value:
        return 0
def Ax4Check():
    if not Ax4.value:
        return 1
    elif  Ax4.value:
        return 0
def Ax5Check():
    if not Ax5.value:
        return 1
    elif  Ax5.value:
        return 0
```
```

### Behavior

Crashes. Requires reset. No REPL error on serial.

We think it might have something to do with this bug: https://github.com/adafruit/circuitpython/issues/2687

### Description

Error related to importing a module with rotaryio and displayio.

### Additional information

Instantiating the display and rotary object AFTER import does NOT cause this bug.

dhalbert, 2022-10-23:
Please try with a recent build, after #6789.

xgpt, 2022-10-23:
Fixed with move to 8.x beta 3. Thanks Neradoc and Dan for your assistance in triaging this!

closed 2022-10-23 by xgpt